letscontrolit/ESPEasy #4753: Feature request - wmbus protocol support?

ghtester, 2023-08-02:
As I have two Zenner Minomess water meters with Wireless M-Bus support (wM-Bus C1 mode) I need to read remotely, I am looking for a most simple solution. Although I was not able to get a detailed technical information related to mentioned hardware yet (except the basic info on local seller's webpage, unfortunately in local language: https://www.vodomery.cz/nabidka-bytovych-vodomeru/bytovy-vodomer-apz-zenner-wireless-m-bus.html ), there's a project for another water meter model, hopefully with the same protocol, using the cc1101 chip:

https://github.com/maciekn/izar-wmbus-esp  

The question is  - could it be possible to add the support for that protocol / chip to the ESP Easy firmware please?

TD-er, 2023-08-02:
You can use the CUL firmware on a Arduino Nano + CC1101 868 MHz module.
Then the CUL reader plugin can be used to send out the raw strings to a MQTT controller.

This must then later be decoded and the key of the encoding should be printed on your WMbus device.

The reason to go for a separate Arduino board is to make sure the reception of data is not interrupted by other stuff running on the ESP.

ghtester, 2023-08-02:
Thanks a lot for the quick hint, I'll consider it. Maybe I'll try to experiment with RPi & RTL2832 receiver at first, as I have both already... There's a nice 'how to': 
https://github.com/xaelsouth/rtl-wmbus
https://github.com/wmbusmeters/wmbusmeters

tonhuisman, 2023-08-02:
FYI: We already support the CC1101 in [P118](https://espeasy.readthedocs.io/en/latest/Plugin/P118.html), though dedicated to the Itho and Orcon protocols 😃

tonhuisman, 2023-08-02:
> unfortunately in local language: https://www.vodomery.cz/nabidka-bytovych-vodomeru/bytovy-vodomer-apz-zenner-wireless-m-bus.html ),

[image]

It seems they don't like us... 🤔

ghtester, 2023-08-03:
Huh, that's really strange... :-( I am sorry for this. :-(
Could you please try this translated link? 
https://www-vodomery-cz.translate.goog/nabidka-bytovych-vodomeru/bytovy-vodomer-apz-zenner-wireless-m-bus.html?_x_tr_sl=cs&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp

Some parameters from the webpage:
<html><body>
<h2><font style="vertical-align: inherit;"><font style="vertical-align: inherit;">Technical data of the M-Bus module</font></font></h2> 
                          

Operating frequency | Radiated transmission power | The length of the broadcast message | Broadcast frequency | Data transfer method | Encryption of radio protocols | Error detection | Sent data | Optical IR interface | Battery capacity | Pointer or display | Power source | Backflow detection | Degree of coverage | Environmental conditions | Conformity according to CE
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
868 MHz | ~14dBm | ~ 1 ms | every 180 s* | wireless M-Bus (mode C1) | yes (encryption mode 5) | CRC (yes) | serial number, date, counter status, values ​​for the previous month (max. 15), status information of the radio module | Yes | 10 years plus reserve from activation of radio transmission | seven-digit roller counter | lithium battery | Yes | IP 68 | +5 °C to +55 °C | according to Directive 2014/53/EU (RED

</body>
</html>

[image]

ghtester, 2023-08-03:
> FYI: We already support the CC1101 in [P118](https://espeasy.readthedocs.io/en/latest/Plugin/P118.html), though dedicated to the Itho and Orcon protocols 😃

Thanks a lot for info, I have already ordered the module and I'll try to contact the local ZENNER's seller to get a more detailed technical info. AFAIK the transfers are encrypted and most likely a decryption key would be needed.

TD-er, 2023-08-03:
The decryption key should be printed on the module.
However in ESPEasy there is no implementation of such decryption, but as I mentioned the CUL reader will output the entire WMBUS packet to MQTT so you can decode it elsewhere.

ghtester, 2023-08-03:
Well, I would like to avoid sending every raw data stream to MQTT broker as there are many other water meters (of the same model, maybe even with the same shared decryption key) around so it would be nice to have the basic filter at the receiver's side.

TD-er, 2023-08-03:
That's where the CUL reader plugin can help :)

N.B. I can also send you a newer version of this plugin which had improved filtering.

ghtester, 2023-08-22:
I have just received the CC1101 module so I have looked at CUL Reader & P118 plugins, compiled the custom firmware with these plugins.
It looks CUL Reader supports T mode while my meters are C mode. Also it's not very clear which arduino module I need (I suppose I would need to buy the CSM module from busware.de ?)
I would prefer ESP8266 node to receive the data directly from CC1101 like P118 plugin does... And send at least the raw data to MQTT controller (I understand decoding the encrypted packets would be a hard task for ESP8266). 

@tonhuisman Do you think it could be possible to create a new plugin, based on P118, to support the ZENNER Minomess protocol please? I tried to e-mail you yesterday with some info, not sure if you received it...

TD-er, 2023-08-22:
The CUL reader allows you to send a command to the CUL, so you can switch modes.
Not sure if that's something you may need to add/enable when compiling the CUL firmware.

tonhuisman, 2023-08-23:
Have responded via email.
Won't be picking this one up now, because of a lack of time, knowledge and unavailability of the used hardware for testing. Maybe (much) later.

ghtester, 2023-08-23:
@TD-er OK but so far I did not find any details about the HW necessary for CUL Reader plugin. It indirectly refers to busware.de and I suppose this module should be supported: https://busware.de/tiki-index.php?page=CSM I also suppose it should perhaps work (as you outlined earlier) with a standard Arduino Nano based on Atmega 328 chip but I did not find an appropriate schema so far.
@tonhuisman   Thanks a lot for your reply, I fully understand and also sent a reply to your e-mail. 
I know how much complicated is to develop a HW related SW if you don't have an access to that HW... I also know there are more important tasks to do now and that your effort here cost a lot of your time. You are doing a great job here which is very appreciated.

closed 2023-08-23 by ghtester