letscontrolit/ESPEasy #4754: P020 add serial config

3dalex07, 2023-08-03:
hello
It is posible to add to serial config  10bit / parity: None / stop  bits: 1
it need to read data from table scales wa301p

Thanks

TD-er, 2023-08-03:
Huh? Not sure if there is any serial port config allowing > 8 bits.
Really strange setup.
Are you sure it isn't like they count start and stop bits?
I never heard of any serial protocol using 10 bits.

TD-er, 2023-08-03:
I did try to find any info via Google, but I can't find anyting about the protocol they use.
Do you have something like a datasheet?

3dalex07, 2023-08-03:
[image]

TD-er, 2023-08-03:
I think this is just normal RS232 communication.
See: https://circuitdigest.com/article/rs232-serial-communication-protocol-basics-specifications
[image]
Since you don't need parity and 1 stop bit, it should be as basic as possible.
It even sends the CR and LF after each message.

uzi18, 2023-08-03:
[STM32](https://github.com/redbear/STM32-Arduino/blob/b35fb3de95ba4faee41026cb38d0cc18e53ff84f/arduino/cores/RedBear_Duo/firmware/hal/inc/usart_hal.h#L55) has 9N1,9N2

tonhuisman, 2023-08-03:
@3dalex07 Have you already tried to hook up the scales and see what data is being sent?
The suggested 8N1 config seems best fitting (and _very_ standard mode of serial connection...)

3dalex07, 2023-08-04:
i found another user manual 
[WA301P.pdf](https://github.com/letscontrolit/ESPEasy/files/12258229/WA301P.pdf)

tonhuisman, 2023-08-04:
> i found another user manual [WA301P.pdf](https://github.com/letscontrolit/ESPEasy/files/12258229/WA301P.pdf)

Well, then it's quite clear:
[image]

9600 baud (suggested) with 8 data bits N (no parity) and 1 stop bit. The start bit is required anyway.
They are just being silly adding the start and stop bits to the data bits stating there is a '10 digits data transfer format', confusing any user. They're describing it wrong, it's still 8 bits of data, with (the usual) start and stop bits. And 'verification' should have been written as 'parity' like the rest of the world does.

3dalex07, 2023-08-04:
Something interesting is happening. Nothing worked yesterday with these settings. everything works perfect today.
Many thanks to everyone for participating.

closed 2023-08-04 by 3dalex07

uzi18, 2023-08-04:
[image]
So it is ASCII, but why K as start code ?

tonhuisman, 2023-08-04:
> but why K as start code ?

Creativity? 🤯

TD-er, 2023-08-04:
It does have quite a lot of 1-0 transitions (1001011), so maybe it is used to 'wake up' some of their own connected displays?