letscontrolit/ESPEasy #4755: Rules are case-sensitive......

JoostDkr, 2023-08-03:
I noticed that existing rules no longer functioned properly. After debugging for a while I found the cause. I had a statement "on CO2#PPM do". That has always worked. Not anymore..... the statement should now be: "On CO2#PPM do".

Is it an idea to make the rules non-case-sensitive?

TD-er, 2023-08-03:
Hmm keywords should be case insensitive...
I will have a look.

What build are you using?
On what build did it still work case insensitive?

JoostDkr, 2023-08-03:
I used the sourcecode of June 2023 (version 1.80) and made a custom build.

TD-er, 2023-08-03:
And the previous build you ran which did allow lower case keywords (as it should) ?

tonhuisman, 2023-08-03:
> I had a statement "on CO2#PPM do". That has always worked. Not anymore..... the statement should now be: "On CO2#PPM do".

Please don't let the Syntax highlighter in the Rules tab confuse you, the all-lowercase syntax should work as intended, nothing has changed in that area. This feature can be turned off in a future release, as added via PR #4742 (though not yet released)

JoostDkr, 2023-08-03:
> And the previous build you ran which did allow lower case keywords (as it should) ?

Yes, it does. I have an application running on an earlier version of easpeasy that works with "on" without a capital letter.

TD-er, 2023-08-03:
> > And the previous build you ran which did allow lower case keywords (as it should) ?
> 
> Yes, it does. I have an application running on an earlier version of easpeasy that works with "on" without a capital letter.

And what build was that, so we have some 'range' to look for changes in the code.

JoostDkr, 2023-08-03:
I don't remember that and I can't find it in the application either. What I see in the application is the date of my own custom build (SP_Easy_mega_20221124_custom_ESP32_4M316k).

But is that important? The casesensitive occurs in the espeasy sourcecode of June 2023 (version 1.80)

TD-er, 2023-08-03:
> But is that important? The casesensitive occurs in the espeasy sourcecode of June 2023 (version 1.80)

Well, just looking at the code (which hasn't changed for a while), I really can't see how it should not work...
```c++
  const bool lineStartsWith_on = line.substring(0, 3).equalsIgnoreCase(F("on "));
```
As you can see, the compare is with "ignore case".
That specific line has changed 4 years ago and the function with that line was changed 16 months ago.

I can assure you that digging through 4 years of commits (or even 16 months) is a lot of work, so having some range to search could reduce this.

JoostDkr, 2023-08-03:
Well, I downloaded and builded the latest espeasy source (I should have done that before posting the issue) and found that the rules are not (anymore) case sensitive.
No sense to analyze this issue further and it can be closed.
Thanks for helping and thinking along.