letscontrolit/ESPEasy #4758: Schriftart ESP Flasher

Rodenberger, 2023-08-08:
Hallo,

wie bekomme ich die Schriftart von ESPEasy geändert???
Bei mir ist sie sehr groß.
DANKE !

tonhuisman, 2023-08-08:
Please use English language here, if possible (or use Google Translate), as the audience exceeds DE/EU, and the common language on Github is English too.

The font used by the ESPEasy Flasher tool can not be changed directly. Most likely you are using a high-res 4k monitor?
Possible solution can be to create a shortcut to the executable, then right-click that shortcut and on the Compatibility tab change the HighDPI settings (you may have to experiment on what's working for you).

(My Windows is configured for English so I can only give the English terms...)

TD-er, 2023-08-08:
The intended fonts for the ESPEasy flasher are included in the fonts directory.

But like Ton mentioned, it is best to use either the Espressif download tool, or the [web flasher](https://td-er.nl/ESPEasy/)
Problem with the ESPEasy Flasher is that it only supports ESP8266 and does have some issues with some USB to serial chips.

tonhuisman, 2023-08-26:
We have removed the, no longer maintained and deprecated, ESP.Easy.Flasher tool from the repository and from the releases since Release 20230822, as there are several alternatives available, mentioned above.
AFAICS, this issue can be closed.

closed 2023-08-26 by TD-er