letscontrolit/ESPEasy #4760: Plugin 133 LTR390 - Different values returned when Resolution changes

iz8mbw, 2023-08-11:
Hello.
I have notice that the values returned by the LTR390 changes when the Resolution changes.
Example in a room when the luminosity is almost the same:

With 18 bit of resolution:
[image]

With 20 bit of resolution:
[image]

With 13 bit of resolution:
[image]

As you can see only "Lux" values is almost the same but "Ambient", "UV" and "UV Index" values are very different.
I expect to have the same values when the resolution changes (I expect changing resolution only effect the "precision" of the values).

ping @tonhuisman 

Thank you!

TD-er, 2023-08-11:
Those 'derived' values often are computed based on a ratio between several sensor values.
Is it possible one of them is clipping?

iz8mbw, 2023-08-11:
These are other examples when sensor is exposed to the sun:

With 18 bit of resolution:
[image]

With 20 bit of resolution:
[image]

With 13 bit of resolution:
[image]

tonhuisman, 2023-08-11:
You seem to be using the Dual mode setting, can you try using either ALS or UV mode, to see if that's closer to your expected results?

iz8mbw, 2023-08-11:
This in single mode by reading "Ambient reading only":

With 18 bit of resolution:
[image]

With 20 bit of resolution:
[image]

With 13 bit of resolution:
[image]

It looks the same behavior of Dual mode.

tonhuisman, 2023-08-11:
The calculated values UVIndex and Lux are both based on the UV and Ambient readings, then calculated using the Gain and Resolution settings. The formulas are included in the library, using the `float` datatype and not altered by me.

The UV and Ambient values displayed are read _directly_ from the sensor, without any calculation whatsoever. Readings are only done once the sensor indicates that new data is ready.

If switching from Dual to ALS or UV mode would have made a significant difference, then it could have been because of a mishap by me, but as it gives similar results, that doesn't seem to be the case 😅.

The Dual mode was 'invented' by me, by alternating between ALS and UV reading mode, effectively halving the number of samples read for each measurement, but still providing all available values.

TL;DR; I think this is how this sensor handles the Resolution and Gain settings. Possibly having a low gain causes a lot of inaccuracy because of the probably `integer` dividing/bit shifting done in the sensor.

tonhuisman, 2023-08-12:
I'm suggesting to close this issue as it seems 'that's how it is' for this sensor. I did notice these differences when originally testing the plugin, but assumed/accepted the behavior as by design.
It appears to me that at 13 bit resolution, the measurements are so inaccurate that these shouldn't be taken too seriously. They allow for a very quick response from the sensor, but as the UV levels won't change significantly within 12.5 msec. it would be better to use a higher resolution, at least the sensor default of 18 bit, but when using an Interval setting of more than a few seconds, the 20 bit max. resolution should be used, as that takes max. 400 msec. to complete.
This is the resolution to conversion time table from the data-sheet:
[image]

iz8mbw, 2023-08-14:
Since UVIndex and Lux values are both calculated based on the UV and Ambient readings, what is strange here is the "raw" values of UV and Ambient reported are very different when resolution changes.
What I mean is that I can expect a litlle different on calculated values but for me it's strange if the sensor reports different values when resolution changes. I cannot see a relationship between resolution and values reported by the sensor.

Anyway if it's "by desgin" on the sensor side and it's not an issue with the Library then OK.

Thanks.

TD-er, 2023-08-14:
It looks like the raw values are just lower, when you lower the resolution.
Which makes sense.
For example 20 bit resolution has a range from 0 ... (2^20 - 1) and 13 bit has a range from 0 ... 8191
So I assume the calculation of the derived values like Lux, will take this into account and thus scale the values when needed.

But there's probably one other aspect when changing the resolution, which is that the exposure time also changes.
Or more accurate, the "integration time".
Light sensitive sensors (also the pixels in a digital camera) sometimes show a slight non-linear behavior when changing the integration time.
Partly because a capacitor needs to be discharged in less amount of time, causing a systematic offset. (this is mainly an issue with multi-pixel sensors like camera sensors, as this offset may differ per pixel)
I can imagine (pun intended :) ) the sensor will internally switch to use a different gain when using a different integration time.
This may also introduce an offset and non-linear behavior.

TD-er, 2023-08-14:
That "Ambient" value is suspiciously close to a power-of-2 value, so I suspect that sensor reading may overflow/clip.

tonhuisman, 2023-08-14:
What Resolution is used there? Ambient value 262143 is 18 bits all 1's, so probably a clipped value.

iz8mbw, 2023-08-14:
> What Resolution is used there? Ambient value 262143 is 18 bits all 1's, so probably a clipped value.

18 bit for both.

TD-er, 2023-08-14:
So that's the 18 bit clipping value.
If any of the sensors used to calculate a derived value is clipping, the resulting value no longer makes any sense.
Thus increasing the gain on a sunny day is not useful.
You may be lucky it won't clip even with gain set to 1x on a sunny day in Italy.
Even in the Netherlands you may occasionally get luminance values close to the limit of that sensor.

iz8mbw, 2023-08-29:
Hello.
So after some weeks of test I can confirm that set to 20 bit of resolution is the best approach when sensor is exposed to the sun and so the luminosity varies a lot.
Thanks as always for support!

closed 2023-08-29 by iz8mbw