letscontrolit/ESPEasy #4768: ADS1115 value are 0.00

mpl1337, 2023-08-16:
Hello,

im trying to read a pressure transducer on my ADS115 with ESP32. Iv got only 0.00 as Value. Whats wrong with my Setup?. Iv tryed allready a Sample Sketch on Arduino IDE and it works. The hardware is working.

Build: [ESP_Easy_mega_20230623_normal_ESP32_4M316k Jun 23 2023](https://github.com/letscontrolit/ESPEasy/releases)
[image]
[image]
[image]
[image]

tonhuisman, 2023-08-16:
Did you change the hardware to be at I2C addres 0x49? As 0x48 is the default 🤔

mpl1337, 2023-08-16:
Yes, ADDR is on +3.3. With 0x48 same.

tonhuisman, 2023-08-16:
And if you leave ADDR disconnected? Don't know how that address has to be set exactly, would need to lookup in the datasheet.

mpl1337, 2023-08-16:
i have a pressure transducer on 5V. SCL/SDA ADS1115 is connected to a levelshifter 5V<>3.3V

i have tryed all Analog Inputs, on every Input 0.00

Samplesketch:

AIN0: -1  -0.00V (A0 connected to GND)
AIN1: 1671  5.01V (A1 connected to +5V)
AIN2: 1105  3.32V (A2 connected to +3.3)
AIN3: 82  0.25V (nc)
[image]

TD-er, 2023-08-17:
OK, so the sensor is working and with the older build you proved it should work with ESPEasy...

Well that's clearly proven there's something wrong in ESPEasy then...
Will have a look at it.

TD-er, 2023-08-17:
I just looked at the history and some changed were merged on Mar 31.
Could you test with a build prior to that date to make sure it may have been caused by this pr: https://github.com/letscontrolit/ESPEasy/commit/afd5299718ca79afc0b7d010571afa415ed2fb18  ?

mpl1337, 2023-08-17:
Build: [ESP_Easy_mega_20230304_normal_ESP32_4M316k Mar 4 2023](https://github.com/letscontrolit/ESPEasy/releases)

still working

mpl1337, 2023-08-17:
Build: [ESP_Easy_mega_20230306_normal_ESP32_4M316k Mar 6 2023](https://github.com/letscontrolit/ESPEasy/releases)

working

mpl1337, 2023-08-17:
Build: [ESP_Easy_mega_20230409_normal_ESP32_4M316k Apr 9 2023](https://github.com/letscontrolit/ESPEasy/releases)

not working

TD-er, 2023-08-17:
Thanks, so it is clear which commit was faulty.
I will look into this.

mpl1337, 2023-08-17:
okay, im ready for the next test :D

TD-er, 2023-08-17:
Could you test with any non working build to enable the calibration and just set the 2nd row of the calibration to something like this:
1 -> 1

This way there is a calibration mapping from 0 -> 0 and 1 -> 1
The second entry can also be 100 -> 100 but I guess you get the idea :)

mpl1337, 2023-08-17:
ok?

[image]

mpl1337, 2023-08-17:
Still 0.00

TD-er, 2023-08-17:
Did you try using the "convert to volt" checked?
My theory is that the conversion to volt may have a multiplier set to 0...
But I don't see it yet in the code how this can be.

TD-er, 2023-08-17:
Also, can you check the debug log (if included in your build) to see if there is any report of the measured analog value

mpl1337, 2023-08-17:
Where can I find the debug log?

TD-er, 2023-08-17:
Can you try on the Tools->Advanced page to uncheck "Check I2C devices when enabled:"

mpl1337, 2023-08-17:
same, nothing about i2c

mpl1337, 2023-08-17:
maybe you want try it with teamviewer?

mpl1337, 2023-08-18:
> 743694: sendcontent free: 185792 chunk size:283
743702: sendcontent free: 180828 chunk size:0
744127: sendcontent free: 185664 chunk size:4096
744144: sendcontent free: 180840 chunk size:1290
744153: sendcontent free: 182156 chunk size:0
744234: static_file: /esp.css
744236: Serve 304: 1692346990 /esp.css
744236: HTML : Request file /esp.css
745600: sendcontent free: 185220 chunk size:4096
745615: sendcontent free: 183572 chunk size:1504
745621: sendcontent free: 174836 chunk size:0
745729: static_file: /esp.css
745730: Serve 304: 1692346990 /esp.css
745731: HTML : Request file /esp.css
746903: sendcontent free: 185704 chunk size:1218
746911: sendcontent free: 180732 chunk size:0
748091: sendcontent free: 185696 chunk size:283
748099: sendcontent free: 180732 chunk size:0
748204: sendcontent free: 185544 chunk size:405
748212: sendcontent free: 180580 chunk size:0
750246: sendcontent free: 185696 chunk size:450
750254: sendcontent free: 180732 chunk size:0
750570: HTTP: taskrun,1
750572: HTTP after parseTemplate: taskrun,1
750575: Command: taskrun
750575: taskrun,1
750576: Par1: 1 Par2: 0 Par3: 0 Par4: 0 Par5: 0
750596: sendcontent free: 185316 chunk size:4096
750612: sendcontent free: 183668 chunk size:1389
750619: sendcontent free: 178512 chunk size:0
750914: static_file: /esp.css
750915: Serve 304: 1692346990 /esp.css
750916: HTML : Request file /esp.css
752095: sendcontent free: 185688 chunk size:1090
752103: sendcontent free: 180736 chunk size:0

0.00

tonhuisman, 2023-08-18:
Can you download the most recent merge build for `mega` from this [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5850022962), to see if that makes a difference?

TD-er, 2023-08-18:
I have made a very 'professional looking' setup to test it...
[image]

[image]

The pins have 3 resistors from 3V3 ... GND.
When I switch GND and 3V3 to these resistors, I get this:
[image]

So it is working here...

TD-er, 2023-08-18:
The only thing I can think of is that the I2C address handling may be wrong???

tonhuisman, 2023-08-18:
Well, I tried on both 0x48 and 0x49 (just connect `ADDR` to `VCC`, like in the OP), and both worked fine. Address 0x4A is the one to avoid, as that requires the `SDA` signal to stay low for some minimal amount of time to be properly recognized, from the datasheet: `If SDA is used as the device address, hold the SDA line low for at least 100 ns after the SCL line goes low to make sure the device decodes the address correctly during I2C communication.`

TD-er, 2023-08-18:
Hmm I noticed the original issue is on ESP32, so I will rewire my setup for ESP32 to make sure there isn't something fishy in the latest ESP32 SDK...

"One moment please" as some Youtuber may say :)

tonhuisman, 2023-08-18:
I've done my testing on an ESP32-C3.

TD-er, 2023-08-18:
Fresh install on ESP32 (including flash erase, just  to be sure)

[image]

mpl1337, 2023-08-18:
ok, i need to order some new ADS1115.

i have connected a second ADS1115 and add an BH1750 to the i2c Bus.

The BH1750 works fine. Both ADS 0.00
[image]
[image]
[image]

TD-er, 2023-08-18:
Not sure if a new one will help here, as you've shown it was working with the older build and the Arduino test code.

Maybe you should try it with only 1 board?
I don't know what may happen if you have 2 boards with different I2C addresses set, but perhaps the set I2C address isn't that clearly set.
So maybe you're simply getting replies from both modules and thus it is considered to be a non-valid reply and thus no update.
On the older builds, there was an issue where we didn't check whether the received value was unambiguously valid, which resulted in a 'shift' of readings.
So the measurement from one pin was assigned to the other task.

mpl1337, 2023-08-18:
2 ADS was my last test.

Is there a way to insert a debug line in the code to get the raw value from the ADS on the serial?

Which environment i need to compile the binary by myself?

TD-er, 2023-08-18:
I have the IDE open here, so what build do you need?
I can add some debug code (INFO log level, so you're not flooded with logs)

TD-er, 2023-08-18:
```c++
bool P025_data_struct::read(float& value) const {
  if (!waitReady025(5)) { 
    addLog(LOG_LEVEL_INFO, F("ADS1115: Not Ready at start read"));
    return false; 
  }

  if (!I2C_write16_reg(_i2cAddress, P025_CONFIG_REGISTER, _configRegisterValue)) {
    addLog(LOG_LEVEL_INFO, F("ADS1115: Start measurement failed"));
    return false;
  }

  // See https://github.com/letscontrolit/ESPEasy/issues/3159#issuecomment-660546091
  if (!waitReady025(10)) { 
    addLog(LOG_LEVEL_INFO, F("ADS1115: Not Ready after start measurement"));

    return false; 
  }

  int16_t raw = 0;
  if (!readConversionRegister025(raw)) {
    addLog(LOG_LEVEL_INFO, F("ADS1115: Cannot read from conversion register"));
    return false;
  }

  value = _fullScaleFactor * raw;
  addLog(LOG_LEVEL_INFO, strformat(F("ADS1115: RAW value: %d, output value: %f"), raw, value));
  return true;
}
```
https://www.dropbox.com/scl/fi/l3gilvbvxmq480tkkicis/ESP_Easy_mega_20230818_energy_ESP32_4M316k_issue4768.zip?rlkey=njn9votljjvyri5tjs9e5t7mx&dl=0

mpl1337, 2023-08-18:
> INIT : Booting version: ESP_Easy_mega_20230818_energy_ESP32_4M316k, (Self built) bugfix/DefaultTaskVarNames_57975f0 (ESP32 SDK 4.4.5.230614)<\r>
<\n>193 : Info   : INIT : Free RAM:244192<\r>
<\n>194 : Info   : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reset by PRO CPU<\r>
<\n>195 : Info   : FS   : Mounting...<\r>
<\n>219 : Info   : FS   : Mount successful, used 141815 bytes of 290156<\r>
<\n>235 : Info   : CRC  : Settings CRC           ...OK<\r>
<\n>270 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>285 : Info   : CRC  : SecuritySettings CRC   ...OK <\r>
<\n><\0>332 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>334 : Info   : INIT : I2C<\r>
<\n>336 : Info   : INIT : Check for Priority tasks<\r>
<\n>338 : Info   : INIT : SPI not enabled<\r>
<\n>339 : Info   : Set Network mode: WiFi<\r>
<\n>442 : Info   : WIFI : Set WiFi to STA<\r>
<\n>549 : Info   : WiFi : Event STA Started<\r>
<\n>651 : Info   : WiFi : Start network scan all channels<\r>
<\n>6867 : Info   : WiFi : Scan finished, found: 5<\r>
<\n>6869 : Info   : WiFi : Best AP candidate: TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-70dBm) - bgn<\r>
<\n>6871 : Info   : WiFi : Added known candidate, try t<\0>o connect<\r>
<\n>6872 : Info   : WiFi : Best AP candidate: TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-70dBm) - bgn<\r>
<\n>6907 : Info   : ESPEasy console using ESPEasySe<\0>rial<\r>
<\n>6908 : Info   : INIT : Free RAM:194536<\r>
<\n>6963 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>6965 : Info   : INFO : Plugins: 60 [Normal][Energy] (ESP32 SDK 4.4.5.230614)<\r>
<\n>6968 : Info   : WiFi : Best AP candidate: TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-70dBm) - bgn<\r>
<\n>6969 : Info   : WIFI : Connecting TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-70dBm) - bgn attempt #0<\r>
<\n>6979 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED<\r>
<\n>7807 : Info   : Webserver: start<\r>
<\n>7896 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>8461 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>8491 : Info   : WIFI : Connected! AP: TP-Link_2FC8 (6C:5A:B0:70:2F:C8) Ch: 2 Duration: 1503 ms<\r>
<\n>8493 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: Conn. <\r>
<\n>8501 : Info   : WIFI : DHCP IP: 192.168.178.125 (Energy) GW: 192.168.178.1 SN: 255.255.255.0 DNS: 192.168.178.62 / 0.0.0.0   duration: 26 ms<\r>
<\n>8529 : Info   : mDNS : Starting mDNS...<\r>
<\n>8534 : Info   : mDNS : Started, with name: Energy.local<\r>
<\n>8659 : Info   : NTP  : NTP replied: delay 31 mSec Accuracy increased by 374 msec<\r>
<\n>8675 : Info   : mDNS : Starting mDNS...<\r>
<\n>8679 : Info   : mDNS : Started, with name: Energy.local<\r>
<\n>8691 : Info   : Time set to 1692358168.374<\r>
<\n>8692 : Info   : Current Time Zone: STD time start: 2023-10-29 03:00:00 offset: 0 min<\r>
<\n>8694 : Info   : Local time: 2023-08-18 11:29:28<\r>
<\n>8703 : Info   : UDP : Start listening on port 8266<\r>
<\n>8704 : Info   : firstLoopConnectionsEstablished<\r>
<\n>9042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>9154 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 178176 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init<\r>
<\n>10042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>11042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>12042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>13042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>14042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>15042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>16042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>17042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>18042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>19042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>20042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>21042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>22042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>23042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>24042 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>25042 : Info   : ADS1115: Not Ready after start measurement<\r>

TD-er, 2023-08-18:
OK, so that 10 msec may be too low here?
Which is surprising, but possible.

I'll make a build with it set to 20 msec, which should be more then enough.

TD-er, 2023-08-18:
[New test build](https://www.dropbox.com/scl/fi/udwjtmgket05iw0s3xgoo/ESP_Easy_mega_20230818_energy_ESP32_4M316k_issue4768_2.zip?rlkey=xedbgdrq7czma8j2jeixegwsb&dl=0)

mpl1337, 2023-08-18:
> INIT : Booting version: ESP_Easy_mega_20230818_energy_ESP32_4M316k, (Self built) bugfix/DefaultTaskVarNames_57975f0 (ESP32 SDK 4.4.5.230614)<\r>
<\n>194 : Info   : INIT : Free RAM:244192<\r>
<\n>195 : Info   : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reset by PRO CPU<\r>
<\n>196 : Info   : FS   : Mounting...<\r>
<\n>220 : Info   : FS   : Mount successful, used 141815 bytes of 290156<\r>
<\n>236 : Info   : CRC  : Settings CRC           ...OK<\r>
<\n>272 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>287 : Info   : CRC  : SecuritySettings CRC   ...OK <\r>
<\n><\0>334 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>336 : Info   : INIT : I2C<\r>
<\n>338 : Info   : INIT : Check for Priority tasks<\r>
<\n>340 : Info   : INIT : SPI not enabled<\r>
<\n>341 : Info   : Set Network mode: WiFi<\r>
<\n>444 : Info   : WIFI : Set WiFi to STA<\r>
<\n>551 : Info   : WiFi : Event STA Started<\r>
<\n>653 : Info   : WiFi : Start network scan all channels<\r>
<\n>7069 : Info   : WiFi : Scan finished, found: 5<\r>
<\n>7071 : Info   : WiFi : Best AP candidate: TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-68dBm) - bgn<\r>
<\n>7073 : Info   : WiFi : Added known candidate, try t<\0>o connect<\r>
<\n>7074 : Info   : WiFi : Best AP candidate: TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-68dBm) - bgn<\r>
<\n>7109 : Info   : ESPEasy console using ESPEasySe<\0>rial<\r>
<\n>7110 : Info   : INIT : Free RAM:194584<\r>
<\n>7165 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>7167 : Info   : INFO : Plugins: 60 [Normal][Energy] (ESP32 SDK 4.4.5.230614)<\r>
<\n>7170 : Info   : WiFi : Best AP candidate: TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-68dBm) - bgn<\r>
<\n>7171 : Info   : WIFI : Connecting TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-68dBm) - bgn attempt #0<\r>
<\n>7184 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED<\r>
<\n>8006 : Info   : Webserver: start<\r>
<\n>8021 : Info   : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 846 ms<\r>
<\n>8035 : Info   : WiFi : Event STA Stopped<\r>
<\n>8124 : Info   : WIFI : Connecting TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-68dBm) - bgn attempt #1<\r>
<\n>8136 : Info   : WiFi : Event STA Started<\r>
<\n>9030 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>9287 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>9383 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 185584 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED<\r>
<\n>11281 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>13277 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>15283 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>17302 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>19276 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>21294 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>23290 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>25273 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>27292 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>28144 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED<\r>
<\n>28145 : Info   : WiFi : WifiDisconnect()<\r>
<\n>28249 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'<\r>
<\n>28350 : Info   : Reset WiFi.<\r>
<\n>28353 : Info   : WiFi : Start network scan all channels<\r>
<\n>34368 : Info   : WiFi : Scan finished, found: 3<\r>
<\n>34371 : Info   : WiFi : Best AP candidate: TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-67dBm) - bgn<\r>
<\n>34372 : Info   : WiFi : Added known candidate, try to connect<\r>
<\n>34373 : Info   : WiFi : WifiDisconnect()<\r>
<\n>34477 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'<\r>
<\n>34479 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED<\r>
<\n>34580 : Info   : WIFI : Set WiFi to OFF<\r>
<\n>34899 : Info   : WIFI : Set WiFi to STA<\r>
<\n>34911 : Info   : WiFi : Event STA Started<\r>
<\n>35013 : Info   : WIFI : Connecting TP-Link_2FC8 6C:5A:B0:70:2F:C8 Ch:2 (-67dBm) - bgn attempt #2<\r>
<\n>35978 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>36457 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>36505 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>36553 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>36560 : Info   : WIFI : Connected! AP: TP-Link_2FC8 (6C:5A:B0:70:2F:C8) Ch: 2 Duration: 1497 ms<\r>
<\n>36561 : Info   : WIFI : Arduino wifi status: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. <\r>
<\n>36563 : Info   : WIFI : DHCP IP: 192.168.178.125 (Energy) GW: 192.168.178.1 SN: 255.255.255.0 DNS: 192.168.178.62 / 0.0.0.0   duration: 28 ms<\r>
<\n>36592 : Info   : mDNS : Starting mDNS...<\r>
<\n>36598 : Info   : mDNS : Started, with name: Energy.local<\r>
<\n>36679 : Info   : UDP : Start listening on port 8266<\r>
<\n>36680 : Info   : firstLoopConnectionsEstablished<\r>
<\n>36729 : Info   : NTP  : NTP replied: delay 31 mSec Accuracy increased by 409 msec<\r>
<\n>36745 : Info   : mDNS : Starting mDNS...<\r>
<\n>36748 : Info   : mDNS : Started, with name: Energy.local<\r>
<\n>36760 : Info   : Time set to 1692359144.410<\r>
<\n>36761 : Info   : Current Time Zone: STD time start: 2023-10-29 03:00:00 offset: 0 min<\r>
<\n>36763 : Info   : Local time: 2023-08-18 11:45:44<\r>
<\n>37255 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>39254 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>39353 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 177596 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init<\r>
<\n>41254 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>43255 : Info   : ADS1115: Not Ready after start measurement<\r>
<\n>45254 : Info   : ADS1115: Not Ready after start measurement<\r>

TD-er, 2023-08-18:
OK, so it isn't about the timeout, but rather receiving either an error from sending the read command to check if it is ready, or the nr of bytes received is wrong.

How is this chip connected to the I2C bus?
If you're using a level converter here, maybe this adds some delay?

Can you try to set this task to use "slow I2C clock"?
On the Hardware tab you can define what should be the slow clock (default 100 kHz)

mpl1337, 2023-08-18:
[image]

No lvl shifter. Minimal jumper wires 

Slow i2c clock with 50,100,200 same

mpl1337, 2023-08-18:
i have tested it on a NODEMCU esp8266 Board. Same Problem. i Thing these ADS1115 are some Bad fakes.

TD-er, 2023-08-18:
[Another testbuild](https://www.dropbox.com/scl/fi/b6cucv762habofdgy3c5z/ESP_Easy_mega_20230818_energy_ESP32_4M316k_issue4768_3.zip?rlkey=m523eoix4jomc3rr4zsewscgn&dl=0)

Main change is that I now allow for `!is_ok` as long as the timeout has not been reached.
```c++
bool P025_data_struct::waitReady025(unsigned long timeout_ms) const {
  const unsigned long timeout = millis() + timeout_ms;
  while (!timeOutReached(timeout)) {
    // bit15=0 performing a conversion   =1 not performing a conversion
    bool is_ok = false;
    const bool ready = (I2C_read16_reg(_i2cAddress, P025_CONFIG_REGISTER, &is_ok) & 0x8000) != 0;

    if (ready && is_ok) { return true; }
    delay(1);
  }
  return false;
}
```

mpl1337, 2023-08-18:
Not working.

Im sure that's my Ads1115 is a fake. It is a ADS1015. They changed only the name and selled the cheaper 1015 as 1115. The label on my 1115 is not readable. I think they scretched it down because a fake.

TD-er, 2023-08-18:
I will take a quick look at their differences, to see how much it will take to support both.

The main differences seem to be 12 bits vs. 16 bit for the ADS111x
And the ADS101x does have different sample rates.

mpl1337, 2023-08-18:
sounds good :)

have order 3 new devices. Hope no fakes :/

TD-er, 2023-08-19:
I can't seem to find in the datasheet why the ADS101x isn't working with the current code.
Could you perhaps (if you live in NL) send me one of those boards via mail?
I looked at various ADS1x15 boards offered on AliExpress and the blue ones specifically have either ADS1015 or ADS1115 written on them.
So I wonder what might be wrong with your specific board.

I will order a few ADS1015 boards myself, but that may also take a while for those to arrive.
But as I said, I'm not sure your board may be an ADS1015. Maybe it is a clone chip?

mpl1337, 2023-08-19:
I should get today my new order.

I'm from Germany. I will send you two chips. I have 5 from this Fakes.

TD-er, 2023-08-19:
That would be great.
Can you email me, so I can give you my address.

uzi18, 2023-08-19:
Try to add debug into I2C_read16_reg

TD-er, 2023-08-19:
Yep, that's the plan, but since it is rather tedious for me to make a test build, others to test and report etc., he agreed to send me one of his modules.

uzi18, 2023-08-19:
If Addressing is issue here, than you need to fight with logic analyser.
But first you can test if you read specifiec ADS, by for example write something into known register like lo/hi threshold.

uzi18, 2023-08-19:
@TD-er by lo/hi threshold registers you can Also detect 1015 vs 1115 ADS. eg. Write there 0x8001 and 0x7ffe if they are same after read it is 1115.

TD-er, 2023-08-19:
One of the things I was thinking of to detect whether it was an ADS1115 or 1015 was to check the least significant bits.
If they are always 0, we're dealing with an ADS1015.

The only real difference in config seems to be the data rate.
The default data rate for the 1115 is 128 samples per sec (thus a little less than 8 msec to take a sample)
However for the 1015 this is 1600 samples per sec.
So it should be ready long before this timeout of 10 msec used for the 1115.

The user has replied via email to tell me the newly arrived ADS1115 boards work just fine, so he will send me one of those non working boards.
Like I mentioned, I can't explain why it should not work with this code on 1015 boards, so I guess those specific ones could be buggy clones, or maybe a different silicon revision of the 1115.
So this does require hands on testing, like with the logic analyser and also putting them under the microscope to see what might be the actual chip being used.

mpl1337, 2023-08-20:
I delivered the package to the post office today. I think you should get it in 3-4 days

mpl1337, 2023-08-26:
Did you get the package?

TD-er, 2023-08-26:
Yep got it yesterday.
I have just looked at it under the microscope and the chip surface has indeed been sanded off.
The one with the pins soldered on was pressed through the enveloppe and all pins were bent flat, so I have turned them 'up' again so it fits in a breadboard, ready to test.

mpl1337, 2023-08-26:
ok, the soldered and the chip in the ESD bag are the same. When the pressed chip is broken, can you take the other one

TD-er, 2023-08-31:
Today I got the ADS1015 and the ADS1115 I ordered so I may have some reference material.

I just inserted my freshly bought ADS1015 module, which clearly states on the PCB as such and found that it was working just fine.
Given it should only output values shifted 4 bits, it should only show multiples of 16 when looking at the raw values.

See here the really strange looking multiples of 16:
[image]

Then looked at the numbering on the chips and I see "BOGI" which is an ADS1115 based on the datasheet.
The ADS1015 should have either one of these: (Last column)
[image]

The ADS1115:
[image]

So I guess I have been duped also.
[image]
N.B. this is a picture taken with my phone, so the entire board can also be seen.

mpl1337, 2023-08-31:
Have you try my chip?

TD-er, 2023-08-31:
I just messaged the seller, got some new beer and was just about to test your modules :)

mpl1337, 2023-08-31:
I needed five beers to accept that I get scammed :-D

[3 more replies]

PR #4786 (2023-09-06): [ADS1x15] Support ADS1015 & add sample rate selector
Make timeout depending on set sample rate + add sample rate selector

ToDo:
- [x] Document
- [x] Show detected version (ADS1015/ADS1115)
- [x] Add option for multiple taskvalues per task

Fixes: #4768 
Fixes: #1677
Fixes: #1564 
Fixes: #926
closed 2023-09-06 by TD-er