letscontrolit/ESPEasy #4769: SD Card send / download files manually or by rule

hrikken, 2023-08-17:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

tonhuisman, 2023-08-17:
Please explain what you think is not working, or your feature request, the text is currently quite undefined.

hrikken, 2023-08-17:
I cannot download files from a mounted SD Card via the ESP Easy GUI. When I power down the ESP and mount the SD Card in a PC I can open the files and see that the files are correct and contain the values I want.

tonhuisman, 2023-08-17:
What exact ESPEasy build are you using? (Is shown at the bottom of each page of the web-UI)

TD-er, 2023-08-17:
And how is the SD card formatted?
Are you using sub directories?
How long is the file name?
What file size do these files have? (and file type as it matters on how files are being served)

hrikken, 2023-08-17:
It is formatted as FAT. Filenames are the standard names that ESPEasy generates. Like ‘log.txt’ and ‘values.csv’. The build is:

ghtester, 2023-08-17:
I would try a concurrent web browser if possible, to confirm the issue is the same...

tonhuisman, 2023-08-17:
> It is formatted as FAT. Filenames are the standard names that ESPEasy generates. Like ‘log.txt’ and ‘values.csv’. The build is: 
> 
> Sent from Proton Mail for iOS

Some interesting information is missing here...?

hrikken, 2023-08-17:
[image]

hrikken, 2023-08-17:
Strange: all of a sudden all files are gone and no more writing at all. Don't no what happened. I relocated the ESP (power down and up again). Will try to reset but not this weekend. I'll let you know...  Thanks so far.

tonhuisman, 2023-08-17:
What size is the SD-card in use?

TD-er, 2023-08-17:
Sounds a bit like a power issue, or maybe just a bad connection between ESP and SD card reader?

hrikken, 2023-08-18:
Don't know what this is. SD not working at all anymore. Maybe HW failure in M5Stack. I'll buy a different SD Card holder and try it again. Close the issue for the time being. Thanks for the efforts...

hrikken, 2023-08-18:
> Are those other cards still working after you tried them in this M5Stack unit?

Yes they do