letscontrolit/ESPEasy #4773: TaskValueSet and Dummy (string)

almirus, 2023-08-20:
Hi! I try to concatenate all sensors value to string, it doesn't work 
```
On Clock#Time=All,**:** do // this event comes in every minute  
  TaskValueSet 2,1,[env#temperature],[env#env],[env#pressure],[uv#uvIndex],[dust#pm1],[dust#pm10],[dust#pm25],[rain#rain]
endon
```
[image]

tonhuisman, 2023-08-20:
ESPEasy doesn't support string values (yet), so 'concatenating' then, and  by using commas, won't work as you might want to intend. At least a + has to be used to add the values, but still nu dice when attempting to add strings.
You can set a single (float/double) value to each (up to 4) value of a Dummy device.

Please explain what you want to achieve.

TD-er, 2023-08-20:
The String data type should be considered as "pre-alpha".
It is currently only present in the Serial Proxy plugin and the CUL reader.
Both receive (large) strings and then can forward these directly to a MQTT controller.

That's the only way how it is now supported via tasks/plugins.

If you want to concatenate values and publish them to a MQTT controller, you can do so via rules.
In there you can simply create a string as argument for the `publish` command.
Just make sure to wrap the 2nd command argument in quotes if you need to use either spaces or commas (or both) in the message.

almirus, 2023-08-21:
> Please explain what you want to achieve.

There is an external web server that can receive requests no more than once every 5 minutes (otherwise it bans me). I wanted to collect data from all sensors through a dummy object and send it with **one request** (as a string). At the same time, I have another controller that sends data asynchronously from each sensor using the MQTT protocol every minute (my HA server).

tonhuisman, 2023-08-21:
Ah, that can be solved in at least 2 different ways, that have a lot in common, only the way they are triggered differs:
The common part:
```
On SendDataExternal Do // Send current data to server
   SendToHTTP,'http://<username>:<password>@your.external.server:port/path/to/send?[env#temperature],[env#env],[env#pressure],[uv#uvIndex],[dust#pm1],[dust#pm10],[dust#pm25],[rain#rain]'
Endon
```
NB: I've used single-quotes around the argument to SendToHTTP, so you can add arguments/parameters using double-quotes if needed.
NB2: You'll have to match the `/path/to/send` and arguments to the external requirements, and also for username, password, your.external.server and port.

Now for scheduling it:
Scheduling option 1:
- Configure a [Cron task](https://espeasy.readthedocs.io/en/latest/Plugin/P081.html#p081-page) (I've used Name: CronTask) using the 6 minute schedule (requires the Cron plugin to be included, in `COLLECTION` and `CLIMATE` builds)
- Cron expression: `* */6 * * * * *`
- Respond to the `On Cron#<taskname> Do` event:

```
On Cron#CronTask Do
  AsyncEvent,SendDataExternal
Endon
```
Scheduling option 2:
Use a counter and the regular `On Clock#Time Do` event:
```
On Clock#Time=All,**:** Do // Every minute
  Let,1,"%sysmin% % 6" // Mod 6
  If %v1%=0 // Every 6th minute
    AsyncEvent,SendDataExternal
  Endif
Endon
```

NB: I've used 6 minute schedules to not tempt the external service with too close successive 'hits' 😺

almirus, 2023-08-21:
@tonhuisman thanks, I'll try
But the external server requires additional correct HTTP headers - like these. That’s why I wanted to collect data in a DUMMY sensor and send data through the standard controller.
```
Content-Type: application/x-www-form-urlencoded
Content-Length: GET_SIZE
```

tonhuisman, 2023-08-21:
We also support `PostToHTTP` and `PutToHTTP` [commands](https://espeasy.readthedocs.io/en/latest/Reference/Command.html) (no direct link, please search the page), that support adding headers. `SendToHTTP` for HTTP `GET` doesn't support headers.

tonhuisman, 2023-08-21:
> That’s why I wanted to collect data in a DUMMY sensor and send data through the standard controller.

One of the issues there is that only 4 values can be stored in a single Dummy device, and the Controller will do 1 transmission for each Dummy device.

------------------

There are more ways to start a schedule, you can also use a Dummy device, and set that to the desired Interval, like 360 seconds, then act on the event:
Scheduler option 3:
```
On Dummy#Dummy do // Expected every 6 minutes
  AsyncEvent,SendDataExternal
Endon
```

almirus, 2023-08-21:
> One of the issues there is that only 4 values can be stored in a single Dummy device

Yes, it’s a serious limitation for me too. At first, I also wanted to transmit through numerical values, but I have 4 physical sensors with two to four values and they don’t fit into the DUMMY object. Is this a software limitation?

[image]

tonhuisman, 2023-08-21:
> I have 4 physical sensors with two to four values and they don’t fit into the DUMMY object. Is this a software limitation?

That has several technical reasons, based on trade-offs vs. available resources, one of them being that ESPEasy acts event-driven, so it can't know if other tasks are ready to send their data combined, and another that the configuration storage is a) limited in size, b) divided in fixed block, c) needs to be backward compatible, to avoid seldom used code to hang around for a long time 'just in case', d) what upper limit would be enough, as a dynamic count would require settings to be stored differently, f.e. in json, but that adds even more code, needs much more storage space and will wear out the flash storage faster because of more often and larger write activities, (such text-based file has to be completely rewritten on every change...).

TL;DR; It's complicated 😃 

There are _ideas_ to handle sending to Controllers differently, but that's a long-term plan only, no actual work in that area is being done yet.

almirus, 2023-08-21:
@tonhuisman Am I doing something wrong? I’m trying to send a POST request. First of all, only one line gets into the log (after save Rule, not every minute), secondly the log is strange, thirdly it seems like the data is not being sent at all.
```
On Clock#Time=All,**:*1 do 
  PostToHTTP,http://192.168.0.145:8081/environment,'Content-Type: application/x-www-form-urlencoded',`ID=%mac%&temperature=[env#temperature]&humidity=[env#humidity]&pressure=[env#pressure]&uvIndex=[uv#uvIndex]&lux=[uv#lux]&pm1=[dust#pm1]&pm25=[dust#pm25]&pm10=[dust#pm10]&rain=[rain#rain]&owner=almirus`
endon
```
```
3472321: EVENT: Clock#Time=Mon,22:41
3472328: ACT : PostToHTTP,http:
```

TD-er, 2023-08-21:
```
On Clock#Time=All,**:*1 do
```
That one would only be triggered on times where the minute ends on 1.
Thus **:01, **:11, **:21, **:31, **:41, **:51

TD-er, 2023-08-21:
Maybe you should first try to see how that string is being formatted when you try to send it to a log.

Also you are using different quotes on the last argument.
Not sure if they match, so maybe best to change them to the same type of single quotes, just to be sure?
Like this:
```
On Clock#Time=All,**:*1 do 
//  PostToHTTP,http://192.168.0.145:8081/environment,'Content-Type: application/x-www-form-urlencoded','ID=%mac%&temperature=[env#temperature]&humidity=[env#humidity]&pressure=[env#pressure]&uvIndex=[uv#uvIndex]&lux=[uv#lux]&pm1=[dust#pm1]&pm25=[dust#pm25]&pm10=[dust#pm10]&rain=[rain#rain]&owner=almirus'
  logentry,"  PostToHTTP,http://192.168.0.145:8081/environment,'Content-Type: application/x-www-form-urlencoded','ID=%mac%&temperature=[env#temperature]&humidity=[env#humidity]&pressure=[env#pressure]&uvIndex=[uv#uvIndex]&lux=[uv#lux]&pm1=[dust#pm1]&pm25=[dust#pm25]&pm10=[dust#pm10]&rain=[rain#rain]&owner=almirus'"
endon
```

almirus, 2023-08-21:
> That one would only be triggered on times where the minute ends on 1.
> Thus **:01, **:11, **:21, **:31, **:41, **:51

It didn’t help, the timer doesn’t work every minute.
```
On Clock#Time=All,**:01 do 
  logentry,"hi"
  PostToHTTP,http://192.168.0.145:8081/environment,'Content-Type: application/x-www-form-urlencoded',`temperature=123`
endon
```

```
4656753: Upload: START, filename: rules2.txt
4656848: Upload: WRITE, Bytes: 168
4656852: Upload: END, Size: 168
4672321: EVENT: Clock#Time=Mon,23:01
4672420: ACT : logentry,'hi'
4672421: hi <---------------------------------------------------
4672423: ACT : PostToHTTP,http: <--------------------- POST rule
4681484: EVENT: uv#uv=0
4681493: LTR390: data read. Mode: 1, UV: 0, UVindex: 0.00, ALS: 33, Lux: 1.65
4681807: EVENT: uv#uvIndex=0.00
4681823: ADC : Analog value: 1024 = 1024
4681865: EVENT: uv#ambient=33
4682039: EVENT: uv#lux=1.65
4682043: EVENT: http#217.25.95.24=200 <- it's my controller 
4682053: EVENT: rain#rain=0
4682056: EVENT: http#217.25.95.24=200
4682966: WD : Uptime 78 ConnectFailures 0 FreeMem 18624 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
4709383: EVENT: env#temperature=17.40
4709555: EVENT: env#humidity=58.89
4709570: EVENT: env#pressure=760.34
4709573: EVENT: http#217.25.95.24=200
4712965: WD : Uptime 79 ConnectFailures 0 FreeMem 18736 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
4732321: EVENT: Clock#Time=Mon,23:02
<------------------------ no timer event and log message
4741485: EVENT: uv#uv=0
4741495: LTR390: data read. Mode: 0, UV: 0, UVindex: 0.00, ALS: 33, Lux: 1.65
4741673: EVENT: uv#uvIndex=0.00
4741682: EVENT: uv#ambient=33
4741685: EVENT: uv#lux=1.65
4741688: EVENT: http#217.25.95.24=200
4741778: ADC : Analog value: 1024 = 1024
4741822: EVENT: rain#rain=0
4741996: EVENT: http#217.25.95.24=200
4742965: WD : Uptime 79 ConnectFailures 0 FreeMem 18512 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
4770376: EVENT: env#temperature=17.38
4770586: EVENT: env#humidity=58.95
4770597: EVENT: env#pressure=760.33
4770602: EVENT: http#217.25.95.24=200
4772965: WD : Uptime 80 ConnectFailures 0 FreeMem 18904 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
4792321: EVENT: Clock#Time=Mon,23:03
<------------------------ no timer event and log message
4801702: EVENT: Rules#Timer=1,16
4801708: ACT : timerSet,2,30
4801712: ACT : pmsx003,wake
4801713: PMSx003: Wake sensor
4801717: EVENT: uv#uv=0
4801896: EVENT: uv#uvIndex=0.00
4801902: EVENT: uv#ambient=32
4801913: ADC : Analog value: 1024 = 1024
4801956: EVENT: uv#lux=1.60
4801967: EVENT: http#217.25.95.24=200
4801971: EVENT: rain#rain=0
4802173: EVENT: http#217.25.95.24=200
4802180: LTR390: data read. Mode: 1, UV: 0, UVindex: 0.00, ALS: 32, Lux: 1.60
4802965: WD : Uptime 80 ConnectFailures 0 FreeMem 11296 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```

almirus, 2023-08-21:
> Also you are using different quotes on the last argument

I used quotes as written in [the documentation](https://espeasy.readthedocs.io/en/latest/Reference/Command.html).

TD-er, 2023-08-21:
To run every minute you need this:
```
On Clock#Time=All,**:** do 
```

tonhuisman, 2023-08-21:
Do you get these logs from the web log viewer?

tonhuisman, 2023-08-21:
Are any logs available from the receiving server? Does something arrive there?

TD-er, 2023-08-21:
The rules parser tries to match parts of the string in the event.

So if there is a specific digit on that clock#time event, then the digit must be there on that position.
N.B. this only applies to time matching as done in the clock#time event.

almirus, 2023-08-21:
> Are any logs available from the receiving server? Does something arrive there?
My Rule now:
```
On Clock#Time=All,**:** do 
  logentry,"hi"
  PostToHTTP,http://192.168.0.145:8081/environment,'Content-Type: application/x-www-form-urlencoded',`test`
endon
```

There are no requests to my test server and a strange log with truncated ‘http’
```
5752322: EVENT: Clock#Time=Mon,23:19
5752328: ACT : logentry,'hi'
5752330: hi
5752332: ACT : PostToHTTP,http:
```
http server log:
```
2023-08-21 23:14:47.795  INFO 15036 --- [           main] c.a.h.HomeMeteoStationApplication        : Started HomeMeteoStationApplication in 4.922 seconds (JVM running for 5.793)
2023-08-21 23:14:47.933  INFO 15036 --- [1)-192.168.56.1] o.a.c.c.C.[Tomcat].[localhost].[/]       : Initializing Spring DispatcherServlet 'dispatcherServlet'
2023-08-21 23:14:47.933  INFO 15036 --- [1)-192.168.56.1] o.s.web.servlet.DispatcherServlet        : Initializing Servlet 'dispatcherServlet'
2023-08-21 23:14:47.935  INFO 15036 --- [1)-192.168.56.1] o.s.web.servlet.DispatcherServlet        : Completed initialization in 2 ms
2023-08-21 23:16:28.771 DEBUG 15036 --- [nio-8081-exec-1] o.s.w.f.CommonsRequestLoggingFilter      : Before request [GET /environment]
2023-08-21 23:16:29.031 DEBUG 15036 --- [nio-8081-exec-1] o.s.w.f.CommonsRequestLoggingFilter      : REQUEST DATA: GET /environment]
2023-08-21 23:16:29.097 DEBUG 15036 --- [nio-8081-exec-2] o.s.w.f.CommonsRequestLoggingFilter      : Before request [GET /favicon.ico]
2023-08-21 23:16:29.107 DEBUG 15036 --- [nio-8081-exec-2] o.s.w.f.CommonsRequestLoggingFilter      : REQUEST DATA: GET /favicon.ico]
```

almirus, 2023-08-21:
@TD-er Now the logic is clear, thank you very much. I haven’t seen the format of writing the time template in the documentation.

TD-er, 2023-08-21:
One suggestion Ton just made to me via chat...
Maybe the `//` in `http://` is being considered as a comment.
So can you place the URL part in quotes too?

TD-er, 2023-08-21:
Something like this:
```
PostToHTTP,'http://192.168.0.145:8081/environment', .....
```
Just shorted the line, but you get the idea....

almirus, 2023-08-21:
> Maybe the `//` in `http://` is being considered as a comment.
> So can you place the URL part in quotes too?

```
On Clock#Time=All,**:** do 
  logentry,"hi"
  PostToHTTP,'http://192.168.0.145:8081/environment','Content-Type: application/x-www-form-urlencoded',`test`
endon
```

```
6052321: EVENT: Clock#Time=Mon,23:24
6052327: ACT : logentry,'hi'
6052329: hi
6052331: ACT : PostToHTTP,'http:
6052332: PostToHTTP: Invalid port argument: 0 will use: 80
6052338: EVENT: http#=-1
```

almirus, 2023-08-21:
@tonhuisman @TD-er Success!
I've found this https://www.letscontrolit.com/forum/viewtopic.php?p=62769#p62769

```
On Clock#Time=All,**:** do 
  logentry,"hi"
  PostToHTTP,192.168.0.145,8081,environment,'Content-Type: application/x-www-form-urlencoded',`ID=%mac%&temperature=[env#temperature]&humidity=[env#humidity]&pressure=[env#pressure]&uvIndex=[uv#uvIndex]&lux=[uv#lux]&pm1=[dust#pm1]&pm25=[dust#pm25]&pm10=[dust#pm10]&rain=[rain#rain]&owner=almirus`
endon
```

```
2023-08-21 23:37:58.990 DEBUG 15036 --- [nio-8081-exec-8] o.s.w.f.CommonsRequestLoggingFilter      : Before request [POST /environment]
2023-08-21 23:37:58.993 DEBUG 15036 --- [nio-8081-exec-8] o.s.w.f.CommonsRequestLoggingFilter      : REQUEST DATA: POST /environment, payload=ID=BC%3ADD%3AC2%3A14%3ABA%3A07&temperature=17.22&humidity=59.62&pressure=760.28&uvIndex=0.00&lux=1.60&pm1=6.8&pm25=7.1&pm10=7.1&rain=0&owner=almirus]
```

tonhuisman, 2023-08-21:
That should work indeed, with the disadvantage you can't easily include a username:password to authenticate with the server.
But this has lead us to a bug that I'm now trying to squash out 👍

closed 2023-09-06 by TD-er