letscontrolit/ESPEasy #4783: [FR] Secure MQTT for ESP32 Devices

chr2002qw, 2023-09-02:
Will ESP Easy get MQTT over SSL like Tasmota has.

On ESP32 Devices Secure MQTT works fine at Tasmota. With ESP8266 its possible, too but very very unstable.

How about adding SSL to ESPeasy ?

TD-er, 2023-09-02:
I have been working on it: https://github.com/letscontrolit/ESPEasy/pull/3788
The reason it has not yet been saved is because of the extreme strange hoops I had to jump through for MbedTLS to make sure it doesn't leak memory.

You can try the last test build made for this PR: https://github.com/letscontrolit/ESPEasy/actions/runs/5968797472

chr2002qw, 2023-09-07:
How can i download thia build for the common ESP32 ? Theres no Link to a File.

TD-er, 2023-09-07:
I literally put a link for you to download.
See this screenshot:
[image]

chr2002qw, 2023-09-07:
Oh yes, found it. My Browser had Issues. Used another and it worked. Just flashed this. But i cant see any SSL or TLS Stuff in controller Settings.

TD-er, 2023-09-07:
You need to test either the "max" build (for 16M flash units) or the "custom" build (which doesn't have all plugins included).
But since it is a test for the MQTT controller, you can simply test with "dummy" or "sysinfo" plugins to test the controller.

chr2002qw, 2023-09-07:
Hmm, on Max. Wifi is broken

35334 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 217528 WiFiStatus WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
65342 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 217528 WiFiStatus WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
85929 : Info   : WIFI : Set WiFi to STA
85930 : Info   : WIFI : Cannot set mode!!!!!
86031 : Info   : WIFI : Cannot set mode!!!!!
86132 : Info   : WIFI : mode not yet set
86232 : Info   : WIFI : mode not yet set
86434 : Info   : WiFi : WiFiConnected(), start AP
86435 : Info   : WiFi : Start network scan all channels
86436 : Error  : WiFi : Scan failed
86539 : Info   : WIFI : Set WiFi to AP
86540 : Info   : WIFI : Cannot set mode!!!!!
86641 : Info   : WIFI : Cannot set mode!!!!!
86742 : Info   : WIFI : mode not yet set
86842 : Info   : WIFI : mode not yet set
87045 : Error  : WIFI : [AP] softAPConfig failed!
87046 : Error  : WIFI : Error while starting AP Mode with SSID: ESPEasy- IP: 192.168.4.1
87048 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
95329 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 217528 WiFiStatus WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED

TD-er, 2023-09-07:
Hmm are you sure about the power supply of that unit?
Can you just try to set the WiFi credentials via the serial port and save them, then power cycle the ESP?

```
wifissid,YourWiFiSSID
wifikey,YourSecretWiFiPassword
save
wifidisconnect
```
WiFiDisconnect will then force a WiFi restart
If that's not working, you can power cycle the ESP.

chr2002qw, 2023-09-07:
Doesnt work.

I´ll drop on this. I need a stable solution for MQTT TLS and will wait until its merged,

chr2002qw, 2023-09-07:
Seems to work with "custom" but whyyyyyyyyyyyyyyyyyyyy MQTT-Import is removed :-( :-( 
I Only need Import and Framed Display.

TD-er, 2023-09-08:
OK, but that's something we can change....
Which build do you need?

chr2002qw, 2023-09-08:
I flashed this "ESP_Easy_mega_20230824_custom_ESP32_4M316k.factory.bin" for testing. It seems, MQTT-TLS gets connected 👍

Is it easy to add MQTT-Import to that build ?

Thank you 👍

TD-er, 2023-09-08:
I will have a look later this evening.
It sure isn't hard to do (maybe Ton will do it as he always has some window open with the ESPEasy code :) ), but I have to get behind a PC that won't take forever to make a build.

TD-er, 2023-09-08:
Just thought about a quick work-around :)

I just edited the file describing the Custom builds on ESP32 and triggered a GH Actions build:
https://github.com/letscontrolit/ESPEasy/actions/runs/6123368939

So it will be ready in about 40 - 45 minutes.

chr2002qw, 2023-09-22:
Had to do a break from this stuff due to massive work :-(
I tried it, and this is working very good with SSL and my external MQTT broker.

Today i tried to compile a custom myself and checked out the Code with this PR. But i get following error in PIO if i try to compile a custom build : 

"error initializations for multiple members of 'ProtocolStruct::<unamed union>"

Can i compile a build myself or is that not as esay with this PRed Version ?

TD-er, 2023-09-22:
Depending on what time you pulled the code, you may have pulled the code that still had a merge conflict.
I solved that later this morning.

chr2002qw, 2023-09-23:
I pulled again and the error gone 👍 But iam to dumb to compile it. I switched to custom_env and renamed the custom-sample.h and selected the Plugins i need.
Then i selected the controller "Homeasisstant MQTT" The bin compiles w/o errors and was uploaded to the ESP32. 
But no Wifi Hotspot is showing up. No Idea what iam doing wrong :-(

TD-er, 2023-09-23:
Maybe it is easier to remove your Custom.h file and just edit the file `tools/pio/pre_custom_esp32.py`
As you can see in that file, there is a line like this:
```
"-DFEATURE_MQTT_TLS=1",
```
This define will enable the feature MQTT TLS.

Or just add to your Custom.h:
```c++
#define FEATURE_MQTT_TLS   1
```

chr2002qw, 2023-09-28:
Thanks, this worked 👍 Everything is running very good with that PR on my ESP32. Not tested with ESP8266.

closed 2024-10-02 by TD-er