letscontrolit/ESPEasy #4784: Not an Issue :)

Sasch600xt, 2023-09-04:
Hey my beloved developer and mods!

just want to let you know, there is a device with a lot to to (heating control, alarm central, light control, power control and so on) which is online since mid of 2019 ^^
And this in an industrial enviroment.

Thats what i call "stable" ^^

Have a great day all :)
[image]

TD-er, 2023-09-04:
So, that's quite a long uptime :)
Not sure if I ever had uninterrupted power at my house for so long.
Well at least I could not have had such uptime in this house as it wasn't built yet around the time you booted yours.

Highest uptime I have on my nodes is about 150 days.

Sasch600xt, 2023-09-04:
in my private house, i have uptimes about 1.5 Years in each ESP. And only because there is always a power interruption after 1- 2 Years.
I am so happy with your work!
Thank you so much for your great work!
Have a great day :)