letscontrolit/ESPEasy #4791: [FR] Client IP filtering using CIDR notation

iz8mbw, 2023-09-06:
Hello.
Currently the "Client IP filtering" under "Main Settings" page accepts only "Allow all", ""Allow Local Subnet" and "Allow IP range" and so it's not possible to specify different subnets.
For example it is not possible to allows "192.168.1.0/24" and "10.10.0.0/24" in the same time. And, as you can understand, create a so big range doesn't make sense.

As Future Request can be the final solution to use the CIDR notation for Client IP filtering (providing a list of CIDR to allow).

Thanks.

TD-er, 2023-09-06:
Hmm that's a bit tricky as adding an arbitrary number of ranges does require significant changes in the settings structure.
So it may sound like a simple request, but it really isn't.

iz8mbw, 2023-09-07:
OK.
So, as "workaround", can be implemented to allow multiple IP range?
Example:
From 192.168.1.0 To 192.168.1.254 AND From 10.10.0.0 To 10.10.0.254 ?

TD-er, 2023-09-07:
Right now the settings struct for this only has 4 bytes for the lower range and 4 bytes for the upper range.