letscontrolit/ESPEasy #4794: [FR] Add Senserion SEN5x (IKEA VINDSTYRKA)

TD-er, 2023-09-11:
The latest [Ikea VINDSTYRKA](https://www.ikea.com/nl/nl/p/vindstyrka-luchtkwaliteitsensor-smart-00498231/) air quality sensor uses the [Senserion SEN5x](https://sensirion.com/search?q=sen5x).
There are 3 types of this sensor:

- [SEN50](https://sensirion.com/products/catalog/SEN50) - Only PM
- [SEN54](https://sensirion.com/products/catalog/SEN54) - PM, RH/T, VOC
- [SEN55](https://sensirion.com/products/catalog/SEN55) - PM, RH/T, VOC, NOx

iz8mbw, 2023-09-11:
@TD-er please replace "Senserion" with "Sensirion".

tonhuisman, 2023-09-11:
This guy Oleksii already has a sort of [tear-down](https://github.com/oleksiikutuzov/IKEA-VINDSTYRKA) for the device (more like an unboxing 😉), that confirms the sensor is indeed a SEN54.
There also is an official Sensirion Arduino library for the SEN5x (using I2C). Not sure if we can 'eaves-drop' on that, like is done with the [Vindriktning plugin](https://espeasy.readthedocs.io/en/latest/Plugin/P144.html#p144-page) (that's using a more 'regular' serial interface). And maybe just implementing this in a more ESPEasy fashion, without introducing _another_ 'standard' way of communication with I2C devices, is a better idea.

TD-er, 2023-09-11:
Yep, was thinking about the same approach.
Problem is that we may need to 'sniff' to see if there is activity on the I2C bus.

TD-er, 2023-09-11:
Ah apparently someone else came up with the same idea: https://github.com/oleksiikutuzov/IKEA-VINDSTYRKA/issues/1#issuecomment-1455082822

TD-er, 2024-02-21:
I guess the Ikea firmware does try to read only on regular intervals, so maybe we can detect this and try to match its read interval, not to disturb those readings?

Another way can be to add a second I2C bus and try to emulate the device on this 2nd bus and thus act as some kind of man-in-the-middle and echo'ing the values while keeping the read results.

N.B. a 2nd bus is probably only limited to ESP32-like devices.

TD-er, 2024-02-21:
Ah this emulation was also mentioned on the link you provided.
Meaning this could be turned into 2 plugins:
- Reading the sensor
- Adding a specific display where we emulate the sensors to show arbitrary values like on any display.

tonhuisman, 2024-02-21:
The SEN55 sensor is planned to be included soon, using the I2C bus.

TD-er, 2024-02-21:
If someone wants to modify such a unit, then quite often new hardware has to be bought.
So why buy any ESP8266 right now if the ESP32-C3 (as example) costs less money, is smaller and has much less power and antenna related issues compared to those "Wemos D1 mini" ESP8266 form factor boards you get when ordering now from Ali et. al. ?

andibaciu, 2024-02-24:
i make some progress for this ESPeasy plugin (for ESP8266)
All description and files could be find here: https://github.com/andibaciu/ESPeasy-plugin-for-IKEA-VINDSTYRKA
Best regards!

TD-er, 2024-02-24:
Why do you use P253?
You can just request a plugin ID: https://github.com/letscontrolit/ESPEasy/issues/3839
And why not directly make a pull request?

andibaciu, 2024-02-24:
Sorry, my mistake, i dont know about this request a plugin ID: https://github.com/letscontrolit/ESPEasy/issues/3839 (it's my first contribution for this project) and i post about this plugin only here, after the owner give approval for this plugin and allocate a number i'll make all necessary changes.

PR #4991 (2024-04-15): Plugin P167 IKEA Vindstyrka
In the file attached you have all description you need.
[ESPeasy plugin for IKEA VINDSTYRKA.pdf](https://github.com/letscontrolit/ESPEasy/files/14418911/ESPeasy.plugin.for.IKEA.VINDSTYRKA.pdf)

Resolves #4794
PR #5038 (2024-05-29): [P167] Enable use of SEN5x sensor, other improvements
Resolves #4794 phase 2: Stand-alone SEN5x support

Features:
- Enable use of Sensirion SEN5x standalone sensor
- Added plugin to `Climate` and `MAX` builds
- Formatted source using Uncrustify
- Clean up code
  - Use existing dew point calculation
  - Remove unused variables and code
  - Fix issue with status bits
  - String handling improved
  - Enable multi-instance use (requires I2C multiplexer because of fixed I2C Address)
  - Use enum classes where applicable
- Add support for `sen5x,startclean` command to clean the fan
- Add support fo [... cut]
closed 2024-04-15 by TD-er