letscontrolit/ESPEasy #4797: [BUG] RTTL not implimented in CLIMATE build ESP_Easy_mega_20230822_climate_ESP8266_4M1M

DeNB3rt, 2023-09-12:
**Describe the bug**
RTTL seems not implimented in CLIMATE build ESP_Easy_mega_20230822_climate_ESP8266_4M1M
but as stated in the documentation, it should be?
[https://espeasy.readthedocs.io/en/latest/Plugin/P000_Buzzer_RTTTL.html](url)

**To Reproduce**
Steps to reproduce the behavior:
I need plugin P068 for my SHT3X so I used the CollectionA
[https://espeasy.readthedocs.io/en/latest/Plugin/P068.html](url)
but RTTL is not supported, so I flashed the CLIMATE version on the same device
[https://espeasy.readthedocs.io/en/latest/Plugin/P000_Buzzer_RTTTL.html](url)

**Expected behavior**
As stated in the documentation, the CLIMATE build should be support both P068 and RTTL plugin.

**Used platform (please complete the following information):**
 - ESP type:  ESP8266
 - Build version: ESP_Easy_mega_20230822_climate_ESP8266_4M1M
 - Build set climate

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M
 - Brand/Model: NodeMCU
 - Browser + OS Windows 10/Chrome

**Additional context**
none

tonhuisman, 2023-09-12:
@DeNB3rt I've added a fix for that, and the build is running. You can download for testing, once finished, from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6164767429)
Please report your findings after testing.

DeNB3rt, 2023-09-13:
Thanks! Just flashed it and it seems to work all again! Thanks, big up!

PR #4798 (2023-09-17): [Build] Restore RTTTL feature in ESP8266 Climate build
Resolves #4797 

Bugfix:
- Restore the RTTTL feature in the ESP8266 `Climate` build, that went missing when we applied `LIMIT_BUILD_SIZE` to it.

- [x] Testing [Confirmed](https://github.com/letscontrolit/ESPEasy/issues/4797#issuecomment-1717645228)
closed 2023-09-13 by DeNB3rt