letscontrolit/ESPEasy #4799: [FR] Plug-In/Native UART Support for HiLink HLK-LD2410 FMCW Motion & Presence Sensor

Wookbert, 2023-09-13:
Just in case you didn't come across this yet: The tiny and dirt cheap [HiLink HLK-LD2410](https://www.hlktech.net/index.php?id=988) is a „new“ Human Presence & Motion sensor based on FMCW (Frequency Modulated Continuous Wave) everyone is integrating and talking about now. The Bluetooth version is just around EUR 4 shipped on [AliExpress](https://www.aliexpress.com/w/wholesale-ld%2525252d2410b.html?catId=0&initiative_id=SB_20230913115659&SearchText=ld-2410b&spm=a2g0o.home.1000002.0).

[image]
[image]

It can detect Motion and Presence (!) of up to 6 meters without the need for line of sight (so installation behind drywall, glass, inside furniture, provides the distance to the detected being and with easily adjustable detection sensitivity and range. The HLK-LD2410-B is the version with Bluetooth connectivity, which can connect to the HLKRadarTool Smartphone App (available for [iOS](https://apps.apple.com/de/app/hlkradartool/id1638651152) and [Android](https://www.pgyer.com/Lq8p)) and make set up even easier.

[image]
[image]
[image]

While the sensor can be easily attached to an ESPEasy host using the Low/High level pin, the sensor also comes with a UART connection, which can be used for displaying detailed information such as motion, presence, distance to object etc. and be used for making adjustment. This has already been implemented in other ESP firmware platforms such as Tasmota and ESPHome.

Full documentation (incl. serial protocol) of the LD2410 Sensor can be found [here](https://drive.google.com/drive/folders/1lCQv3mfHJ3XKXzweeHPFnJ_8_D_EWEKk).

Note that there are also a couple of [other FCMW sensors from HiLink](https://www.hlktech.net/index.php?id=product&cate=RadarModule), such as the LD2410C, LD2420 etc., which are, as I understand almost identical to the LD-2410B, but come in a different form factor and lack Bluetooth.

Here’s an early YouTube review of the non-Bluetooth version LD2410 by Andreas Spiess:
[image]_Click image to watch on YouTube_

tonhuisman, 2023-09-13:
Duplicate for #4673

closed 2023-09-14 by TD-er

Wookbert, 2023-09-14:
@TD-er
Completed? So does that mean that the LD2410 plug-in is actually already around, but just hasn’t been documented in [ReadTheDocs](https://espeasy.readthedocs.io/en/latest/index.html) yet?

TD-er, 2023-09-14:
Nope, I closed this as it is a duplicate :)