letscontrolit/ESPEasy #4814: [BUG] I2C Scan provides redundant devices in list

iz8mbw, 2023-09-22:
Using the latest build (mega_bb22164) and the I2C devices Scan provide redundant devices.
Take these examples:
[image]

the device (sensor) "SHT35" is listed two times.

[image]

the device (sensor) "LTR390" is listed two times.

tonhuisman, 2023-09-22:
No, that's not a bug, that's a feature 😄.
The `(Device)` prefix indicates that the plugin/device is _actually_ included in the build, and the short name an indication that it is supported/recognized by ESPEasy as a valid device.
As there is no direct relation between those short names and long plugin descriptions, as multiple devices can have the same I2C address, but a plugin usually only supports 1 type of device, they can't be separated or ignored.

NB: Some devices are not listed with their long name, as they support the full range of I2C addresses, and would always show up in this list for every address. They most likely will show their address only.

iz8mbw, 2023-09-22:
OK, clear.
Going to close it.

Thanks!

closed 2023-09-22 by iz8mbw