letscontrolit/ESPEasy #4821: [BUG]https://td-er.nl/ESPEasy/

dosipod, 2023-09-24:
[image]

TD-er, 2023-09-24:
Can you email me your IP-address?
I have put in a large IP-blocklist because of incredible amounts of hack-attempts.
Just to put it into context, I block about 3 TB of requests a month.

TD-er, 2023-09-24:
Based on the logs, I think I found your IP.
Can you try again?

dosipod, 2023-09-24:
Sure it is fixed faster then expected and i can install it just fine .Thank you for the quick action 
Since you are fixing things m you might want toalso  look at the  pre-release test build button  listed at https://td-er.nl/ESPEasy/latest/latest/ as that will give 404 Not Found error

TD-er, 2023-09-24:
Ah that's a link I add in the Python script to generate the HTML page.

dosipod, 2023-09-24:
All okay now , impressive response time

closed 2023-09-24 by dosipod