letscontrolit/ESPEasy #4822: [BUG] SI7021/HTU21D false value

padbol2000, 2023-09-24:
**Describe the bug**

After an update from 20211105 to 20230822 (without changing the wiring) the sht21 probe returns incorrect values ​​(-46°C instead of 23°C)

**To Reproduce**
Steps to reproduce the behavior:

put a sht21 sensor

**Used platform (please complete the following information):**
 - ESP type: ESP8266
 - Build version: mega-20230822]
 - Build set normal

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M
 - Brand/Model: Wemos D1

**Additional context**
tested with an another sensor, same result. ref sensor : TPH Board seeedstudio SKU 114990245

tonhuisman, 2023-09-24:
Does the value change when the temperature changes, or is it stuck at -46 ?

padbol2000, 2023-09-24:
Stuck at -46, 
with the firmware 20221105 stuck at +128.9
and works like a charm with firmware 20220809 !
hope it will help

tonhuisman, 2023-09-24:
Ah, I've done some digging in the code; with the 20221105 release there has been a major overhaul of this P014 plugin, and one of the features is a check on the Chip ID during initialization, as support for the SI70xx series sensors was added, but support for the SHT21 has been removed (😱), because that chip is not (fully) compatible with the command structure of the HTU21D/SI70xx chips. (This should have been documented somewhere, but unfortunately it isn't 😞)

The SHT2x series sensors are (now) supported by a separate plugin, [P122 - Environment - SHT2x](https://espeasy.readthedocs.io/en/latest/Plugin/P122.html) that's included in `Collection F` and `Climate` builds.

Depending on the other plugins configured on your device you should be able to upgrade to Collection F or Climate via the Update Firmware button, and reconfigure the task using the SHT2x plugin (or use a separate task and keep the current settings for older releases). When keeping the old task, it must be disabled for the new task to be able to work properly.
Only known 'issue' when upgrading to a Collection or the Climate build could be that P037 MQTT Import doesn't support all features it has in the Normal build you are currently using, and there is no DEBUG level logging included in these builds (to be able to include more plugins).

padbol2000, 2023-09-24:
Ok thanks, works on the last Collection F with plugin 122.

it would be useful to add in the log that sht21 disappeared from plugin 14 to go into plugin 122

closed 2023-09-24 by padbol2000