letscontrolit/ESPEasy #4827: [FR] Add support for PN5180 - RFID

iz8mbw, 2023-09-26:
Hi!
Is it possible to add NXP Semiconductors PN5180 RFID?

[image]

Docs Libraries:

https://github.com/ATrappmann/PN5180-Library
https://github.com/taymuur/ESP32-PN5180-PN532
https://github.com/playfultechnology/arduino-rfid-PN5180
https://github.com/dotnet/iot/tree/main/src/devices/Pn5180
https://docs.nanoframework.net/devicesdetails/Pn5180/README.html

Many thanks!

tonhuisman, 2023-09-26:
What's the USP (Unique selling point) for adding support for this board? We already have support for several, differently interfaced, RFID boards...

iz8mbw, 2023-09-26:
The PN5180 can read the card/tag up to 15 cm.
RC522 and PN532 can reach some cm only.

TD-er, 2023-09-26:
The second library you linked is apparently also supporting the PN532, which we do support.
So maybe it is just a matter of adding the right I2C addresses to the selection?

iz8mbw, 2023-09-26:
PN5180 uses different library than PN532.

tonhuisman, 2023-09-26:
PN532 does support SPI, I2C and serial (AFAIR), but we support it in I2C mode only. PN5180 only has support for SPI interface.

TD-er, 2023-09-26:
As far as I can remember is SPI support for the PN532 far from stable.
Thus when this PN5180 will be added, it must then be a new plugin as it can't use the same bus as the PN532.

Have to see if it is already available for a reasonable price at the usual channels.

iz8mbw, 2023-09-26:
> Have to see if it is already available for a reasonable price at the usual channels.
>
It costs about 5 Euro on AliExpress.