letscontrolit/ESPEasy #4842: [%lcltime% in Display - OLED SSD1306  fail updating]

IW2FAG-11, 2023-10-07:
[%lcltime% in Display - OLED SSD1306  fail updating] 

ESP8266 platform with ESP_Easy_mega_20230930_normal_ESP8266_4M1M.bin 

Using the latest mega-20230930 tag, it becomes inoperative to update the date retrieved from the [%lcltime%] display lines.
With the tag ESP_Easy_mega_20230822_normal_ESP8266_4M1M and identical config and rules it worked perfectly.
I use ADCS1115 with readings of 3ADC every 6/1 second plus rules.
With the mega-20230930 La release update I was forced to go into the ADC configurations and re-commit in order to see the samplings working again, this was probably because of the changes I saw in the management.
Going back to the mega_20230822 version via OTA downgrade I got stuck and had to delete the FLASH with in the empty 4Mbin. Otherwise I could no longer downgrade even by serial. It probably failed OTA downgrade due to my too fast restart.

[image]
[image]
[rules1.txt](https://github.com/letscontrolit/ESPEasy/files/12837326/rules1.txt)
[rules2.txt](https://github.com/letscontrolit/ESPEasy/files/12837327/rules2.txt)

tonhuisman, 2023-10-07:
Ah, that probably fell off the wagon somewhere recently. I'll fix that.

tonhuisman, 2023-10-07:
Hmm, seems to be working here (though on latest `mega` sources, there hasn't been any changes in this area).
I'll connect an OLed display to see if something is not working as intended there.

Is the `%lcltime%` result empty, or doesn't it get updated?

IW2FAG-11, 2023-10-07:
Start with 1970 xxxxx and stay at that date.

tonhuisman, 2023-10-07:
> Start with 1970 xxxxx and stay at that date.

Looks like your ESP system time isn't being updated.
Check your NTP setting, best would be to enable the Use NTP checkbox, and leave NTP Hostname _empty_ (Tools/Advanced page)
It may take a couple of seconds/minutes after boot/reset for the time to be updated.

Alternatively you can enable the ESPEasy P2P protocol (Tools/Advanced ESPEasy p2p UDP Port, default: 8266) and have at least 1 ESPEasy instance with access to the correct time, either NTP, GPS or a connected RTC. That will be automatically broadcasted/shared via the P2P network.

IW2FAG-11, 2023-10-07:
I will try leaving the NTP url blank as you indicate. What I don't
understand is that with the exact same configurations, the August release
works but the September release does not. I will try again starting with a
total reset.
Thanks

TD-er, 2023-10-07:
If you have NTP disabled, but some other ESPEasy node in your network does have some idea of the time, then they will all sync.

IW2FAG-11, 2023-10-08:
I'd male a video of the problem... in this case on ESP8266 wemos board with
16MB Flash and energy firmware.
The problem still present
To back in prevoious 20230822 release i news to do two update.....
I'm tring to compress below 25MB anyway i can't send with email.

IW2FAG-11, 2023-10-08:
sorry for language mine t9 is on for Italian language 🥹

TD-er, 2023-10-08:
What I meant to say is that no changed configuration doesn't mean something else could have changed.

ESP-1: Has NTP checked
ESP-2: NTP not checked

What if ESP-1 was online the last time you setup ESP-2.
This then means they will sync their time via P2P.
If you reboot an ESP (and keep it powered), they will still have some idea of the system time as it is also stored in RTC memory.
So you can go on "forever" and still have some idea of time/date, even when the original time source (ESP-1) has been removed. (without NTP sync somewhere, it will eventually become quite inaccurate)

If you have several ESPEasy nodes, they can keep their time synced for quite some time, without time source, until all units have lost power at the same time (e.g. power outage)

IW2FAG-11, 2023-10-08:
If I do not change any parameters, in a device with NTP active for example on en.pool.ntp.org, equipped with firmware 20230822 and with 1-second refresh of the display, as soon as I turn on, the correct date is taken from the system variables and shown.
If I use the firmware 20231003 instead, this does not happen, basically the date in the system variables is correct, but the display does not show the correct date. It displays 1970 .... xxxx and it will never be updated.
I have the video of what I am claiming, but I cannot upload because of the large size.

You are however developing a beautiful environment for "user-friendly" management of all the connected devices, I know how much time development costs I hope I can help you improve the interaction of all the codes with each other. As soon as I notice a new release I will update my test devices and then all my Iot devices at home.
Thanks again for the great work you are doing.

Translated with www.DeepL.com/Translator (free version)

IW2FAG-11, 2023-10-08:
Sorry but 10M is toooo smalll in resulution.

https://github.com/letscontrolit/ESPEasy/assets/69007986/d4dad2f4-49cf-44a5-877c-4030384e183c

tonhuisman, 2023-10-08:
I think there is an issue with that NTP url `en.pool.ntp.org`, because when I use it I can't get the time (but get time via P2P 😉), and after I change the url to `pool.ntp.org` I _instantly_ get the time from NTP...
Can't understand why this would work with an older release, as there is no filtering on NTP Hostnames.

tonhuisman, 2023-10-08:
According to the [usage instructions](https://www.ntppool.org/en/use.html) for the NTP.org pool, you are supposed to use `pool.ntp.org` optionally prefixed with `0.`, `1.`, `2.` or `3.` to get an address that's randomly changed evry few hours.
Additionally, you can prefix the url with a country-code, like `it.`, `ch.`, `nl.` etc. but it seems the `en.` prefix doesn't work anymore or is temporarily off-line. That country-prefixed url can have the `0.` .. `3.` prefixes.
A valid NTP hostname for your situation would be `it.pool.ntp.org`. I've used that locally, and it seems to work as intended 😃

iz8mbw, 2023-10-13:
A good NTP Server in Italy is time.inrim.it
I use it without any issues.

Info: https://www.inrim.it/it/ricerca/settori-scientifici/tempo-e-frequenza/ntp-network-time-protocol

tonhuisman, 2023-10-13:
I've _never_ set the NTP server (because of the proper use of the pool), except for tests like this, and currently have only a few of my ESPEasy units get their time from NTP, where the rest of them (just over a dozen of units) get their time via the P2P network. If only to avoid a request-storm when there's a power failure and all units go online at nearly the same time 😸.
The basic issue here seems to be that the `en.pool.ntp.org` server doesn't seem to exist (I can't find it in the NTPPool.org website), so if it either wasn't set in ESPEasy, or set to an existing server name, it most likely would have worked without issues.

What I still don't understand is the claim it worked/works with older versions of ESPEasy. It may have either been handled by a global redirect at the pool.ntp.org site, or there could have been a server at that url that now no longer exists. When I now set this url als the NTP Hostname I don't get time data, and when set to `pool.ntp.org`, or leave that field empty as suggested, time is retrieved instantly and without problems.

As the problem seems to have been solved, or at least _sort of_ explained, this issue can IMHO be closed,

IW2FAG-11, 2023-10-13:
The problem is NOT NTP the problem is that display don't refresh the
visualizzati in of %lcttime% variabile, so also if system time is correct
DISPLAY show wrong data and doesn't refresh.

May be..... PROBLEM IS NON NTP.

tonhuisman, 2023-10-13:
> display don't refresh the visualizzati in of %lcttime% variabile

That's even stranger, as the display plugin you are using (P023) hasn't had changes in a long time...
What is the Interval set to? As that's the rate at which the display is updated. 0 means there is no update 😃

IW2FAG-11, 2023-10-13:
Update is set at 1s.
With last release autoupdate doesen't work.
I'll revert al mine at 20230822

IW2FAG-11, 2023-10-16:
Problem still present also on last 20231013 release.
Seems  that scheduled action is not satisfied
1sec time base on rules cycle, OLED-ssd1306 refresh still paused also if there is 1sec in refresh select menu.

TD-er, 2023-10-16:
Hmm that has been touched recently indeed, but I didn't think of looking there as other (task) interval related code is working fine.
Will have a look at it later this evening.

tonhuisman, 2023-10-16:
@IW2FAG-11 We eventually found and fixed the issue, can you test using this [GH Action run](https://github.com/letscontrolit/ESPEasy/actions/runs/6538769938), once that's finished?

IW2FAG-11, 2023-10-17:
I'll try this evening when at home.

tonhuisman, 2023-10-17:
I've done a [small performance enhancement](https://github.com/letscontrolit/ESPEasy/actions/runs/6551717618), that shouldn't interfere with the fix, so it doesn't really matter which run you test. Though testing on the actual code that's intended to be merged is preferred 😉

TD-er, 2023-10-17:
So it is working again?

IW2FAG-11, 2023-10-18:
Yesss is ok ! 👍

IW2FAG-11, 2023-10-18:
Yessss boe is already ok !!👌

PR #4851 (2023-10-18): [P023] Bugfix: Template parsing was overwriting the template
Resolves #4842 

Bugfix:
- Template parsing was overwriting the template

TODO:
- [x] Testing ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4842#issuecomment-1767154577))
closed 2023-10-18 by TD-er