letscontrolit/ESPEasy #4855: [BUG]  integer conversion of %eventvalue% not working or wrongly documented

chromoxdor, 2023-10-23:
As you can see, an inline conversion of an %eventvalue% always returns zero....

This is an example of the documentation:
[image]

This is the command:
`event, start=12.3
`
This is a rule to test:
```
On start Do
  Let,1,%eventvalue1%
  LogEntry,'[int#1], [int#%eventvalue1%]'
Endon
```

This is the log output:
```
4914777: EVENT: start=12.3
4914874: ACT : Let,1,12.3
4914880: ACT : LogEntry,'12, 0'
4914882: 12, 0
```

TD-er, 2023-10-23:
And that's exactly why we have an alias for `[var#N]` like `%vN%`, so those could be processed first.

chromoxdor, 2023-10-23:
...the worst thing about it is, that i think this is not the first time i fell for that... i am getting old 😳

tonhuisman, 2023-10-23:
> Most likely, `[int#12]` will hold the value 0

I maybe should have written here: An unset `[var#N]` will return 0 😺

chromoxdor, 2023-10-23:
Thanks for your time guys....closing it now

closed 2023-10-23 by chromoxdor