letscontrolit/ESPEasy #4859: Setting for "Force WiFi B/G" is inverted

microkid42, 2023-10-28:
Enabling the "Force WiFi B/G" does not force B/G. Instead, disabling this setting enables a B/G connection.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Advanced
2. Enable Force WiFi B/G

**Expected behavior**
B/G wif is forced

**Screenshots**

**Used platform (please complete the following information):**
 - ESP type: ESP8266
 - Build version: mega-20231013
 - Build set normal

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M
 - Brand/Model: NodeMCU
 - Browser + OS : Mac Brave

**Additional context**

tonhuisman, 2023-10-28:
Ah, yes, this is a known issue, work is in progress to fix this.

iz8mbw, 2023-10-28:
On ESP32 "Force WiFi B/G" works fine:
[image]

From Wireless Controller:
[image]

iz8mbw, 2023-10-28:
Oh yes, issue is only on ESP8266 board.
See https://github.com/letscontrolit/ESPEasy/issues/4678

PR #4863 (2023-10-30): [WiFi] Fix force B/G mode on ESP8266
Also add automatic fallback to a mode supported by the access point when settings force a mode not supported by the access point.  (only on ESP32 or ESP8266 SDK3.x)

Fixes: #4859 
Fixes: #4678
closed 2023-10-30 by TD-er