letscontrolit/ESPEasy #4869: VSCode + PlatformIO

RBroks, 2023-11-05:
Hello, cloud someone add steps how to add plugin from playground.
Or in best scenario video tutorial.

Have tried to active plugin, without succes, not clear where need to add plugin file, what to uncomment in which files...
Someone could help with clear info, for people who try to write code only on sunday eavning.

Thankyou

TD-er, 2023-11-05:
Quite a lot of plugins from the Playground have already been included in ESPEasy.
And those that haven't are likely quite old and may need quite a lot of work to make them compatible with the current code base of ESPEasy.

Which Playground plugin were you thinking of?

There is already some documentation on how to start contributing to ESPEasy: https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#starter-guide-for-local-development-on-espeasy

RBroks, 2023-11-05:
I wanted to add P152_MCP42010.ino. But no luck, i found where to add .ino file, but when compile a build nothing shows up in Espeasy. Compaile without any errors. Do you need to write anything in other files to make it work? 

Info video about VSC + Platform video was great guide!!!

tonhuisman, 2023-11-05:
I've added it to the list of planned plugins. Will take care of migrating it from the Playground, as it is now single-instance only, needs a new PluginID as 152 is already in use, and needs some code-fixup too.

tonhuisman, 2023-11-08:
Hmm, trying to find a board that has this chip, but this type of device seem to only be available as separate chips. Also this type doesn't store the last set state, so after a power cycle, the value needs to be restored/resent from the plugin, that also doesn't keep the last value...
@RBroks Do you really need this exact type, or would an I2C model, that also keeps the last state, and needs less GPIO pins to operate, be a consideration?
For implementing the plugin it won't make much difference, controlling this kind of device is rather straightforward.

RBroks, 2023-11-09:
If there is better solution, i will use that. There is no special need to make it on this chip. It was the only one i could find, where someone has started to do some coding. I am not so good in code writing thats way i was serching atleast some solution…

tonhuisman, 2023-11-10:
I'm currently looking at the [MCP4461](https://www.microchip.com/en-us/product/MCP4461) I2C Digital potentiometer, that has 4 independent channels and can store the power-on value. Not sure if I can get my hands on a DIP-version of the chip (want to put it on a breadboard...), as most suppliers only seem to have the QFN-20 and TSSOP-20 packages available.
Delivery cost is also an issue, at around 6x the price of these chips 😱 as I won't need more than 2 for testing, so quantity discounts won't apply 😉 and the local electronics-parts shop doesn't keep these in stock. Not sure what number of chips you intend to purchase?

tonhuisman, 2023-11-15:
@RBroks Do you have other suggestions, or should I proceed with the chip I mentioned above? (it's also available in 1 and 2 channel versions, I don't expect any changes to use that with this plugin, you just shouldn't try to use the not-available channels 😉)

tonhuisman, 2024-04-10:
@RBroks This turned out a bit different then expected, as I could only find a [DFRobot board with this MCP42xxx](https://wiki.dfrobot.com/Dual_Digital_Pot__100K__SKU__DFR0520) chip installed. It uses a SPI connection, and while the chip supports Reset and Shutdown pins, these are not available in this board configuration, so these pins are optional, and should be set to None when not connected. The MCP42xxx chip also supports daisy-chaining, but as that's also not available on this board, I haven't implemented that in the plugin.

I hope you can obtain such board (or similar, though I haven't searched for alternatives), and test if it suits your situation... 🤔 Please report your findings here.

NB: You can download the latest build via the PR linked right above this comment.

Nevi0s, 2024-04-16:
Thanks for the work @tonhuisman ! I'm really in need of this plugin too, I will try it asap.

tonhuisman, 2024-04-16:
> I will try it asap. With a MCP41010 by the way.

You'll only have the first wiper (W0) available and working, and W1 _will_ still be shown, but not working, of course 😃

Nevi0s, 2024-04-22:
@tonhuisman Hello, It works well, it do the job! So tanks a lot for your work !

tonhuisman, 2024-04-24:
@RBroks Are you available for testing this? Link for downloading is a few messages up in Github.

PR #5030 (2024-05-01): [P162] Add plugin for Digipot MCP42xxx/MCP41xxx
Resolves #4869 

Features:
- Add support for [P162] Digipot MCP42xxx (dual channel)/MCP41xxx (single channel) (other models are prohibitively expensive or very hard to obtain)
  A digipot is a Digital Potentiometer or variable resistor.
- Device is connected via SPI
- Configuration options for:
  -  Initial value
  - Initially in Shutdown state
  - Output value when in Shutdown state
  - Send out data when the wiper value is changed

TODO:
- [x] Add documentation
  - [x] Update EasyColorCode configuration
- [x] Testing
- [ ] Testing by requester  [... cut]
closed 2024-05-01 by TD-er