letscontrolit/ESPEasy #4870: Doesn't compile when used by Custom.h

alex-kiev, 2023-11-08:
I didn’t change anything in the custom.h file, I just renamed the custom-sample.h

src\src\Helpers_tmp\__tmpfile.cpp:29280:103: error: array subscript out of bound
 constexpr size_t Highest_CPlugin_id = ProtocolIndex_to_CPlugin_id[ProtocolIndex_to_CPlugin_id_size - 1];
                                                                                                       ^
src\src\Helpers_tmp\__tmpfile.cpp:29287:77: error: size of array 'CPlugin_id_to_ProtocolIndex' is not an integral constant-expression
 protocolIndex_t CPlugin_id_to_ProtocolIndex[CPlugin_id_to_ProtocolIndex_size]{};
                                                                             ^
Compiling .pio\build\custom_ESP8266_4M2M\FrameworkArduino\FS.cpp.o
Compiling .pio\build\custom_ESP8266_4M2M\FrameworkArduino\FSnoop.cpp.o
src\src\Helpers_tmp\__tmpfile.cpp: In function 'cpluginID_t getHighestIncludedCPluginID()':
src\src\Helpers_tmp\__tmpfile.cpp:29327:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1plus.exe: some warnings being treated as errors
*** [.pio\build\custom_ESP8266_4M2M\src\src\Helpers_tmp\__tmpfile.cpp.o] Error 1
======================================================================================== [FAILED] Took 40.87 seconds ========================================================================================
Environment          Status    Duration
-------------------  --------  ------------
custom_ESP8266_4M2M  FAILED    00:00:40.873
=================================================================================== 1 failed, 0 succeeded in 00:00:40.873 ===================================================================================
 *  The terminal process "C:\Users\big_a\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'custom_ESP8266_4M2M'" terminated with exit code: 1.

TD-er, 2023-11-08:
Ah I see, there should be a check in the code whether there is any CPlugin (or Plugin) included in the build.

alex-kiev, 2023-11-08:
Yes, if you enable any CPlugin it will work without problems

TD-er, 2023-11-08:
I already added a PR to fix it so you can also use just the default Custom.h file.

PR #4871 (2023-11-08): [Build] Fix compile error when using default Custom.h (#4870)
Fixes: #4870
commit dc3bca5f (2023-11-08): [Build] Fix compile error when using default Custom.h (#4870)

Fixes: #4870
closed 2023-11-08 by TD-er