letscontrolit/ESPEasy #4873: [BUG] AP mode does not turn on

alex-kiev, 2023-11-10:
After flashing a file compiled using Custom.h, the access point does not appear, the STA mode is turned on.
If you compile the normal version, the access point works
I used esp8266

INIT : Booting version: ESP_Easy_mega_20231110_custom_ESP8266_4M1M, (Self built) My Build: Nov 10 2023 17:30:55 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443
e), LWIP: 2.1.2 PUYA support)

825 : Info   : INIT : Free RAM:28280

827 : Info   : INIT : Cold Boot - Restart Reason: External System

828 : Info   : FS   : Mounting...

8080 : Info   : FS   : Mount successful, used 0 bytes of 957314

RESET: Resetting factory defaults... using defa
ult settings

RESET: Warm boot, reset count: 0

10986 : Info   : ESPEasy console using ESPEasySerial

RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)

12211 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'

12312 : Info   : Reset WiFi.

 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x4010f000, len 3584, room 16 

tail 0

chksum 0xb0

csum 0xb0

v2843a5ac

~ld

ªU717 : Info   : 

INIT : Booting version: ESP_Easy_mega_20231110_custom_ESP8266_4M1M, (Self built) My Build: Nov 10 2023 17:30:55 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)

718 : Info   : INIT : Free RAM:28280

720 : Info   : INIT : Soft Reboot #1 - Restart Reason: Software/System restart

721 : Info   : FS   : Mounting...

746 : Info   : FS   : Mount successful, used 71284 bytes of 957314

767 : Info   : ESPEasy console using ESPEasySerial

783 : Info   : ESPEasy console using ESPEasySerial

784 : Info   : INIT : I2C

784 : Info   : INIT : SPI not enabled

785 : Info   : Set Network mode: WiFi

890 : Info   : WIFI : Set WiFi to STA

3255 : Info   : Setup: Scan all channels

5445 : Info   : ESPEasy cos5448 : Info   : ESPEasy console using ESPEasySerial

5449 : Info   : INFO : Plugins: 5 [No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(3
8a443e), LWIP: 2.1.2 PUYA support)

7641 : Info   : Webserver: start

8993 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 21800 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED

38996 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 21800 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED

68995 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 21800 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED

TD-er, 2023-11-10:
Can you test with a build based on this PR: https://github.com/letscontrolit/ESPEasy/pull/4841
There are also pre-compiled binaries available based on that PR on the 'latest' web flasher: https://td-er.nl/ESPEasy/latest/
Or you can use the latest GH Actions build based on that PR: https://github.com/letscontrolit/ESPEasy/actions/runs/6824131357

alex-kiev, 2023-11-10:
I flashed the file from here https://github.com/letscontrolit/ESPEasy/actions/runs/6824131357 and it works.
Then I compiled the Custom_8266_4M1M, but the Custom.h file was missing and that also works.
The access point mode does not work only if there is a Custom.h file and the firmware is smaller in size.

[image]

tonhuisman, 2023-11-10:
> Then I compiled the Custom_8266_4M1M, but the Custom.h file was missing

Did you use the sourcecode from the referenced pull request? As that's not available as a separate download, you'll have to use git commands to pull that to your system ([docs](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#get-a-pull-request-by-someone-else-on-your-system)), and then build.
If the `Custom.h` file isn't found, a default custom configuration, as defined in the `pre_custom*.py` scripts is used, so some plugins and features are enabled (different for ESP8266 and ESP32 and in that PR, since very recent, also for ESP32-C2 and ESP32-C6).

alex-kiev, 2023-11-10:
unfortunately I can't do it

PS C:\Users\big_a\Documents\PlatformIO\ESPEasy> git checkout mega
Already on 'mega'
Your branch is up to date with 'origin/mega'.
PS C:\Users\big_a\Documents\PlatformIO\ESPEasy> git pull upstream mega
fatal: 'upstream' does not appear to be a git repository
fatal: Could not read from remote repository.

TD-er, 2023-11-10:
Maybe you can try:

```
git fetch origin +refs/pull/4841/merge:
```

alex-kiev, 2023-11-10:
PS C:\Users\big_a\Documents\PlatformIO\ESPEasy> git fetch origin +refs/pull/4841/merge:
remote: Enumerating objects: 3510, done.
remote: Counting objects: 100% (1453/1453), done.
remote: Compressing objects: 100% (408/408), done.
Receiving objects: 100% (1156/1156), 321.13 KiB | 2.11 MiB/s, done.eused 920 (delta 740), pack-reused 0
Resolving deltas:  12% (116/964)
Resolving deltas: 100% (964/964), completed with 204 local objects.
From https://github.com/letscontrolit/ESPEasy
 * branch                refs/pull/4841/merge -> FETCH_HEAD

TD-er, 2023-11-10:
OK, so you have checked out the code from that PR.
Now you can build it based on that code.

tonhuisman, 2023-11-10:
> unfortunately I can't do it

The docs is based on previously documented commands to set up the `upstream` reference for remote, but you seem not to have set that. The git +ref... command seems to circumvent that, so that should be fine too 😃

alex-kiev, 2023-11-10:
I've never worked with git and the PlatformIO before, I've been studying for literally three days

alex-kiev, 2023-11-10:
Please tell me what should I choose here? 4841 not visible
[image]

tonhuisman, 2023-11-10:
The previous log you showed suggestws you already have the correct sourcecode local, just build the Custom build assuming you have a local `Custom.h` file it should result in the correct Custom build when selecting a Custom environment from the PlatformIO environments list

alex-kiev, 2023-11-10:
In this case, unfortunately nothing has changed. If you have a Custom.h file and select a custom build, the binary file is smaller than with normal compilation and the access point mode does not work (

TD-er, 2023-11-10:
What do you have included in your Custom.h and how do you create your build?
For example, is it possible your build process is fetching the 'mega' branch again when you start building?

alex-kiev, 2023-11-11:
It's probably on the mega branch again

`#ifndef ESPEASY_CUSTOM_H
#define ESPEASY_CUSTOM_H

/*
    To modify the stock configuration without changing the EspEasy.ino file :
    1) rename this file to "Custom.h" (It is ignored by Git)
    2) define your own settings below
    3) define USE_CUSTOM_H as a build flags. ie : export PLATFORMIO_BUILD_FLAGS="'-DUSE_CUSTOM_H'"
 */

/*
 #######################################################################################################
   Your Own Default Settings
 #######################################################################################################
    You can basically ovveride ALL macro defined in ESPEasy.ino.
    Don't forget to first #undef each existing #define that you add below.
    But since this Custom.h is included before other defines are made, you don't have to undef a lot of defines.
    Here are some examples:
 */

// --- Feature Flagging ---------------------------------------------------------
// Can be set to 1 to enable, 0 to disable, or not set to use the default (usually via define_plugin_sets.h)

#define FEATURE_RULES_EASY_COLOR_CODE    1   // Use code highlighting, autocompletion and command suggestions in Rules
#define FEATURE_ESPEASY_P2P       1     // (1/0) enables the ESP Easy P2P protocol
#define FEATURE_ARDUINO_OTA       1     //enables the Arduino OTA capabilities
// #define FEATURE_SD                1     // Enable SD card support
// #define FEATURE_DOWNLOAD          1     // Enable downloading a file from an url

#ifdef BUILD_GIT
# undef BUILD_GIT
#endif // ifdef BUILD_GIT

#define BUILD_GIT           "My Build: "  __DATE__ " "  __TIME__

#define DEFAULT_NAME        "MyEspEasyDevice"                        // Enter your device friendly name
#define UNIT                0                                        // Unit Number
#define DEFAULT_DELAY       60                                       // Sleep Delay in seconds

// --- Wifi AP Mode (when your Wifi Network is not reachable) ----------------------------------------
#define DEFAULT_AP_IP       192, 168, 4, 1                           // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_SUBNET   255, 255, 255, 0                         // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_KEY      "configesp"                              // Enter network WPA key for AP (config) mode

// --- Wifi Client Mode -----------------------------------------------------------------------------
#define DEFAULT_SSID                         "MyHomeSSID"            // Enter your network SSID
#define DEFAULT_KEY                          "MySuperSecretPassword" // Enter your network WPA key
#define DEFAULT_SSID2                        ""                      // Enter your fallback network SSID
#define DEFAULT_KEY2                         ""                      // Enter your fallback network WPA key
#define DEFAULT_WIFI_INCLUDE_HIDDEN_SSID     false                   // Allow to connect to hidden SSID APs
#define DEFAULT_USE_STATIC_IP                false                   // (true|false) enabled or disabled static IP
#define DEFAULT_IP                           "192.168.0.50"          // Enter your IP address
#define DEFAULT_DNS                          "192.168.0.1"           // Enter your DNS
#define DEFAULT_GW                           "192.168.0.1"           // Enter your Gateway
#define DEFAULT_SUBNET                       "255.255.255.0"         // Enter your Subnet
#define DEFAULT_IPRANGE_LOW                  "0.0.0.0"               // Allowed IP range to access webserver
#define DEFAULT_IPRANGE_HIGH                 "255.255.255.255"       // Allowed IP range to access webserver
#define DEFAULT_IP_BLOCK_LEVEL               1                       // 0: ALL_ALLOWED  1: LOCAL_SUBNET_ALLOWED  2:
// ONLY_IP_RANGE_ALLOWED
#define DEFAULT_ADMIN_USERNAME               "admin"
#define DEFAULT_ADMIN_PASS                   ""

#define DEFAULT_WIFI_CONNECTION_TIMEOUT      10000 // minimum timeout in ms for WiFi to be connected.
#define DEFAULT_WIFI_FORCE_BG_MODE           false // when set, only allow to connect in 802.11B or G mode (not N)
#define DEFAULT_WIFI_RESTART_WIFI_CONN_LOST  false // Perform wifi off and on when connection was lost.
#define DEFAULT_ECO_MODE                     false // When set, make idle calls between executing tasks.
#define DEFAULT_WIFI_NONE_SLEEP              false // When set, the wifi will be set to no longer sleep (more power
// used and need reboot to reset mode)
[... 455 lines cut ...]
// #define USES_C006   // PiDome MQTT
// #define USES_C007   // Emoncms
// #define USES_C008   // Generic HTTP
// #define USES_C009   // FHEM HTTP
// #define USES_C010   // Generic UDP
// #define USES_C011   // Generic HTTP Advanced
// #define USES_C012   // Blynk HTTP
// #define USES_C013   // ESPEasy P2P network
// #define USES_C014   // homie 3 & 4dev MQTT
// #define USES_C015   // Blynk
// #define USES_C016   // Cache controller
// #define USES_C017   // Zabbix
// #define USES_C018   // TTN/RN2483

/*
 #######################################################################################################
 ###########     Notifiers
 #######################################################################################################
 */

// #define USES_N001   // Email
// #define USES_N002   // Buzzer

#endif // ESPEASY_CUSTOM_H
`

TD-er, 2023-11-11:
So you just wanted a build with:

-  switch
-  ADC
-  DS18b20
-  ser2Net
-  Cron 
- C005

alex-kiev, 2023-11-11:
For starters, yes.
But in the future I would like to change the set of plugins and initial settings.

alex-kiev, 2023-11-11:
And so I compiled the firmware usingPR4841 and Custom.h, but the problem remained

PS C:\Users\big_a\Documents\PlatformIO\ESPEasy> git fetch origin pull/4841/head:BRANCH_4841
remote: Enumerating objects: 3526, done.
remote: Counting objects: 100% (1475/1475), done.
remote: Compressing objects: 100% (409/409), done.
Receiving objects:  95% (1115/1173)sed 937 (delta 756), pack-reused 0
Receiving objects: 100% (1173/1173), 327.53 KiB | 4.89 MiB/s, done.
Resolving deltas: 100% (980/980), completed with 204 local objects.
From https://github.com/letscontrolit/ESPEasy
 * [new ref]             refs/pull/4841/head -> BRANCH_4841

TD-er, 2023-11-11:
Ton also mentioned to me he also noticed the AP is not starting on a freshly installed ESP node, so that must be looked into first.

alex-kiev, 2023-11-11:
Here is the normal firmware log

`

11023 : Info   : SaveToFile: ControllerSettings index: 0

11024 : Info   : SaveToFile: free stack: 3440

11086 : Info   : FILE : Saved config.dat offsýé11091 : Info   : ESPEasy console using ESPEasySerial

11092 : Info   : SaveToFile: free stack: 3568

11167 : Info   : FILE : Saved config.dat offset: 0 size: 1324

11168 : Info   : SaveToFile: free stack after: 3568

11169 : Info   : SaveToFile: free stack: 3504

11204 : Info   : FILE : Saved security.dat offset: 0 size: 593

11205 : Info   : SaveToFile: free stack after: 3504

11208 : Info   : SaveToFile: ExtendedControllerCredentials index: 0

11209 : Info   : SaveToFile: free stack: 3232

11219 : Info   : FILE : Saved security.dat offset: 1024 size: 6

11221 : Info   : SaveToFile: 
free stack after: 3232

RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)

12222 : Info   : WiFi : WifiDisconnect()

12324 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'

12325 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED

12426 : Info   : Reset WiFi.

 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x4010f000, len 3584, room 16 

tail 0

chksum 0xb0

csum 0xb0

v2843a5ac

~ld

ªU744 : Info   : 

INIT : Booting version: ESP_Easy_mega_20231110_normal_ESP8266_4M1M, (Self built) pygit2_not_installed (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)

746 : Info   : INIT : Free RAM:27600

748 : Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: ResetFactory Last systime: 12 - Restart Reason: Software¯Þ5802 bytes of 957314

794 : Info   : CRC  : Settings CRC           ..þO805 : Info   : ESPEasy console using ESPEasySerial

806 : Info   : INIT : I2C

807 : Info   : INIT : SPI not enabled

808 : Info   : Set Network mode: WiFi

912 : Info   : WIFI : Set WiFi to STA

1088 : Info   : WiFi : Start network scan all channels

3274 : Info   : WiFi : Scan finished, found: 11

3277 : Info   : WiFi : Start network scan all channels

5463 : Info   : WiFi : Scan finished, found: 13

5466 : Info   : Setup: Scan all channels

5467 : Info   : WiFi : Start network scan all channels

7654 : Info   : WiFi : Scan finished, found: 17

7660 : Info   : ESPEasy console usiîö7663 : Info   : ESPEasy console using ESPEasySerial

7664 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)

7667 : Error  : WIFI : No valid wifi setti
ngs

7669 : Info   : WiFi : WiFiConnected(), start AP

7670 : Info   : WiFi : Start network scan all channels

9857 : Info   : WiFi : Scan finished, found: 8

9961 : Info   : WIFI : Set WiFi to AP+STA

10954 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1 ch: 1

10957 : Info   : WiFi : Start network scan all channels

13143 : Info   : WiFi : Scan finished, found: 12

13147 : Info   : WiFi : Start network scan all channels

15335 : Info   : WiFi : Scan finished, found: 16

15339 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED

15342 : Info   : Webserver: start

16681 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 17648 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED`

alex-kiev, 2023-11-11:
This is the custom firmware log

`INIT : Booting version: ESP_Easy_mega_20231111_custom_ESP8266_4M1M, (Self built) My Build: Nov 11 2023 14:35:18 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443
e), LWIP: 2.1.2 PUYA support)

813 : Info   : INIT : Free RAM:28232

815 : Info   : INIT : Cold Boot - Restart Reason: External System

816 : Info   : FS   : Mounting...

7967 : Info   : FS   : Mount successful, used 0 bytes of 957314

RESET: Resetting factory defaults... using defa
ult settings

RESET: Warm boot, reset count: 0

10825 : Info   : ESPEasy console using ESPEasySerial

RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)

12047 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'

12148 : Info   : Reset WiFi.

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 

tail 0

chksum 0xb0

csum 0xb0

v2843a5ac

~ld

ªU724 : Info   : 

INIT : Booting version: ESP_Easy_mega_20231111_custom_ESP8266_4M1M, (Self built) My Build: Nov 11 2023 14:35:18 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)

726 : Info   : INIT : Free RAM:28232

727 : Info   : INIT : Soft Reboot #1 - Restart Reason: Software/System restart

728 : Info   : FS   : Mounting...

754 : Info   : FS   : Mount successful, used 71284 bytes of 957314

775 : Info   : ESPEasy console using ESPEasySerial

791 : Info   : ESPEasy console using ESPEasySerial

792 : Info   : INIT : I2C

792 : Info   : INIT : SPI not enabled

793 : Info   : Set Network mode: WiFi

897 : Info   : WIFI : Set WiFi to STA

3260 : Info   : Setup: Scan all channels

5450 : Info   : ESPEasy cÿû5453 : Info   : ESPEasy console using ESPEasySerial

5454 : Info   : INFO : Plugins: 5 [No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(3
8a443e), LWIP: 2.1.2 PUYA support)

7647 : Info   : Webserver: start

8996 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 21576 WiFiStatus WL_IDLE_STATUS 0 ESPeasy internal wifi status: DISCONNECTED`

alex-kiev, 2023-11-11:
It can be seen that files with default settings are not created

TD-er, 2023-11-11:
N.B. you need 3 backticks on a separate line to mark a block as 'code formatted'
The single backtick is to do it `inline like this`

tonhuisman, 2023-11-11:
If you type the command `wifiapmode` in your serial terminal to the ESP (there is no echo until you press <Enter>) you will manually activate AP mode, so you can setup the device via the accesspoint.
As TD-er said, the not-automatically-starting of the AP is a point of concern that's being investigated.

tonhuisman, 2023-11-11:
@alex-kiev I've found a possible culprit and made a fix for that, can you test either the (rather small, the method rename is just for clarity) fix in #4875 or download a build from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6836421268) when that's finished, for testing?

alex-kiev, 2023-11-12:
```git fetch origin pull/4875/head:BRANCH_4875```

I compiled the firmware using PR#4875 and Custom.h for esp8266, the access point is now working.
Now I can create truly custom firmware).
Thank you very much for your help and patience.

closed 2023-11-12 by alex-kiev

tonhuisman, 2023-11-12:
FYI: No need to manually close the issue, it will be auto-closed once the linked PR is merged. Other users having the same issue can more easily find this when it's still visible/active until the fix is merged, which can take some time as more people are testing the fix. You can leave it now as-is, just a heads-up for a next time 😃