letscontrolit/ESPEasy #4876: [FR] Please add support for LCD Display - MIDAS DISPLAYS MD21605B6W-FPTLWI3

bkrajnc, 2023-11-15:
**Is your feature request related to a problem? Please describe.**
I can not connect LCD display 2x16 with I2C to ESP8266F, becouse it is not supported

**Describe the solution you'd like**
I woul like to add LCD whic I buyed at https://si.farnell.com/midas/md21605b6w-fptlwi3/lcd-display-cob-transflective/dp/3565078
It could be using the ST7032 controller.

**Describe alternatives you've considered**
Unfortunately there is no alternatives for this LDC

**Additional context**

fly74, 2023-11-17:
Looks cool too https://www.[winstar.com.tw/de/products/oled-module/graphic-oled-display/wea025664a.html](https://www.winstar.com.tw/de/products/oled-module/graphic-oled-display/wea025664a.html)

tonhuisman, 2023-11-17:
> Looks cool too https://www.winstar.com.tw/de/products/oled-module/graphic-oled-display/wea025664a.html

That is a _completely_ different display... 🤔

bkrajnc, 2023-11-18:
> The user requesting this support already has bought a bunch for a project. (was first mentioned on the forum) So in this use case I don't think he would like to see a different display being supported :)

Yes, TD-er you are right. I already bought 24 displays :-( .

tonhuisman, 2023-11-18:
I'm looking at the code for both the current P012 LCD plugin and the ST7032 controller that this display is most likely using, and though there are some differences, these differences are not huge, so I'll try to add an option to the used library to make it work with this display.
@bkrajnc I don't own such display (though I do own a 20x4 LCD I2C display that's already supported by ESPEasy), so I can't test myself and must rely on you for testing 😸 
I'll let you know when there is something to test.

bkrajnc, 2023-11-18:
> I'm looking at the code for both the current P012 LCD plugin and the ST7032 controller that this display is most likely using, and though there are some differences, these differences are not huge, [...]

@tonhuisman, many, many thank you. Yust inform me here, and I will test immediately ;-)

tonhuisman, 2023-11-19:
@bkrajnc I have patched the LCD library to also support the ST7032 controller (that's supposed to be used in your display).
This [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6922496177) is building the binaries. This plugin is included in nearly all builds, so you can pick the best fitting for your project for testing.
You'll have to select the correct I2C address `0x3E`, and Display Size `2 x 16 (ST7032)` for this test.

Please report your findings here, or in the pull request I linked above this comment (not in both, that's just confusing 😉).

bkrajnc, 2023-11-19:
@tonhuisman , thank you for your fast response. I tryed, but unfortunately it doesn't work. I can see i2c communication but  Nothing come to the display. I have only one open question and this is, if i must connect pul up resistor on the display, becouse it is not described in documentation. See attached files.
What do you suggest? Can I debug somehow?
[image]
[image]

tonhuisman, 2023-11-19:
I2C should always have pull-up resistors on the SDA and SCL lines, something like 4k7 or 10k should be fine, but many devices already have them installed, so maybe you won't need them (you can measure it when the device isn't connected and not powered).
And most likely you will have to pull-up the reset line too, as it might see disconnected as low == reset active.

Does this board work with 5V or 3.3V logic? When using with 5V you'll need a level converter for the SDA/SCL lines, because the ESP is not guaranteed to handle 5V well on the GPIOs, and possibly/probably the 3.3V 'high' level from the ESP isn't enough to trigger a 5V input to high level.

tonhuisman, 2023-11-20:
In review, it seems I forgot a small piece of code, specific to the ST7032. A new [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6934803728) is building new binaries.

bkrajnc, 2023-11-20:
@tonhuisman , ........unfortunately....still doesnt't work

tonhuisman, 2023-11-20:
Well, if it finds the device, then usually the wiring rather OK, though I have seen I2C devices that still report they work if SDA and SCL are switched. You could try that, just for kicks, and see if it still is detected? (but usually it won't work).

I've stumbled on another ST7032 library that does the initialization quite differently, so I'll do another build with that configuration, so you can test that too.

bkrajnc, 2023-11-20:
I switch SDA and SCL but no. I will wait for new build ;-)

tonhuisman, 2023-11-20:
A new build is running [from here](https://github.com/letscontrolit/ESPEasy/actions/runs/6936462923)

bkrajnc, 2023-11-21:
I tryed, but no, it doesn't work. I waiting for LCD manufacturer response.

tonhuisman, 2023-11-21:
> I tryed, but no, it doesn't work

It's a bit disconcerting that after trying a couple of really different initializations it still isn't working at least a bit. The last change I had high hopes of, as it sets the contrast of the display, because having 0 contrast isn't going to show much readable on the display.
Probably it is better if I test it myself on real hardware, but maybe best if we try to ship one from your location (?) to me (NL), so there is no confusion on that point. You can contact me via PM in the Forum (name: **Ath**, same avatar 😉).

tonhuisman, 2023-11-21:
> I waiting for LCD manufacturer response.

I assume you'll be sent to this page: https://midasdisplays.com/shop/lcd/lcd-character/md21605b6w-fptlwi3/
Did find the technical docs there, but no software interface or description 😞

tonhuisman, 2023-11-21:
It looks like you will need to pull up pin 6 (A) to VCC using a resistor between 470 - 1000 ohm, and connect pin 7 (K) to GND. That should activate the LED backlight, and might help make the display content visible.

bkrajnc, 2023-11-21:
@tonhuisman , thank you for all effort and advices and I will try tomorow in the morning what you suggest. Of course will also report about findings.

TD-er, 2023-11-21:
> It looks like you will need to pull up pin 6 (A) to VCC using a resistor between 470 - 1000 ohm, and connect pin 7 (K) to GND. That should activate the LED backlight, and might help make the display content visible.

Backlight shouldn't make a difference when you don't even see anything on the display.
Well... with the whoppin' contrast ratio of 5:1 anything can help, but still..

And the rise/fall times of the display are with 150 msec awfully slow.
So better not try to scroll text as it will be a big smeer..

I still think you really need to pull the RST pin (pin 5) up to 3V3.

bkrajnc, 2023-11-22:
Hi all. I am at the point where I think  I will surrended :-). I tryed everything you suggested @tonhuisman and @TD-er , but nothing comes to the LCD :-( .

tonhuisman, 2023-11-22:
If you're not returning the displays to the supplier, you could send one over for further experimenting (as long as that's financially viable, sending it to NL), as said before: you can contact me via PM in the Forum (name: **Ath**, same avatar 😉.

bkrajnc, 2023-11-23:
Hi again. I receive email from manufacturer en he said:
I’ve attached Arduino demo code along with some initialisation code of the display controller. 
The Reset pin is defined in this code,  as I believe it’s required to initialise the ST7032 on the display, you can use any digital GPIO for his purpose.
[ST7032.zip](https://github.com/letscontrolit/ESPEasy/files/13449442/ST7032.zip)

Demo code is in attachement

TD-er, 2023-11-23:
So, that's a lot of delay() calls in the `LCD_Init()` function....
And also that it needs to call this init function 3 times in the `setup()` function with 50 msec delay inbetween is not looking very reliable.

At least they did include quite extensive datasheets.

tonhuisman, 2024-12-21:
I've closed the PR that was supposed to get this display working. I've not found ground, nor the technical documentation, to try and support this.
Case closed.