letscontrolit/ESPEasy #4885: Can not connect hidden SSID

ganzales, 2023-11-19:
Can not connect to hidden SSID by Mikrotik router, but same board under tasmota connecting well to this network. Not work only with Mikrotik, other routers works well.

TD-er, 2023-11-19:
Just to make sure, you have checked the "Include Hidden SSID:" checkbox on the Setup page or on the Config page? (both the same checkbox)

And do you have put a limit on what can connect to the Mikrotik? 
For example set to "only 802.11n" or "only 802.11g"?
On the ESP8266 there was a bug where the checkbox for forcing "b/g only" was actually inverted.
So best to uncheck it or use one of the latest GH Actions builds to test if you're running an ESP8266.

What board are you using and which model of Mikrotik?
Can you connect to the Mikrotik when SSID is not hidden?

ganzales, 2023-11-20:
Option "Include Hidden SSID" checked and ESP8266 connect to other devices hidden network. Router MikroTik hAP ac², same network, but open working with out problem.

TD-er, 2023-11-20:
Which firmware build are you using?

ganzales, 2023-11-20:
Fresh: 7.11.2 (stable)

TD-er, 2023-11-20:
Ehhh... that's not an ESPEasy build nr.

TD-er, 2023-11-20:
OK, you mean the RouterOS firmware...
I just confirmed my ESP8266 test board cannot connect to a Mikrotik unit here as soon as I set it to hidden SSID, so will look into it.

TD-er, 2023-11-20:
I have it working here, but I'm not 100% sure why this solution works and the others don't...
So I will make a PR for it, but I'm still thinking about how to make this a generic fallback option or maybe a user setting specific to Mikrotik.

TD-er, 2023-11-20:
I added a setting "Hidden SSID Slow Connect:" which is made available on the Setup page, Config page and Advanced settings page.

It is enabled by default.

PR #4887 (2023-11-27): [WiFi] Fix connect to Hidden SSID on Mikrotik AP (#4885)
Fixes: #4885