letscontrolit/ESPEasy #4888: Can't enable I2C Device

xxxturboxxx, 2023-11-23:
**Describe the bug**
When I add an I2C device and click on "enable" the blue check mark appears and when I click on "submit" it disapears again so I can't activate my sensor!

[image]

**Used platform (please complete the following information):**
 - ESP type: ESP8266
 - Build version: 20231013 - Mega
 - Build set ESP_Easy_mega_20231013_normal_ESP8266_4M1M

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: Espressif Generic ESP8266 ESPEasy 4M Flash 1M FS
 - Brand/Model: Wemos D1 mini pro
 - Browser + OS Chrome, Win11

xxxturboxxx, 2023-11-23:
additional info...

Tryed it with Edge instead of Chrome and I could activate it once so the green check mark appeared in the deviceoverview. Going into the device settings the second time then it is deactivated again and can't be reaktivated anymore...

tonhuisman, 2023-11-23:
What is the device you selected? "An I2C device" is a bit too global/unspecific.

An I2C device *must* be connected and working to be enabled. A protection has been instated to fix the issue that some I2C device-libraries block the unit for a long time if they can't find the device. To keep the unit working this protection has been instated.

In the Tools/Advanced page there is a setting to enable that protection, default it's on, so it can be turned off.

xxxturboxxx, 2023-11-23:
You'right. Sorry for this missing info...

[image]

When I tick "Force Slow I2C speed" it works fine.

tonhuisman, 2023-11-23:
> When I tick "Force Slow I2C speed" it works fine.

Ah, then that's a limitation of the sensor. Though many sensors work fine at 400 kHz, some are limited too 100 kHz communication speed, this is usually specified in the datasheet.

xxxturboxxx, 2023-11-23:
Ok, Thanks for the qick help!

TD-er, 2023-11-23:
@tonhuisman 

So maybe that should be added to the plugin, to add a note it only supports upto 100 kHz and set its default to "slow I2C" ?
Preferrably in the `DevicesStruct` so you only have to add this code once :)

PR #4889 (2023-11-27): [I2C] Add Device flag for 100 kHz-only devices
Resolves: #4888 

Features:
- Add device-flag for 100 kHz-only devices
  - Enable **Force Slow I2C speed** checkbox by default when *adding* the device
  - Show message when having such device that it should be using 100 kHz max I2C speed
- Updated documentation

100 kHz-Only devices:
- P007 - Analog input - PCF8591
- P024 - Environment - MLX90614
- P117 - Gases - SCD30 CO2, Humidity, Temperature
- P135 - Gases - CO2 SCD4x
closed 2023-11-27 by TD-er