letscontrolit/ESPEasy #4890: [BUG] bin gets rejected because of file size when building in vscode

chromoxdor, 2023-11-27:
Just wanted to test the new CDN option..
but when i want to build a "normal_ESP8266_4M1M" i get this:

[image]

any ideas?

TD-er, 2023-11-27:
Hmm I tried merging PRs faster than the build size would increase, but apparently I failed...

I will have a look at it.

Maybe even the 'normal' build on ESP8266 should now use the CDN and not include CSS/JS etc?

chromoxdor, 2023-11-27:
But the flash size seem to be at first ok (`Flash: [==========]  99.7% (used 1041496 bytes from 1044464 bytes)`)... where is the additional data coming from?

TD-er, 2023-11-27:
> But the flash size seem to be at first ok (`Flash: [==========] 99.7% (used 1041496 bytes from 1044464 bytes)`)... where is the additional data coming from?

Bootloader stuff.
I have to check whether it is a false negative, but I think the actual bin file size on the disk is what should be considered (as what I'm doing now)

It even got worse with the latest PRs being merged:

```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  58.9% (used 48216 bytes from 81920 bytes)
Flash: [==========]  99.8% (used 1042080 bytes from 1044464 bytes)
 gzip:   .pio\build\normal_ESP8266_4M1M\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.bin
bin_elf_copy([".pio\build\normal_ESP8266_4M1M\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.bin"], [".pio\build\normal_ESP8266_4M1M\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.elf"])
 copy to:   build_output\debug\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.elf
 file size: 1046240 > 1044464
 copy to:   build_output\reject\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.bin
 JSON to:   build_output\json\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.manifest.json
 copy to:   build_output\bin\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.bin.gz
 copy to:   build_output\debug\ESP_Easy_mega_20231127_normal_ESP8266_4M1M.env.txt
 Timestamp: 2023-11-27 17:15:02.112595
```

TD-er, 2023-11-27:
You could look at https://github.com/letscontrolit/ESPEasy/pull/4891

chromoxdor, 2023-11-27:
Perfect!

PR #4891 (2023-11-27): [ESP8266 build] Set all ESP8266 'normal' builds to use CDN for CSS/JS
Fixes: #4890
commit e5f4e44c (2023-11-27): [ESP8266 build] Set all ESP8266 'normal' builds to use CDN for CSS/JS

Fixes: #4890
closed 2023-11-27 by TD-er