letscontrolit/ESPEasy #4892: [Wanted] Need testers for upcoming build (end Nov '23)

TD-er, 2023-11-27:
Lots and lots of changes, so we're up for a new build.

@Fly74 @ghtester @chemmex @alex-kiev  @iz8mbw @wookbert @dickesplumpel
Can you guys please have a look at this test build: https://td-er.nl/ESPEasy/testbuild/

N.B. I already re-organized the list of items in the webflasher, sorted by file system type.

The upcoming ESP-IDF5.1 will be merged after the next build, so that will be for the "Christmas build" and will give you a few really nice new things :)

chemmex, 2023-11-28:
Missed this in previous discussions: after swithcing from AP to STA mode on first connection, internal http sever is not reachable on newly obtained IP address. It was different in previous builds.

ESP8266, normal build

TD-er, 2023-11-28:
Yep, this is a known issue and I don't yet know how to fix this.

What happens is this:
- Client is connected via ESP's AP
- Client fetches `/setup` + defers loading CSS (which may or may not get served)
- User configures WiFi credentials
- ESP tries to connect to WiFi, which causes client to disconnect from ESP's AP
- Webserver on ESP tries to serve CSS to no longer connected client
- Since client is no longer connected, the buffered packets are not fetched from the webserver's buffer and remain in memory
- Free Memory is too little to serve pages to the client connected through ESP's STA interface.

TD-er, 2023-11-28:
@uzi18 Not sure what you mean?

ghtester, 2023-11-28:
> 
> @fly74 @ghtester @chemmex @alex-kiev @iz8mbw @Wookbert @Dickesplumpel Can you guys please have a look at this test build: https://td-er.nl/ESPEasy/testbuild/
> 
Well, as I need a Custom build with my preferences, I'll make it with already merged changes and I can test it on some of my ESP nodes.

> Missed this in previous discussions: after swithcing from AP to STA mode on first connection, internal http sever is not reachable on newly obtained IP address. It was different in previous releases.
> 
> ESP8266, normal build

I think  I am experiencing the similar issue (on 'recent' builds - a couple weeks/months old) despite I don't use the AP mode at all (it's disabled). But after cold boot or node reboot, the very first attempt to open the ESP node web GUI fails (usually). Did not test it thoroughly yet.

Dickesplumpel, 2023-11-28:
Hi, at the moment I have not much time to test "intensive" and I don't know what explicit you wanted to test, so:
20231127 normal 4M LittleFS ESP32 can't connect to Wifi.
Wifi configuration first via iPhone to AP-Web, then via Arduino IDE serial but both:
686821 : Info   : WiFi : Added known candidate, try to connect
686821 : Info   : WiFi : WifiDisconnect()
686926 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
686928 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
687028 : Info   : WIFI : Set WiFi to OFF

TD-er, 2023-11-28:
> I have Edge only on my gaming PC, usually I use Firefox. Is there a way to download the whole stuff "the normal way"?

Sure, you can take this latest GH Actions build:
https://github.com/letscontrolit/ESPEasy/actions/runs/7011191383

TD-er, 2023-11-28:
Yep, latest merge to 'mega' branch.
The files on the web flasher I linked are from the PR which was merged into that build you linked.
So should be the same code.

iz8mbw, 2023-11-28:
All OK here on ESP32 boards, two 4MB and one 16MB.
Using Climate, Energy and Custom (via Custom.h).

fly74, 2023-11-28:
No issues at all.

TD-er, 2023-11-28:
> Hi, at the moment I have not much time to test "intensive" and I don't know what explicit you wanted to test, so: 20231127 normal 4M LittleFS ESP32 can't connect to Wifi. Wifi configuration first [...]

Could you reproduce this behavior?

chemmex, 2023-11-28:
> Yep, this is a known issue and I don't yet know how to fix this.
> 
> * Free Memory is too little to serve pages to the client connected through ESP's STA interface.

Can this be avoided by embedding CSS in custom build?

TD-er, 2023-11-28:
> > Yep, this is a known issue and I don't yet know how to fix this.
> > 
> > * Free Memory is too little to serve pages to the client connected through ESP's STA interface.
> 
> Can this be avoided by embedding CSS in custom build?

If you have the Gzipped CSS included in the build, then this cannot be embedded inline in the HTML.
I was thinking about including some decompressor like sox. This way we can still have stuff compressed embedded in the build and serve it decompressed.

chemmex, 2023-11-30:
And this build does not include #4875, right?

TD-er, 2023-11-30:
> And this build does not include #4875, right?

Correct
The way how Ton fixed it, was actually not the correct way to do it as far as I could see.
So maybe it is better to leave a known bug in than to introduce several others.

As far as I could see, his fix -when applied- will cause issues connecting to hidden SSIDs.

However, it might be that I accidentally fixed this one too when fixing connecting to Mikrotik hidden SSIDs?

TD-er, 2023-11-30:
A build has been made, so this can be closed.

Thanks for testing.

closed 2023-11-30 by TD-er