letscontrolit/ESPEasy #4893: [BUG][P076] calculation with multiplier values is broken

chromoxdor, 2023-11-28:
Todays build with p076 causes it to display nonsense. (last uses/working build ESP_Easy_mega_20220722)
Seems to be an issue with the multiplier values. 
If i use the same values as before or try the recalibrate with `hlwcalibrate` the output is not as expected.
e.g. with pure resistive load (80W): `hlwcalibrate,228,0.352,80`  -> 
[image]

chromoxdor, 2023-11-28:
Changes i found related to P076:

https://github.com/letscontrolit/ESPEasy/pull/4663/files#diff-7ec68dc04e4be4c0170970cfcd3ef1f88d6ab1cb1cd4edd87b325481315ab044

TD-er, 2023-11-28:
A lot has been changed to that plugin since 20220722.
Does your unit have significantly different default multipliers compared to the used defaults in the code?

chromoxdor, 2023-11-28:
> Does your unit have significantly different default multipliers compared to the used defaults in the code?

I dont´t think so.
default:
[image]

calibrated:
[image]

TD-er, 2023-11-28:
My voltage multiplier is 462022.94
I never calibrated the current/power on that node.

You could try to use the calibration command with only the voltage, just to see if that's working.

chromoxdor, 2023-11-28:
> You could try to use the calibration command with only the voltage, just to see if that's working.

It doesn´t

> My voltage multiplier is 462022.94

I have around 8 devices here (all gosund sp1 or sp111)
They all have similar values and they definitely needed calibration...

TD-er, 2023-11-28:
Maybe there needs to be an actual floating point value?

Can you try 
```
hlwcalibrate,228.0,0.352,80.0
```

TD-er, 2023-11-28:
I think I see what may be happening here....
The function to apply calibration needs a proper last measurement value.
However it is somewhat of a trial-and-error to hit the command at just the right moment to get a proper `_voltage` value as this `getVoltage` function may every now and then return 0 as the reading is not yet ready.

TD-er, 2023-11-28:
Made a fix for it, can you please test it?

chromoxdor, 2023-11-28:
Command works! Sadly calculation is still wrong.....

TD-er, 2023-11-28:
"current" as in the last value, or the actual current  as in amps?

chromoxdor, 2023-11-30:
just realized that the power multiplier also doesn´t get changed.
So it´s only voltage that works.

TD-er, 2023-11-30:
Just curious, what is the actual current being used the moment you try to perform the calibration?

Like if you tell the unit there is some load like 100 Watt, you really should have it present and the task already must have it measured.

So if you have the task set to 5 minutes and you apply some load, perform the calibration and disconnect the load, then it will fail as the measured current will still be 0.

TD-er, 2023-11-30:
Since this is already an older bug, it doesn't really make a difference whether I will fix it today or tomorrow (and you will be perfectly capable to make your own builds ;) )

Deadline for a fix for this one to be included in the November build is ~20h today.