letscontrolit/ESPEasy #4896: [BUG][P159] "Generate Events only when changed" not working.

chromoxdor, 2023-11-28:
Finally a pair of LD2410 arrived and i could also do a bit of testing. Very interesting devices...

@tonhuisman, not sure this is a bug but i also couldn´t find anything about it.
When i check `Generate Events only when changed` and hit submit, it will be unchecked again.

tonhuisman, 2023-11-28:
Hmm, looks like a bug indeed, it's not properly displayed it seems. Will fix

tonhuisman, 2023-11-28:
Hmm2, not able to reproduce here, using an ESP32 MAX build. @chromoxdor What build are you using?

chromoxdor, 2023-11-28:
ESP_Easy_mega_20231128_collection_G_ESP8266_4M1M

iz8mbw, 2023-11-28:
Here seems to works well.
Using mega_29cfc92 om MAX (16M) build.

tonhuisman, 2023-11-28:
Testing that the ESP8266 Collection G too, and can confirm it's not saved correctly,
have to go now, continue tomorrow

TD-er, 2023-11-29:
> @chromoxdor The bugfix is PR'd, please test/confirm (tested by myself already, of course 😉)

Huh?
Do you actually test before committing?
Why? ;)

PR #4897 (2023-11-29): [Bugfix] [P159] Saving settings checkbox failed on ESP8266
Resolves #4896 

Bugfix for the settings checkbox 'Generate Events only when changed' that wasn't being saved on ESP8266.
closed 2023-11-29 by TD-er