letscontrolit/ESPEasy #4898: [BUG] DOCs on Website is not updated

iz8mbw, 2023-12-01:
Hi.
I noticed the DOCs on Website https://espeasy.readthedocs.io/en/latest/ is not updated.
Just some examples of merged commits not published on website:

These commits https://github.com/letscontrolit/ESPEasy/commit/936d5b24e47e906288a3cc59fe9ce113a70d0ecd and  https://github.com/letscontrolit/ESPEasy/commit/2a2d6451d01a0a4cbcd958d9e045185219f82079 are not published on https://espeasy.readthedocs.io/en/latest/Plugin/P014.html

This commit https://github.com/letscontrolit/ESPEasy/commit/2a2d6451d01a0a4cbcd958d9e045185219f82079 is not published on https://espeasy.readthedocs.io/en/latest/Plugin/P014.html

This commit https://github.com/letscontrolit/ESPEasy/commit/d80948f396c56f41a0b9be90ad0d3045a7a6f30b is not published on https://espeasy.readthedocs.io/en/latest/Plugin/P037.html

This commit https://github.com/letscontrolit/ESPEasy/commit/787316fd25600d6cb1e6230c5ff731ca45f3f51b is not published on https://espeasy.readthedocs.io/en/latest/Plugin/P021.html

This commit https://github.com/letscontrolit/ESPEasy/commit/c7344df03f6fddc97874adf028da7e819c1ef9f6 is not published on https://espeasy.readthedocs.io/en/latest/Plugin/P159.html

and so on...

TD-er, 2023-12-01:
Ton also mentioned something similar.
I will have a look at why this isn't updated as it should be done automatically when something gets merged.

TD-er, 2023-12-01:
Let's see if it is now acutally fixed.
There was a missing `.readthedocs.yaml` file

commit 8f257f48 (2023-12-01): [Docs] Add .readthedocs.yaml to fix auto build

Fixes: #4898
closed 2023-12-01 by TD-er