letscontrolit/ESPEasy #4901: To use system variables Otput clock plugin[FR]

mariolazarov, 2023-12-05:
Output clock variables like %sunset% or %sunrise% to be used instate of hh:mm.

tonhuisman, 2023-12-05:
Mmm, IMHO, this can better be handled via Rules. Anything this plugin offers can be done using rules, and probably more flexible too.
Only exception to that could be that it's not in a full UI, with fields to fill in etc. though that also imposes some limits, and rules _do_ have a kind of UI, when using the syntax highlighting, though probably a little less guiding as a plugin UI.

mariolazarov, 2023-12-06:
Hello Ton,
Yes through rules I use to turn on and off the outdoor lighting in my home, also the water heaters on day and night tariff here in my country we have 2 tariffs 06:00-22:00 and respectively 22:00-06:00 as the day price is twice the nightly price and I use Output clock plugin for that. Thanks for the attitude to my question and the elder reaction, I just decided that many of us will use it in our home and it will still be easier and faster, so the idea is ESPEasy to be easier if posible.

chromoxdor, 2023-12-06:
I am also unhappy with the timer plugin. 
What if we steal the timer from tasmota 😁 I really like it and is has all the features...

[image]

TD-er, 2023-12-06:
What timer plugin do you mean?

chromoxdor, 2023-12-06:
I mean [p043] clock plugin... (sorry i got confused because in tasmota it is called timer )

TD-er, 2023-12-06:
> [...] What if we steal the timer from tasmota 😁 I really like it and is has all the features...

We don't "steal" from Tasmota, we just borrow :)

tonhuisman, 2023-12-06:
I've thought out a _possible_ solution for handling `%sunrise%` and `%sunset%` in P043, but I just have to confirm it doesn't get 'confusing' as that notation is more often used in rules like the `on System#Clock=All,%sunset% do` events.

tonhuisman, 2023-12-07:
It's not the kind of UI as available in Tasmota, but I've made it possible to store, retrieve and process the Clock time configurations having `%sunrise%` and `%sunset%` variables (including offsets).

@mariolazarov Can you test please, once [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/7134263379) is completed?

mariolazarov, 2023-12-08:
[image]
My idea is to just use it like that if possible of course

tonhuisman, 2023-12-08:
Well, that would be a downgrade in features, and as that's not backward compatible, we can't go that way.

I'll think of how to possibly combine this with the existing feature set.

mariolazarov, 2023-12-08:
I understand though it would be nice to have an option to use system variables. Thanks for the comprehensive answer and for the attitude.

tonhuisman, 2023-12-08:
> it would be nice to have an option to use system variables

Well, that'll be a challenge, the value is stored as a `long`, meaning numeric only... I've been able to squeeze in the `%sunrise%` and `%sunset%` values as these are _quite_ relevant, but adding other variables won't be feasible, AFAICS. 😞

TD-er, 2023-12-08:
@tonhuisman 
I told you so, whenever you try to add 1 single system variable like %sunset%, there will be demand for any other system variable.
And now it was even asked before it was merged, so even more quickly than expected.

At least you have some bits left, so in theory, you could allow for a few variables like `[int#1#D2]:[int#2#D2]`

chromoxdor, 2023-12-09:
Thanks for adding `%sunrise%` and `%sunset%`!

I personally think there is no need in adding more options timewise and for everything else i cite ton:
>  Anything this plugin offers can be done using rules, and probably more flexible too.

The clock plugin is a nice graphical option which is a fast and easy way of setting specific times. The only thing i personally would add is an easier way to set the time (and days) as shown in my tasmota post but i also understand that this is more complicated to implement...

That sounds a bit hypocrite since i wanted to have an option to set and read the values of the plugin via a html request.
But this was to make a nicer UI in JS (as an addition to easyfetch) for a Timer which merely uses the clock plugin for storing the values. But i realized that i would need to change the structure of the clock plugin (e.g.  choosing multiple days for one time event) in a way that would make this UI obsolet. And i also already added  a feature for Clock Plugin in easyfetch long time ago :) :
[image]

chromoxdor, 2023-12-09:
But since this plugin gets changed: Can we remove the "change way the plugin works when no GPIO is selected"- feature.
It is not intuitive to use and i don´t see what the added value is in this other than being confusing.
Or did somebody added this for a good reason?

tonhuisman, 2023-12-09:
I've added a set of predefined values to choose from for non-`LIMIT_BUILD_SIZE` builds (for size reasons): (`<day>,00:00`, `<day>,%sunrise%` and `<day>,%sunset%`, that can be edited after selection, `<day>`=All,Sun,Mom,Tue,Wed,Thu,Fri,Wrk,Wkd)

[image]

tonhuisman, 2023-12-09:
> Or did somebody added this for a good reason?

That's usually the reason for this kind of special features, and I can see a use for this when not having a GPIO configured. You just need to handle the event for that in Rules, and then you want to know what line in the settings you're triggered from.

mariolazarov, 2023-12-10:
That's exactly what I had in mind when I started the thread. Now this plugin is much more flexible and useful. Thank you guys for the good work.

tonhuisman, 2023-12-10:
Well, I've done some more improvements 😺 (parts suggested by TD-er)

[image]

1) Set the number of Day,Time fields in range 1..16 (Submit the page to update the fields)
2) Make the Day part a separate dropdown list with all allowed values
3) Make the Time part a text field with a limited datalist (3 items, optionally appended with user-entered values, depending on the browser)

tonhuisman, 2023-12-11:
> @tonhuisman is it possible to add an option for changing to the "GPIO is used" layout when no GPIO is used.

@chromoxdor Is this what you intended? (No option, just an alternative, more basic, layout)

[image]

mariolazarov, 2023-12-11:
In my case, it has been working for three days now, it turns on and off the lights in the yard of my home right on time. Thanks again for the good work.

tonhuisman, 2023-12-11:
You don't have to close this issue, it will be auto-closed by Github when the pull request is merged, as I linked that to this issue.

chromoxdor, 2023-12-11:
> @chromoxdor Is this what you intended? (No option, just an alternative, more basic, layout)

Looks good :)
But i meant an option where people can choose between this layout and the "old" layout when no gpio is selected.
Like i stated earlier, this "new" layout is not very intuitive. I mean i got it now :) but i don´t want to explain a non-nerd every time he wants to set the timer how it works... 😁

mariolazarov, 2023-12-11:
Sorry I didn't expect it to be a problem. I am satisfied with the result of my question and I think that what I proposed was achieved

chromoxdor, 2023-12-11:
> Sorry I didn't expect it to be a problem.

There is no problem. I hijacked your feature request :) . I should have probably posted this directly in the PR. Sorry...

mariolazarov, 2023-12-11:
Няма проблем за мен ако имащ други идеи давай ;)

tonhuisman, 2023-12-11:
> choose between this layout and the "old" layout when no gpio is selected.

And make it even more confusing? 🤣

PR #4903 (2023-12-28): [P043] Add support for %sunrise% and %sunset% based time-values
Resolves #4901 
Resolves [this forum request](https://www.letscontrolit.com/forum/viewtopic.php?t=9931#p67184)

Features:
- Add support for time configurations using `%sunrise%` and `%sunset%` instead of explicit HH:MM format, including the possible offsets like `+1h`, `-20m` and `+13200s` (220 minutes...)
- For non-`LIMIT_BUILD_SIZE` builds a list of a separate Day selector (All, Sun,Mon-Fri,Sat, Wrk = Workday, Wkd = Weekend) and predefined Time configurations with `00:00`, `%sunrise%` and `%sunset%` that can be selected and then [... cut]