letscontrolit/ESPEasy #4905: nodemcu wont start

mpl1337, 2023-12-08:
i try to install the 20231130 climate 4m solo1 on my Nodemcu (AZ-Delivery ESP32-WROMM-32) over the Webflasher and the ESP tool, but the ESP wont start.

I tryed 3 ESPs but on every esp the same problem.

Whats my problem?

<\n><\r>
<\n>rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r>
<\n>configsip: 0, SPIWP:0xee<\r>
<\n>clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r>
<\n>mode:DIO, clock div:2<\r>
<\n>load:0x3fff0030,len:184<\r>
<\n>load:0x40078000,len:12736<\r>
<\n>load:0x40080400,len:2716<\r>
<\n>entry 0x40080594<\r>
<\n>ªU199 : Info   : <\n><\n><\r>
INIT : Booting version: ESP_Easy_mega_20231130_climate_ESP32solo1_4M316k, (GitHub Actions) mega-20231130 (ESP32 SDK 4.4.5.230614)<\r>
<\n>201 : Info   : INIT : Free RAM:296392<\r>
<\n>203 : Info   : INIT : Soft Reboot #304 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: CPU0: Software reset CPU CPU1: for APP CPU, reset by PRO CPU<\r>
<\n>204 : Info   : FS   : Mounting...<\r>
<\n>214 : Info   : FS   : Mount successful, used 141815 bytes of 290156<\r>
<\n>225 : Info   : CRC  : Settings CRC           ...OK<\r>
<\n>261 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>267 : Info   : CRC  : SecuritySettings CRC   ...OK <\r>
<\n><\0>308 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>312 : Info   : INIT : SPI not enabled<\r>
<\n>313 : Info   : Set Network mode: WiFi<\r>
<\n>418 : Info   : WIFI : Set WiFi to STA<\r>
<\n>ets Jun  8 2016 00:22:57<\r>
<\n><\r>
<\n>rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r>
<\n>configsip: 0, SPIWP:0xee<\r>
<\n>clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r>
<\n>mode:DIO, clock div:2<\r>
<\n>load:0x3fff0030,len:184<\r>
<\n>load:0x40078000,len:12736<\r>
<\n>load:0x40080400,len:2716<\r>
<\n>entry 0x40080594<\r>
<\n>ªU199 : Info   : <\n><\n><\r>
INIT : Booting version: ESP_Easy_mega_20231130_climate_ESP32solo1_4M316k, (GitHub Actions) mega-20231130 (ESP32 SDK 4.4.5.230614)<\r>
<\n>201 : Info   : INIT : Free RAM:296392<\r>
<\n>203 : Info   : INIT : Soft Reboot #305 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: CPU0: Software reset CPU CPU1: for APP CPU, reset by PRO CPU<\r>
<\n>204 : Info   : FS   : Mounting...<\r>
<\n>214 : Info   : FS   : Mount successful, used 141815 bytes of 290156<\r>
<\n>225 : Info   : CRC  : Settings CRC           ...OK<\r>
<\n>261 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>267 : Info   : CRC  : SecuritySettings CRC   ...OK <\r>
<\n><\0>308 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>312 : Info   : INIT : SPI not enabled<\r>
<\n>313 : Info   : Set Network mode: WiFi<\r>
<\n>418 : Info   : WIFI : Set WiFi to STA<\r>
<\n>ets Jun  8 2016 00:22:57<\r>
<\n><\r>
<\n>rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r>
<\n>configsip: 0, SPIWP:0xee<\r>
<\n>clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r>
<\n>mode:DIO, clock div:2<\r>
<\n>load:0x3fff0030,len:184<\r>
<\n>load:0x40078000,len:12736<\r>
<\n>load:0x40080400,len:2716<\r>
<\n>entry 0x40080594<\r>
<\n>ªU199 : Info   : <\n><\n><\r>
INIT : Booting version: ESP_Easy_mega_20231130_climate_ESP32solo1_4M316k, (GitHub Actions) mega-20231130 (ESP32 SDK 4.4.5.230614)<\r>
<\n>201 : Info   : INIT : Free RAM:296392<\r>
<\n>203 : Info   : INIT : Soft Reboot #306 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: CPU0: Software reset CPU CPU1: for APP CPU, reset by PRO CPU<\r>
<\n>204 : Info   : FS   : Mounting...<\r>
<\n>214 : Info   : FS   : Mount successful, used 141815 bytes of 290156<\r>
<\n>225 : Info   : CRC  : Settings CRC           ...OK<\r>
<\n>261 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>267 : Info   : CRC  : SecuritySettings CRC   ...OK <\r>
<\n><\0>308 : Info   : ESPEasy console using ESPEasySerial<\r>
<\n>312 : Info   : INIT : SPI not enabled<\r>
<\n>313 : Info   : Set Network mode: WiFi<\r>
<\n>418 : Info   : WIFI : Set WiFi to STA<\r>
<\n>ets Jun  8 2016 00:22:57<\r>
<\n><\r>
<\n>rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)<\r>
<\n>configsip: 0, SPIWP:0xee<\r>
<\n>clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<\r>
<\n>mode:DIO, clock div:2<\r>
<\n>load:0x3fff0030,len:184<\r>
<\n>load:0x40078000,len:12736<\r>
<\n>load:0x40080400,len:2716<\r>
<\n>entry 0x40080594<\r>
<\n>ªU199 : Info   : <\n><\n><\r>
INIT : Booting version

mpl1337, 2023-12-08:
i found the problem. My USB Port is broken. with each other can i flash the ESP. On one Port the ESP wont start after flash.

i have now running this one

Build: [ESP_Easy_mega_20231130_climate_ESP32solo1_4M316k Nov 30 2023](https://github.com/letscontrolit/ESPEasy/releases)

Thx

closed 2023-12-08 by mpl1337

TD-er, 2023-12-08:
Hmm what kind of board do you have that you only got a solo1 board running?
The solo1 build was only added as "fallback" whenever the manufacturer decided to go for squezing out absolutely the last penny and then some...
It runs at 160 MHz instead of 240 MHz and uses only a single core.
Also the flash speed is set at the lowest possible.

mpl1337, 2023-12-08:
the cheapest from amazon.

https://www.amazon.de/AZDelivery-NodeMCU-Development-Nachfolgermodell-ESP8266/dp/B074RGW2VQ/ref=sr_1_3_pp?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=S6TMI64OHAO3&keywords=esp32&qid=1702071297&sprefix=esp32%2Caps%2C106&sr=8-3

tonhuisman, 2023-12-08:
Hmm, you can have those 3 for nearly half the price at Aliexpress https://aliexpress.com/item/1005006267484374.html or https://aliexpress.com/item/32959541446.html🤔 maybe have to wait a little longer, but still > € 10,- so delivered within 12 days

TD-er, 2023-12-08:
Maybe also check the used USB cable and/or try another USB port.
Preferrably on a powered USB hub or an USB3 port directly on the PC as those tend to be a bit more 'beefy' for power delivery.