letscontrolit/ESPEasy #4909: T6963 plugin

Jirka10, 2023-12-11:
Hi, is it hard to make a plugin for 128x64 LCD display with paralel T6963 driver? It would help me a lot. Thank you.

tonhuisman, 2023-12-11:
Hm, these parallel interface displays aren't a very good match with ESP's, as these don't usually have as many [GPIO pins](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html) as Arduino boards seem to have (though ESP32 MCUs do have more GPIO pins available, it's still not often used).
We currently only support serial connected displays (I2C, SPI and Uart serial).

Jirka10, 2024-03-03:
Thanks for your help, this is understandable, I used an I2C converter for the display.

tonhuisman, 2024-03-03:
> I used an I2C converter for the display.

- Do you have a link available, for others interested in such a solution?
- What display protocol/emulation is used by that interface?