letscontrolit/ESPEasy #4913: UART speed converter

Vladimir-by, 2023-12-17:
There is a need to make a software UART speed converter. Pre-configured GPRS and LORA-868MHz modules with an operating speed of 9600 bit/s are pre-connected to the ESP32 microcontroller. The initial speed of the optical (infrared) reading device can be 300, 600, 1200, 2400, 9600 bit/s. How is it possible to convert UART speed using ESP32 and ESPEasy using two internal UART ports and software processing?

TD-er, 2023-12-17:
Right now you can't make such a conversion using existing plugins.

Also I don't know what you want to 'convert' here as the serial things you mentioned are not really compatible like you can copy data from one to the other without interpreting data.

Another issue with baud rate conversion is that you also need to buffer data. If the other end uses a lower baud rate, you also need to make sure the sending side will stop sending till you have flushed the data or else you have to drop data.

Vladimir-by, 2023-12-17:
It is clear that the implementation of the project will require at least an additional buffer and data control.

Vladimir-by, 2023-12-17:
Maybe the forum participants can suggest a ready-made solution, the use of specialized integrated circuits? As I understand it, all USB-UART interface converters (from FTDI, Silicon Labs, Prolific and others) operate on a similar frequency division principle. The specified bitrate is obtained by dividing the fixed reference frequency of the clock generator. Frequency division occurs through the use of multiple D-flip-flops.

tonhuisman, 2023-12-17:
What kind of data is to be transported over these GPRS and LORA connections? Both could be suitable to send out data from a sensor to a data-collection facility, but trying to build a bridge between both technologies doesn't seem very plausible.

Vladimir-by, 2023-12-17:
Data can be transmitted via several communication channels of your choice, based on the distance of the device Wi-Fi, bluetooth, GPRS, LORA-868MHz, all communication modules are configured at a speed of 9600 bit/s. The ESP-32 controller with ESPEasy firmware controls the inclusion of modules and switching of the UART multiplexer. The UART input connection can be at different speeds: 300, 600, 1200, 2400, 9600 bit/s. Reconfiguring each communication module on both the transmitting and receiving sides requires putting the modules into programming mode, launching service programs and additional time. Because of this, it became necessary to convert the speed as a faster method of coordination, if such a thing can be implemented.

TD-er, 2023-12-17:
OK, just put all questions about baudrates etc. aside as those are by far the least of your problems here.
Let's first make a few steps backwards and please explain why you think you could simply transfer data (without any processing) from IR, WiFi, Bluetooth, GPRS to LoRa?
It is absolutely not something you can simply pipe from one to the other.
You need to decode packets, maybe process data and recode packets to say the least.
Then if all of this is done, then the baudrate may be your next concern.

So please explain your use case what you want to do?

Vladimir-by, 2023-12-17:
Packets are decoded on the receiving equipment. Modules with an operating speed of 9600 bit/s Wi-Fi (Communication - Serial Server) or bluetooth or GPRS or LORA-868MHz are used to forward the IR port of energy meters. Manufacturers of meters have provided different speeds of interaction: 300, 600, 1200, 2400, 9600 bit/s.

Vladimir-by, 2023-12-17:
Interestingly, LORA E22-900T22D modules from Chengdu Ebyte Electronic Technology Co., Ltd. can operate at a connection speed of 9600 bit/s, and over a radio channel - for example, 2400 bit/s. They have their own operating algorithm and speed translation.
[image]

TD-er, 2023-12-17:
But still, even if you manage to get all your input and output devices to work at the same baudrate, you still cannot just forward the data without processing from one end to the other.

You still need to package the data into chunks, wrap them with some commands to tell the other device what to do with it etc.
It is not just a matter of converting baud rates, like you may do with voltages or other aspects of your low level data transmission.

chemmex, 2023-12-18:
E22-900T22D are serial LoRa modules with SX1262 and STM32 inside, they are not LoraWAN capable and can rather be used as UART extenders or modems. They are in fact very popular as longest range transceivers for drones and other UAVs. 
They have "transparent" point-to-point mode which is half-duplex and the data packet is limited to 200 bytes.
In all other modes the data should be framed with mode, address and probably checksum.
So, I think there is a chance that this module can be directly connected to the sensor or energy meter and even transmit some data assuming it is preconfigured with EBYTE's utility, and there is no need for ESPEasy in between.

TD-er, 2023-12-18:
But you still need quite a lot of buffering as LoRa is quite slow.
These modules might perhaps interface at 9600 baud, but the actual transfer rate is much, much slower.

And if these are compliant with local RF regulations, then you can only send 1% of the time (on some channels upto 10%)
So this leaves much less bandwidth (as in bytes/hour) for you to transfer data.

If you plan on connecting it to an energy meter, then you really have to filter or else you will end up with lots of broken packets as the input is just flooded with data.

chemmex, 2023-12-18:
That's correct, this is just a guess. I myself prefer an ESPEasy approach with sensors/actuators as plugins and data connections as controllers. 
This, again, returns us to recent forum discussion about controller template for async data transfer. For these particular modules there is a [library](https://github.com/xreef/EByte_LoRa_E22_Series_Library) which I sometimes used to upload sensor data using LoRa. But writing an ESPEasy controller is too big a challenge for me, so I would either ask for assistance or wait until somebody more proficient implements this

Vladimir-by, 2023-12-21:
The topic has quite a lot of criteria for implementation. Mainly the use of a buffer (which is important when transmitting from higher to lower speeds 9600->300 baud) and tracking data packets. Also, in industrial applications, it is necessary to achieve 100% packet transmission; errors in speed conversion are not acceptable. It is difficult to say how, in my case, to implement the task of quickly switching (matching) the speed, without changing the configuration of the configured modules (LORA-868 MHz E22-900T22D, GPRS E840-TTL-GPRS03 from Chengdu Ebyte Electronic Technology Co., Ltd.). If converting UART speed is interesting, there is a similar thread.
https://radiokot.ru/forum/viewtopic.php?t=17328

chemmex, 2023-12-22:
> https://radiokot.ru/forum/viewtopic.php?t=17328

I think that you are following the wrong path. Your use case is not about converting the UART speed, but about sending sensor data to the remote server using either LoRa or GPRS modules. 
However, it is very likely that you can use two  E22-900T22D modules as "serial extender" without ESP32, all you need is to setup the modules to match interface baud rates and radio parameters. Google for "E220-900T22D transparent mode".

For the GPRS module, there is no support for the devices controlled by AT commands yet