letscontrolit/ESPEasy #4917: [Test Build] Upcoming X-mas build, testers wanted

TD-er, 2023-12-23:
We're approaching Christmas and I know I mentioned there will be a X-mas release, but I'm not 100% sure it will be ready.
This all depends on the amount of critical bugs found by all of you.

Please test the [latest](https://td-er.nl/ESPEasy/latest/) build (same as [this GH Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/7304395764) ) and let me know what critical issues you may find.

The ESP32 builds with "LittleFS" are all running the latest ESP-IDF 5.1 code (upto commits of a few days ago from Espressif)
New platforms are ESP32-C2 and ESP32-C6.
Those only have a 'custom' and 'normal' build right now as I plan on adding a 2-step OTA for those so these can all have a small partition for the OTA and a large partition for the build. This concept allows for all 4M boards to run a "Max" build and thus will reduce the number of build files significantly.
This will later also be made available for all other ESP32-variants including some steps to migrate existing setups.

All ESP32-builds still running SPIFFS are based on ESP-IDF 4.4

I've been working on this migration to IDF5.x for over 2 months full time, as a lot had to be changed and upstream bugs had to be fixed for Arduino and ESP-IDF.
So I would be pleasantly surprised if no critical bugs were found.

If you plan on migrating a build using SPIFFS to one  using LittleFS, please keep in mind you will loose the settings.
So please backup your settings and restore them when making this migration.

There is also some extra control of persistent settings like WiFi credentials etc. which will be stored outside the file system.
So you could first update to a SPIFFS build, then check all the boxes on the "Factory Reset" page, save and wait for at least 5 minutes to let ESPEasy know the WiFi connection is considered stable. Then extra WiFI info will be stored in the NVS partition.
If you then migrate to a LittleFS build, your WiFi will be kept (and all other things checked on that factory reset page)

Dickesplumpel, 2023-12-23:
Esp8266-4M1M seems to work generaly perfect but I have trouble with dummy device or with taskvalueset, even if "taskvalueset to all plugins" is enabled taskvalueset to dummy device seems not working anymore.

fly74, 2023-12-23:
> Esp8266-4M1M seems to work generaly perfect but I have trouble with dummy device or with taskvalueset, even if "taskvalueset to all plugins" is enabled taskvalueset to dummy device seems not working anymore.

Would be important to me.

TD-er, 2023-12-23:
Hmm sounds like a bug which needs to be fixed before any release (and thus exactly the reason I asked for testing :) )
Happy to hear about this bug :)

TD-er, 2023-12-23:
That was easy to fix, or a stupid bug... ;)

When the [build](https://github.com/letscontrolit/ESPEasy/actions/runs/7309930058) is ready, I will upload it to the webflasher.

TD-er, 2023-12-23:
Latest build with fix is now uploaded to the web flasher.

Dickesplumpel, 2023-12-23:
Taskvalueset works! 👍 
Maybe tomorrow I have a closer look at ESP32-S2 and ESP32 builds.

TD-er, 2023-12-23:
There is still one issue I'm looking into right now and that's when calling `wifidisconnect`, the ESP sometimes reboots with a crash.
At least when using LittleFS builds on the ESP32-C6.

fly74, 2023-12-23:
> That was easy to fix, or a stupid bug... ;)
> 
> When the [build](https://github.com/letscontrolit/ESPEasy/actions/runs/7309930058) is ready, I will upload it to the webflasher.

Looks good in my landscape.

fly74, 2023-12-23:
Houston there is a problem:

I have a 
[image]

wich is filles remotely with MMddHHmm to compare in a rule, if the time is longer than 6 minutes away.

```
let,1,[D1#Dummy]+6

if [var#1]>%sysmonth%%sysday_0%%syshour_0%%sysmin_0%
...
else
```

in this release it always comes to the else path, even if D1 is not longer than 6 minutes away.

2nd if I try to change a rule, save does nothing. (MS Edge)

tonhuisman, 2023-12-23:
> 2nd if I try to change a rule, save does nothing. (MS Edge)

What build environment is that on? (as in: Works on my ESP32 machine 😛)

fly74, 2023-12-23:
> > 2nd if I try to change a rule, save does nothing. (MS Edge)
> 
> 
> 
> What build environment is that on? (as in: Works on my ESP32 machine 😛)

In MS Edge?

tonhuisman, 2023-12-23:
no, ESP8266, ESP32, etc. Normal, Collection, Display?

fly74, 2023-12-23:
```
<html>
<body>
<!--StartFragment-->
Firmware
--
Build:⋄ | 20231223 - Mega32
System Libraries:⋄ | ESP32 SDK 4.4.6
Git Build:⋄ | HEAD_40df5ee
Plugin Count:⋄ | 48 [Normal]
Build Origin: | GitHub Actions
Build Time:⋄ | Dec 23 2023 19:15:05
Binary Filename:⋄ | ESP_Easy_mega_20231223_normal_ESP32_4M316k
Build Platform:⋄ | Linux-6.2.0-1018-azure-x86_64-with-glibc2.35
Git HEAD:⋄ | HEAD_40df5ee
System Status
Syslog Log Level: | None
Serial Log Level: | None
Web Log Level: | None
Enable Serial Port Console: | true
Console Serial Port: | HW Serial0 RX:3 TX:1 @ 115201 baud
Network Services
Network Connected: | ✔
NTP Initialized: | ✔
ESP Board
ESP Chip ID: | 16169629 (0xF6BA9D)
ESP Chip Frequency: | 240 MHz
ESP Crystal Frequency: | 40 MHz
ESP APB Frequency: | 80 MHz
ESP Chip Model: | ESP32-D0WDQ6
ESP Chip Features: | Wi-Fi bgn / BLE
ESP Chip Revision: | 1.0
ESP Chip Cores: | 2
ESP Board Name: | Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS

<!--EndFragment-->
</body>
</html>
```

tonhuisman, 2023-12-23:
And what build did that still work on?, as it seems to be related to parsing the `if` condition, and that hasn't recently changed, AFAICS.
if I change your rule to:
```
Let,1,[D1#Dummy]+6
Let,2,%sysmonth%%sysday_0%
  If [var#1]>%v2%%syshour_0%%sysmin_0%
```
it works as intended, though I can see what you expect

fly74, 2023-12-23:
If I could save the rule I would test it.

tonhuisman, 2023-12-23:
Can you hard-refresh the browser page, so it can re-fetch the js code?

tonhuisman, 2023-12-23:
or watch the log in a second browser tab (or serial log) to see if it throws an error when saving the rules file?

TD-er, 2023-12-23:
What kind of "sensor type" have you used for the dummy?
This looks like the float resolution, which is essentially 6 decimals and you're using 8 decimals here.
This is exactly why I added other integer types to the dummy.

fly74, 2023-12-23:
> Can you hard-refresh the browser page, so it can re-fetch the js code?

Firefox don't work too. But download and edit and upload works.
@tonhuisman  The rulechange works. Thx.

So only the save would be interesting.

Browserdebug:
[image]

fly74, 2023-12-23:
> What kind of "sensor type" have you used for the dummy? This looks like the float resolution, which is essentially 6 decimals and you're using 8 decimals here. This is exactly why I added other integer types to the dummy.

[image]

TD-er, 2023-12-23:
What I meant is this selection on the Dumy task:
[image]

You're referring to the value of the dummy task value, which is by default a float.  (Single/dual/quad etc are all float)

fly74, 2023-12-23:
> What I meant is this selection on the Dumy task: [...]

But Ton's rulechange works with single :/

tonhuisman, 2023-12-23:
Seems to be a parsing issue, as I got an error with the original rule that didn't have the `%sysmonth%` part replaced

TD-er, 2023-12-23:
> But Ton's rulechange works with single :/

Yep, it looks like he is splitting the value into separate floats.
But depending how the value is being set via remote calls, this will still fail on new updates.

If you set the new value via an event value, then you can split it with enough resolution just fine.
But if it is stored in the dummy as a single value, then a new update may not appear to change the task value.

fly74, 2023-12-23:
> > But Ton's rulechange works with single :/
> 
> Yep, it looks like he is splitting the value into separate floats. But depending how the value is being set via remote calls, this will still fail on new updates.
> 
> If you set the new value via an event value, then you can split it with enough resolution just fine. But if it is stored in the dummy as a single value, then a new update may not appear to change the task value.
> 
> Edit: This has always been a problem, that's exactly why I added these other options to store int values and even int64 types if really needed.
> 
> Edit2: Any integer value outside the range of -8388608 ... 8388607 will be stored with reduced resolution as a float simply can't represent those values without loss of resolution.

@TD-er I can't recognize problem. The 8 digit are set remotely to
[image]
to single.

TD-er, 2023-12-23:
Hmm tested here and indeed, you're still in the proper range.
Just looked [here](https://en.wikipedia.org/wiki/Single-precision_floating-point_format#Precision_limitations_on_decimal_values_(between_1_and_16777216) and it seems I was off by a factor of 2 for the range.
Strange as I thought it had a range of 23 bits, but apparently it allows upto 2^24 -1.

TD-er, 2023-12-23:
Just had a chat with Ton, apparently 3 variables replaced in a single line seems to work, but 4 doesn't work.
Have to dive into this a bit further...
So let me get a fresh beer and have a look :)

fly74, 2023-12-23:
> Hmm tested here and indeed, you're still in the proper range. Just looked [...]

[image]

> Hmm tested here and indeed, you're still in the proper range. Just looked [...]

```
Precision limitations on integer values
Integers between 0 and 16777216 can be exactly represented (also applies for negative integers between −16777216 and 0)
```
That makes it clear. Int is enough to store 8 digit when the left 2 is the month 0..12.

fly74, 2023-12-23:
> Just had a chat with Ton, apparently 3 variables replaced in a single line seems to work, but 4 doesn't work. Have to dive into this a bit further... So let me get a fresh beer and have a look :)

But worked in the last official.

TD-er, 2023-12-23:
> But worked in the last official.

Never underestimate my super powers to introduce sneaky bugs.
But you guys managed to find them within hours and it probably took me a lot more to create these bugs. ;)

chromoxdor, 2023-12-23:
Ohhh.. here is a testing party going on... i almost missed it. 🙂

fly74, 2023-12-23:
> Ohhh.. here is a testing party going on... i almost missed it. 🙂

@chromoxdor It's not to late come in.

TD-er, 2023-12-23:
I expect there will be some bugs left for you to find and report :)
@fly74
I think I found it... have to test it 
What was the exact line you used with the variables?
```
%sysmonth%%sysday_0%%syshour_0%%sysmin_0%
``` 
??

chromoxdor, 2023-12-23:
> @chromoxdor It's not to late come in.
Thanks! But first things first. Where do i find the buffet?

> I expect there will be some bugs left for you to find and report :)
I hope i find these before the release...

fly74, 2023-12-23:
> I expect there will be some bugs left for you to find and report :)
> 
> @fly74
> 
> I think I found it... have to test it 
> 
> What was the exact line you used with the variables?
> 
> ```
> 
> %sysmonth%%sysday_0%%syshour_0%%sysmin_0%
> 
> ``` 
> 
> ??

As above 👆. Before Ton fixed it.

chromoxdor, 2023-12-23:
Almost all notes have no proper formatting
[image]

TD-er, 2023-12-23:
If I place the variables in a different order, they will be replaced without any error:

```
logentry,"%sysday_0%%syshour_0%%sysmin_0%%sysmonth%"

03:40:23.464 : (305112) Info   : HTTP: logentry,"%sysday_0%%syshour_0%%sysmin_0%%sysmonth%"
03:40:23.471 : (305128) Debug  : HTTP after parseTemplate: logentry,"23232412"
03:40:23.475 : (305096) Debug  : Command: logentry
03:40:23.476 : (305096) Debug  : logentry,"23232412"
```

Here the order you used:
```
03:40:01.958 : (312032) Info   : HTTP: logentry,"%sysmonth%%sysday_0%%syshour_0%%sysmin_0%"
03:40:01.963 : (312032) Debug  : HTTP after parseTemplate: logentry,"%sysmonth%232324"
03:40:01.968 : (312016) Debug  : Command: logentry
03:40:01.969 : (312016) Debug  : logentry,"%sysmonth%232324"
03:40:01.969 : (312000) Debug  : Par1: 12232324 Par2: 0 Par3: 0 Par4: 0 Par5: 0
03:40:01.970 : (312016) Info   : 12232324
```

So it seems like the order makes a difference here, which it obviously shouldn't.

N.B. the reason why this could now be broken is that I have also added quite some improvements in parsing speed and reduction of the bin size.
So a lot (!!) of code has changed.

chromoxdor, 2023-12-23:
Ok...i need to sleep. So far it looks good here except the notes.
@TD-er how much time do i have for testing? I guess you want to release it tomorrow?

TD-er, 2023-12-23:
Well if it isn't done, it isn't going to be released.
And here in the Netherlands we have 2 Christmas days, so even the 26th is still Christmas :)

TD-er, 2023-12-24:
OK these are all fixed now.
See GH Actions build: https://github.com/letscontrolit/ESPEasy/actions/runs/7311657637

I'm not going to wait till 3:30 am to upload these to the web flasher.
So please use the GH Actions build for testing :)

TD-er, 2023-12-24:
OK, found 1 possible other issue which I will look into tomorrow... (well after sleeping...)
The `config` command might not work right now or if it does we can speed up command parsing even more and reduce build size along the way. :)

fly74, 2023-12-24:
Anyway merry Christmas!

chemmex, 2023-12-24:
Not sure if current code is affected, but it seems that latest ESP32x custom builds do not apply `#define DEFAULT_NAME` from Custom.h 
Instead, the name is set to ESPEasy-  which is broadcasted as AP SSID. ESP8266 builds are Ok

Dickesplumpel, 2023-12-24:
Nothing to test - build failed!

chemmex, 2023-12-24:
> Nothing to test - build failed!

There are at least ESP32 images in the binaries.zip

TD-er, 2023-12-24:
Just added a quick fix, hopefully this is enough to complete the build

Dickesplumpel, 2023-12-24:
For me it makes no sense to test when build failed, I wait for next build.

TD-er, 2023-12-24:
I have no idea why it doesn't trigger a new build, I will try pushing the commit from my main computer...

iz8mbw, 2023-12-24:
Hi guys. Sorry but for now I cannot give my contribution for testing. I'll try to do my best in next days (if I'm still in time).

TD-er, 2023-12-24:
No idea why the GH Actions build didn't start, but now it is running...
https://github.com/letscontrolit/ESPEasy/actions/runs/7314499893

TD-er, 2023-12-24:
Updated the web flasher with the latest build.

Dickesplumpel, 2023-12-24:
In my case everything okay with ESP8266-4M1M climate and normal, ESP32s2 climate, ESP32-4M and ESP32Solo1 so far.

fly74, 2023-12-24:
> No idea why the GH Actions build didn't start, but now it is running... https://github.com/letscontrolit/ESPEasy/actions/runs/7314499893

There is an issue:

```
On SW#State do
      Event,pswitch
endon

On pswitch do
   if [D1#Dummy]=1
     Event,poff
   else
     Event,pon
   endif
endon

On pon do
   neopixelall,255,255,255
   TaskValueSet 3,1,1
endon

On poff do
   neopixelall,0,0,0
   TaskValueSet 3,1,0
endon
```

don't switch static anymore ?!?!
When the SW is firing it is static. But remotely it do both events on/off or off/on depending on state before (flicker short).

[image]

TD-er, 2023-12-24:
Not 100% sure I get what you're trying to tell here.

I assume `[D1#Dummy]` is task 3, taskvalue 1?

What remote event are you sending? `event,pswitch` ?
How is the switch input `SW` being toggled, since you don't have any debounce configured?
What is the value of `[D1#Dummy]`  when you send the remote event?

N.B. Rules parsing and event handling is now quite a bit faster, so it is very well possible you're seeing faster response when the input switch task does receive a number of pulses.

Better have the debounce set to something like 100 msec when you actually use an actual button and not a logical output of another electronic device or other microcontroller.

chromoxdor, 2023-12-24:
I am testing with ESP8266-4M1M normal and ESP32 energy. No issues so far…

TD-er, 2023-12-24:
I think I see what you mean...
Tested here on an ESP32-S2 (with LittleFS and thus IDF5.1 code)

Sending the event via the web page:

```
575752: HTTP: Event,pswitch
575758: EVENT: pswitch
575786: ACT : Event,poff
575790: EVENT: poff
575799: ACT : neopixelall,0,0,0
575806: Command unknown: neopixelall,0,0,0
575809: ACT : TaskValueSet 3,1,0
575897: EVENT: pswitch
575913: ACT : Event,pon
575916: EVENT: pon
575924: ACT : neopixelall,255,255,255
575930: Command unknown: neopixelall,255,255,255
575933: ACT : TaskValueSet 3,1,1
```

TD-er, 2023-12-24:
Found the bug, I introduced it yesterday while fixing your previous bug...

TD-er, 2023-12-24:
Will be fixed in this GH Actions build:
https://github.com/letscontrolit/ESPEasy/actions/runs/7317175703

Keep them reports coming @fly74 
Really truly appreciated!

fly74, 2023-12-24:
> Not 100% sure I get what you're trying to tell here.

Was difficult to describe ;)

fly74, 2023-12-24:
> Will be fixed in this GH Actions build:
> https://github.com/letscontrolit/ESPEasy/actions/runs/7317175703

What is the build time actually? 1h?

[61 more replies]

closed 2023-12-25 by TD-er