letscontrolit/ESPEasy #4920: [BUG] [P012] Display LCD2004 in Line 1 show ESP Easy if it is blank

iz8mbw, 2023-12-26:
Hello.
In LCD2004 display if the Line 1 (first line) is blank (not valorized) the display show "ESP Easy" text instead to show nothing.

tonhuisman, 2023-12-26:
The code doesn't replace the current content for an empty template line, and hasn't been changed for over 3 years (and probably has worked this way much longer), so it's not a _new_ bug...

iz8mbw, 2023-12-26:
Users expect this text to be displayed can simply write this text. Users want to have the first line blank can simply leave the field blank.

tonhuisman, 2023-12-26:
The actual reason that empty template lines are skipped is that you can use the `lcd` command to put custom/external content on an empty template-line and standard text or variable content using the template.

iz8mbw, 2023-12-26:
Ton but if I leave blank the second or the third or the fourth lines it works as expected, so lines remain blank.
The bug is only for the first line.

tonhuisman, 2023-12-26:
Only the first line is used to display the splash text 🙈 
I'll add some code to initially clear the display, but the ESPEasy splash text may then be displayed only very briefly.

TD-er, 2023-12-26:
Just check if this "ESPEasy" string is still present on the screen and then completely clear the line if someone wants to write to it.
Why wouldn't  anyone have "ESPEasy" being displayed like forever? ;)

iz8mbw, 2023-12-26:
When I put some text on the first line, it display the text I wrote.
When I clear the field, it display "ESP Easy".

iz8mbw, 2023-12-26:
When the board boot I see the splash text "ESP Easy" (only first line with this text). After the lines are correctly valorized except the first line if/when the filed is blank.

tonhuisman, 2023-12-26:
Yes, the issue is clear, it's just that it is also (at least partly) by design, except that the design didn't take the splash into account 😉 
As said, the intent of blank lines in the template is that they are (usually) filled from rules or via http commands, then there is no issue...

iz8mbw, 2023-12-26:
Oh yes, if I valorize the field with text or via rules the issue disappeared. 😀

tonhuisman, 2024-07-03:
@iz8mbw This can be closed as it has been resolved via #5043

iz8mbw, 2024-07-03:
Yes, sorry if I miss to close it.

iz8mbw, 2024-07-03:
https://github.com/letscontrolit/ESPEasy/pull/5043

closed 2024-07-03 by iz8mbw

tonhuisman, 2024-07-03:
> Yes, sorry if I miss to close it.

Well, I could have auto-closed it when the PR was merged if properly annotated, but I didn't, so it's most likely _my_ mistake 😄