letscontrolit/ESPEasy #4923: [BUG] stall

IW2FAG-11, 2023-12-27:
I have many esp8266 device.
I have many devices based on ESP8266. I have upgraded two devices, both with Display - OLED SSD1306, in one there is also an ADC ADS1115.
By upgrading from ESP_Easy_mega_20230822_energy_ESP8266_4M1M version to ESP_Easy_mega_20231225_energy_ESP8266_4M1M version, the one without ADC r<and flash size of 16M estarted perfectly. Upgrading the second one, the one that has the ADC connected and configured,< with 4M flash, switching from ESP_Easy_mega_20230822_normal_ESP8266_4M1M to ESP_Easy_mega_20231225_normal_ESP8266_4M1M shows on the display the time that seems to advance slowed down by 30 times, basically the time on the display advances by one second every thirty or so, it does not connect to WiFi. It seems slowed down and unresponsive..

TD-er, 2023-12-27:
Hmm this sounds like the ADS1115 is maybe not a real one.
Not sure when, but a few months ago I did investigate those ADS1x15 chips and made some changes to the code as there was a very annoying bug where the wrong pin was sampled.

When changing this, another issue came to light that you can almost be 100% sure you're not getting what you order when buying an ADS1x15 from sites like Ali Express.

Can you via the serial port disable the ADS1x15 task to see if the ESP will work fine?
```
taskdisable,N
save
```
With `N` being the task index of the ADS1x15 task.

I don't think the ESP itself is running slow, but probably the events used to update the display will be handled much more slowly.
And if the event queue is being filled, the ESP will probably run out of memory and crash/reboot.
If the time hasn't been updated in between the ESP may continuously load the last known time from RTC memory and only occasionally update the time, which makes it appear to run slow.

IW2FAG-11, 2023-12-27:
I have to disassemble the device to get to the serial, I will still have to do that in order to get back to at least the previously used FW.
The ADS1115s that is in use by alemo a couple of years in the power meter I built myself, I think I bought them through ebay in a German store but I think they were from Aliexpress. They looked the same as the ones I got a few weeks ago.
As soon as I disassemble I will try to give the command to disable the task.
Thanks

IW2FAG-11, 2023-12-27:
I loaded the configuration file I was using in the one that after the upgrade no longer connects on the other one that was still working after the upgrade. In response it stopped working by stalling as well.
I tried now disttivating two tasks pointing to the ADC, having the serial connection available there, then the esp started working again. If from serial I try to reactivate the two tasks, they do not reactivate not having the ADS device physically connected to the esp. I think that is why the tasks do not reactivate.

TD-er, 2023-12-27:
Did you power cycle the node after you found the node to be stalling?

IW2FAG-11, 2023-12-27:
Yes several times. without solving the problem.
In the device with 16M Flash that did not have ADS1115 connected and was working after the update, after loading the configuration file of the device that had stopped working, it too had stopped working despite repeated HARD-RESET cycles (powerUP cycle).
Then I gave from serial the command taskdisable,x to the two tasks that were programmed to use ADC even though not physically present, and the device after soft reboot started to go,
Now I connected an ADC1115 and reloaded the original config with the tasks enabled and the device continues to go.

Is it not by chance some information is stored in config flash that with the new FW alters the operation of the task with the new release and it "unlocks" only after a taskdisable -... taskenable forced ?

One symptom when it does not work is that it does not connect WiFi.

TD-er, 2023-12-27:
> Is it not by chance some information is stored in config flash that with the new FW alters the operation of the task with the new release and it "unlocks" only after a taskdisable -... taskenable forced ?

It shouldn't.
Only thing I can imagine is that there is something not working well in your setup while the ESP also needs to connect to Wifi.
Like I said, is your ESP perhaps in some boot-loop?
You might also see the onboard LED flashing at slightly over 1 Hz if the ESP is indeed in some crash/boot loop. (if there is some LED connected to GPIO-2)

IW2FAG-11, 2023-12-27:
[image]
At present moment is not possible to wiev anything.
Tomorrow morning i'll open case, solder some cable on serial to capture data.
Anyway, before upgrading from autostop release to present one, device was working for two year with same config and rules to measure mine home production/consumption energy.

IW2FAG-11, 2023-12-28:
[By attaching the serial to the stalled device, when I power up I get what is visible in the log I sent in the LOG file.
I think I disabled both the serial and the LED, since the device had never needed debugging for at least two years anyway.
So I guess there is not enough information to do anything.
Unless there is a command to re-enable the serial from the console.

teraterm.log](https://github.com/letscontrolit/ESPEasy/files/13784255/teraterm.log)

IW2FAG-11, 2023-12-28:
After reinstalling the 20230822 version, I reconnected to the web server, disabled tasks 1,2,3 calling ADCs, re-enabled the serial.

Then I reinstalled the 20231225 version from the web and again everything crashes. Now, however, it is clear that the problem is in the WiFi connection.

[Uploading teraterm.log…]()

TD-er, 2023-12-28:
I don't see any logs attached to your post.

IW2FAG-11, 2023-12-28:
[Uploading teraterm.zip…]()

IW2FAG-11, 2023-12-28:
Ä                                                                                                                                                                           
dì                                                                                                                                                                          
c<ìÛc$                                                                                                                                                                      
#                                                                                                                                                                           
ónglgnãì                                                                                                                                                                    
cp                                                                                                                                                                          
dc$sdpûgà                                                                                                                                                                   
l                                                                                                                                                                           
Ü                                                                                                                                                                           
c                                                                                                                                                                           
nâ|ìl?                                                                                                                                                                      
?c                                                                                                                                                                          
ûg'çl?d`'                                                                                                                                                                   
l`or?'                                                                                                                                                                      
                                                                                                                                                                            
cd`{                                                                                                                                                                        
bl`b                                                                                                                                                                        
Çd;`üÃgªU00.620 : (29536) Info   :                                                                                                                                          
                                                                                                                                                                            
INIT : Booting version: ESP_Easy_mega_20231225_normal_ESP8266_4M1M, (GitHub Actions) mega-20231225 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA su
pport)                                                                                                                                                                      
00.622 : (29400) Info   : INIT : Free RAM:29400                                                                                                                             
00.624 : (29344) Info   : INIT : Cold Boot - Restart Reason: Power On                                                                                                       
00.625 : (29304) Info   : FS   : MountinçÉùed!                                                                                                                              
00.673 : (26856) Info   : ESPEasy console using ESPEasySerial                                                                                                               
00.714 : (26792) Info   : INIT : I2C                                                                                                                                        
00.715 : (26752) Info   : INIT : SPI not enabled                                                                                                                            
00.716 : (26680) Info   : Set Network mode: WiFi                                                                                                                            
00.821 : (27320) Info   : WIFI : Set WiFi to STA                                                                                                                            
03.112 : (26920) Info   : ESPEasy console using ESPEasySerial                                                                                                               
03.113 : (26880) Info   : INIT : Free RAM:26880                                                                                                                             
[... cut ...]
30.881 : (20840) Info   : WIFI : Connecting IW2FAG_G 0C:B6:D2:26:C5:B0 Ch:6 (-47dBm)WPA2/PSK attempt #6                                                                     
30.898 : (20984) Info   : EVENT: WiFi#Disconnected                                                                                                                          
30.923 : (21048) Info   : EVENT: WiFi#Disconnected                                                                                                                          
31.837 : (21112) Info   : EVENT: WiFi#Disconnected

IW2FAG-11, 2023-12-28:
Looks like having connection problem when i upload mine config file.

TD-er, 2023-12-28:
Can you try the command `clearwifirfcal` ?

TD-er, 2023-12-28:
Is this a hidden SSID by the way?

IW2FAG-11, 2023-12-28:
I see that after reboot device reconnect to mine wifi spot.  Is ok ?

IW2FAG-11, 2023-12-28:
00:09:25.991 : (21016) Info   : ADS1x15: RAW value: 32, output value: 32.000000
00:09:34.637 : (20936) Info   : WD   : Uptime 10  ConnectFailures 0 FreeMem 21072 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
clearwifirfcal
>clearwifirfcal
Cleared RFcal partition. Please reboot!
00:09:42.430 : (20808) Info   : ADS1x15: RAW value: 2912, output value: 2912.000000
reboot
>reboot
00:09:46.730 : (21712) Info   : WIFI : Disconnected! Reason: '(8) Assoc leave'
00:09:46.832 : (21744) Info   : Reset WiFi.
00:09:46.934 : (21728) Info   : WIFI : Set WiFi to OFF

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
ªU00.757 : (29536) Info   :

INIT : Booting version: ESP_Easy_mega_20231225_normal_ESP8266_4M1M, (GitHub Actions) mega-20231225 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
00.759 : (29400) Info   : INIT : Free RAM:29400
00.761 : (29328) Info   : INIT : Soft Reboot #9 - Restart Reason: Software/System restart
00.762 : (29288) Info   : FS   : Mounting...
00.788 : (27040) Info   : FS   : Mount successful, used 75802 bytes of 957314
00.811 : (26840) Info   : ESPEasy console using ESPEasySerial
00.853 : (26840) Info   : INIT : I2C
00.854 : (26800) Info   : INIT : SPI not enabled
00.855 : (26760) Info   : Set Network mode: WiFi
00.960 : (27440) Info   : WIFI : Set WiFi to STA
03.251 : (27104) Info   : ESPEasy console using ESPEasySerial
03.252 : (27064) Info   : INIT : Free RAM:27064
03.262 : (26768) Info   : ESPEasy console using ESPEasySerial
03.264 : (26632) Info   : INFO : Plugins: 47 [Normal][No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
03.267 : (26184) Info   : WIFI : Connecting IW2FAG_G 0C:B6:D2:26:C5:B0 Ch:6 (-50dBm)WPA2/PSK attempt #0
03.277 : (23608) Info   : Webserver: start
04.283 : (22536) Info   : ADS1x15: RAW value: 3024, output value: 3024.000000
04.345 : (22448) Info   : ADS1x15: RAW value: 32, output value: 32.000000
04.657 : (22304) Info   : WD   : Uptime 0  ConnectFailures 0 FreeMem 22440 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
05.281 : (22120) Info   : WIFI : DHCP IP: 192.168.0.51 (ESP-Power-7) GW: 192.168.0.1 SN: 255.255.255.0 DNS: 151.5.216.150 / 151.5.216.15   duration: 1037 ms
05.295 : (22232) Info   : firstLoopConnectionsEstablished
05.298 : (22184) Info   : WIFI : Connected! AP: IW2FAG_G (0C:B6:D2:26:C5:B0) Ch: 6 Duration: 967 ms
05.299 : (22288) Info   : Webserver: stop
05.357 : (21912) Info   : NTP  : NTP replied: delay 20 mSec Accuracy increased by 670 msec
05.359 : (22272) Info   : Time set to 1703754641.671
05.361 : (22176) Info   : Current Time Zone:  DST time start: 2023-03-26 02:00:00 offset: 120 min STD time start: 2023-10-29 03:00:00 offset: 60 min
05.366 : (22232) Info   : Local time: 2023-12-28 10:10:41
05.368 : (22232) Info   : Webserver: start
34.627 : (21632) Info   : WD   : Uptime 1  ConnectFailures 0 FreeMem 21768 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
00:01:03.360 : (21424) Info   : ADS1x15: RAW value: 2928, output value: 2928.000000
00:01:03.399 : (21440) Info   : ADS1x15: RAW value: 32, output value: 32.000000
00:01:04.625 : (21440) Info   : WD   : Uptime 1  ConnectFailures 0 FreeMem 21576 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init

IW2FAG-11, 2023-12-28:
Hidden SSID Slow Connect:    is enabled after restart looking advanced setup

IW2FAG-11, 2023-12-28:
For testing, i anabled the switch force B/G ande rebooting mine.
Now device have problem on connection.

IW2FAG-11, 2023-12-28:
I did more testing and confirm that if I activate "force B/G" as I always had in my configs to avoid N mode, the WiFi connection becomes unstable and no longer connects on reboot.

Is there any way to edit my config.dat and remove the "force B/G" switch without having to redo the whole device configuration by hand ?

TD-er, 2023-12-28:
Hmm I think I know what the issue is you're experiencing.
There was a bug where the "force B/G" was essentially inverted.
So it seems like it is now working and your access point is blocking "802.11g" clients.

After 10 failed connect attempts it should ignore this flag and retry using the 'n' option.

TD-er, 2023-12-28:
As a fallback you can also configure an access point on your mobile and configure ESPEasy to connect to it (or any other access point)
To keep your current WiFi setting, you can use `wifissid2` and `wifikey2` instead of `wifissid` and `wifikey` to set your WiFi credentials.

```
wifissid2,MyWiFiSSID
wifikey2,MySecretPassword
save
wifidisconnect
```
Or `reboot` instead of `wifidisconnect`

IW2FAG-11, 2023-12-28:
In all my devices I have always used B/G forcing to avoid network N mode.
Until the 20230822 version, I never encountered any problems starting. I
have always forced not to use N mode as the latter is less tolerant to
electrical disturbances. I have two homes and in both I use access poi t
with dual SSID Notmal and foldback. So far I have never experienced any
connection problems. Same network same device.... new release and problems
arise. Ok I uncheck B/G to make it work. My problem is that I can NOT do
serial updates on all devices, I would have to disassemble the system. That
is why I usually have copies of the devices I try to update before I try on
the ones actually connected to the system. Basically if I want to use the
new version, first I would have to uncheck the B/G mode and then do OTA
upgrade of the devices. Thank you. p.s. I would highlight these "issues"
well in the releases. Thanks anyway for the great work you are doing.

TD-er, 2023-12-28:
I think the 'fallback' options should be more user friendly.
Apparently your unit did have multiple issues or else it would have used the fallback I mentioned where it would eventually ignore the force b/g if it fails to connect.

On builds with the SDK 2.7.4 on ESP8266 I cannot query the access points on allowed protocols.
On SDK 3.x.x builds I do query the access point for its allowed protocols, and thus ignore "force b/g" when it would result in an unusable connection.

Anyway, I will look into the code to see if I might have messed up somewhere and by adding this optional ignore for SDK3.x perhaps have broken the existing fallback again.

There is also another issue with NeoPixel plugins on ESP8266 in the X-mas build, so maybe there will be a bugfix release this year to deal with these issues.

IW2FAG-11, 2023-12-28:
Thanks.
At present moment I remove forcing B/G on mine vevice prior upgrading.

IW2FAG-11, 2024-03-04:
I updated a few days ago some ESP-12Fs with the 20240229 release and the problem of WiFi forcing in B/G mode persists causing the ESP to stall where it actually keeps trying to connect in G mode, dicosnnette, turns off the wifi, turns WiFi back on and then starts the loop again. I did some more testing. 
I went to another home where two APs exist. What happens I can define in these terms. When I try to connect the SSID of the older ASUS DSL-N14U router that has the 2.4G band only it works both by forcing B/G and by letting the connection decide that it is in fact configured as N. 
Whereas if I try to force B/G connection with router DVA-5592_A1_WI_20191219 to its SSID I get the same result I see on another dual band router. The new router accepts the connection but after 1 or 2 seconds disconnects.
I should mention that I have always disabled Band Stering so that I can configure the two bands independently, one band I use for home automation (2.4G) and the other band I use for multimedia devices.  The strange thing is that if I use old firmware released before August, even the newer modems would accept B/G connections. Basically it would switch to the G connection also identifiable by the maximum power you see going up to 17.0 dBm, in the advanced function box.

Other note, in the two homes I also have ZigBee devices, obviously the APs have configurations such that there is no ovrlapping of channels on the 2.4G bands either between the APs or with the ZigBee channel. 
Newer APs have freedom to select HT20 or HT40 channels having left the yes Auto option.
I have loaded Platformio and the various additional tools indicated. 
The copilations of the released codes with the tag does them.
If you need to do cmpilation tests I am available.
I use the esp8266_normal_4M1M bin on almost all devices, I tried to generate the compilation trying to use a newer core, 3.1.2 which I was already seeing employed elsewhere. Unfortunately, it fails on the serial drivers. As soon as I have some time I'll promise to look at it again.
Thanks again for this wonderful project.

TD-er, 2024-03-05:
Can you check on the access point and on the ESPEasy sysinfo page that the actual mode (802.11n or 802.11b/802.11g) is used according to what is being set in ESPEasy (tools->Advanced page Force B/G mode) ?
There is of course some fallback if the AP doesn't support b/g while set to use it.
And on 3.1.2 code, I can actually see what is being advertised by the access point and decide to ignore the "b/g" forcing if the access point doesn't advertise it as an option.

As far as I can remember, that's the only change in the past 6 months for ESP8266 regarding WiFi.

TD-er, 2024-03-10:
Can you check the [GH Actions test build](https://github.com/letscontrolit/ESPEasy/actions/runs/8225625268) which is now running for this PR: https://github.com/letscontrolit/ESPEasy/pull/5004

Your issues might be related to some weird behavior on ESP8266 regarding how union types are handled.

IW2FAG-11, 2024-03-11:
Hi, I'll try sooner.
I've had some problem.

IW2FAG-11, 2024-03-11:
I just tried to load the indicated version.
Without checking the B/G switch.
Newer Router accepts the connection and then immediately disconnects.
The older Asus Router accepts the connection and indicates....
MAC RSSI PSM PhyMode BW MCS SGI STBC Rate Connect Time.
B4:E6:2D:6D:61:DE -66dBm Yes OFDM 20M 6 NO NO 48M 00:05:06
Reasonably it is in n as also indicated in the ESP8266 device
WiFi Connection: 802.11n (RSSI -53 dBm)
But the connection from the newer router is not accepted.
Forcing in B/G I observe that the indicated pMax goes from 14 to 17dBm,
WiFi Connection: 802.11g (RSSI -52 dBm)
 On the older router nothing changes.
MAC RSSI PSM PhyMode BW MCS SGI STBC Rate Connect Time
B4:E6:2D:6D:61:DE -65dBm Yes OFDM 20M 6 NO NO 48M 00:25:08

With 20240229 release tag, when NOT in B/G, pmax is 14dBm and old router indicates.
MAC RSSI PSM PhyMode BW MCS SGI STBC Rate Connect Time
B4:E6:2D:6D:61:DE -64dBm Yes HTMIX 20M 7 NO Yes 65M 00:00:16
In this mode my New router is also able to connect.

So I think the change made in the temporary version makes the situation worse.

I am looking for information as at this point I believe it is some new routers that do not accept mixed connections, i.e. G and N at the same time if only one other device requires n.

20240229 Tag is better than the trial version you submitted.

TD-er, 2024-03-11:
The change in max TX power is exactly as expected as it can only send at max. 14 dBm on 802.11n and max. 17 dBm on 802.11g (and 20 dBm on 802.11b, but please do not use that on modern routers/AP)

With Force B/G unchecked, it defaults to "n" mode.
However on (newer) routers with "band steering" enabled, your router will disconnect any client to let it reconnect on either another AP which may be closer to the client (or less busy) or to force it to use 5 GHz WiFi.

If you advertise you only can handle "b" or "g" mode, the router will not try to do these kinds of disconnects.
However if your router is set to "N-only" as is the default on quite a few newer routers/APs, then ESPEasy may not be able to connect to it in "b" or "g" mode.
For ESP8266 builds based on SDK3.x ("beta" or with "312" in the name) I can see what the router advertises as options and if "b" or "g" is not listed, then I just ignore the "force B/G mode" flag and just connect to the AP using "n" mode.

However on SDK2.7.4 builds I don't have this info and thus I need to retry and eventually switch to the fallback option.

Now the bug I tried to fix.
In order to decide whether I can try to connect using either "g" or "n", I check the reported flags by the AP.
Since I don't know this info on SDK2.7.4 builds, I have to set those explicitly to "false". And that apparently was not always done.
Thus it was at random whether an ESP8266 could connect to WiFi on an AP which has limited the options.
If your AP allows all (b/g/n), then there is no problem.

Not sure why you conclude this build is worse than the previous one though.

TD-er, 2024-03-11:
By the way, I just updated the build on the webflasher: https://td-er.nl/ESPEasy/latest/
It now also contains a possible fix for ADS1x15 as this was also using a `union` to interact with register values.
The way how we used `union` types (to make conversions) is not working well on ESP8266.
So I'm now looking into every use of `union` to see if it can be done differently.

PR #5004 (2024-03-21): [ESP8266 WiFi] Initialize flags for AP capabilities
May fix issues where the ESP may not properly (re)connect  to an AP

Fixes: #4996
Fixes: #4923
closed 2024-03-21 by TD-er