letscontrolit/ESPEasy #4929: Happy new year!

iz8mbw, 2023-12-31:
Just to say HAPPY NEW YEAR to all of you guys.
Many thanks for your effort.to improve this great project.
All the best to you and your families.

Ciao!

@TD-er @tonhuisman still thanks

tonhuisman, 2024-01-01:
Happy New Year, and all the best to you and your closest 😃

TD-er, 2024-01-01:
Happy new year to 'ya all :)

ghtester, 2024-01-02:
And thanks a lot to all developers & contributors for their hard and great work on ESP Easy during the last year! :-)

fly74, 2024-01-14:
Oh missed this. Happy new year to all. Let's keep it this way.

TD-er, 2024-01-14:
> Oh missed this. Happy new year to all. Let's keep it this way.

Well eventually 2024 will no longer be considered 'new' I guess :)

iz8mbw, 2024-01-14:
Yes.
Going to close this!

closed 2024-01-14 by iz8mbw