letscontrolit/ESPEasy #4931: SoftAp not start and can't config[BUG]

mehdi-song, 2024-01-05:
Hi

ESP-WROOM-32 can't 'Set WiFi to AP+STA' to start AP Mode.

**To Reproduce**
In `Build: ESP_Easy_mega_20230304_normal_ESP32_4M316k Mar 4 2023` everything is OK but after update to another builds AP Mode not working.

**Expected behavior**
After update to `ESPEasy_mega_20230505_ESP32` (ESP_Easy_mega_20230505_normal_ESP32_4M316k.bin) from web ,
Without any change in settings if I only change the SSID name in `Config` tab and reboot the esp board , `AP Mode` not activated.

**Used platform when `AP Mode` works:**
 - ESP type: ESP32
 - Build version: ESP_Easy_mega_20230304_normal_ESP32_4M316k Mar 4 2023
 - Build set : normal

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: [4M]
 - Brand/Model: [ESP -WROOM-32]
 - Browser + OS [Windows 10/Firefox]\
 

**Used platform when `AP Mode` not works:**
 - ESP type: ESP32
 - Build version: ESP_Easy_mega_20230505_normal_ESP32_4M316k
 - Build set : normal

Thanks

TD-er, 2024-01-05:
Yep, working on it.
ESP-IDF debug log:
```
[ 11104][D][WiFiGeneric.cpp:1043] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 11108][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Started
[ 11109][D][WiFiGeneric.cpp:1043] _eventCallback(): Arduino Event: 10 - AP_START
[ 11216][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
[ 11218][V][WiFiGeneric.cpp:146] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[ 11230][V][WiFiGeneric.cpp:193] set_esp_interface_ip(): DHCP Server Range: 192.168.4.2 to 192.168.4.12
[ 11239][E][WiFiGeneric.cpp:201] set_esp_interface_ip(): DHCPS Set Netmask Failed! 0x5001
```

mehdi-song, 2024-01-05:
In console with usb cable : 

Console log (`ESP_Easy_mega_20230304_normal_ESP32_4M316k, (GitHub Actions) mega-20230304 (ESP32 SDK 4.4.4.20230208)`) : 

`INIT : Booting version: ESP_Easy_mega_20230304_normal_ESP32_4M316k, (GitHub Actions) mega-20230304 (ESP32 SDK 4.4.4.20230208)
7789 : Info   : INIT : Free RAM:263580
7791 : Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 1704475305 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
7792 : Info   : FS   : Mounting...
7816 : Info   : FS   : Mount successful, used 141815 bytes of 290156
7831 : Info   : CRC  : Settings CRC           ...OK
7837 : Info   : CRC  : SecuritySettings CRC   ...OK 
7853 : Info   : INIT : SPI not enabled
7967 : Info   : WIFI : Set WiFi to STA
7979 : Info   : WiFi : Event STA Started
8081 : Info   : WiFi : Start network scan all channels
14496 : Info   : WiFi : Scan finished, found: 9
14498 : Info   : WiFi : Start network scan all channels
18093 : Info   : WiFi : Scan finished, found: 12
18197 : Info   : WIFI : Set WiFi to AP+STA
18200 : Info   : WiFi : Event AP Started
18761 : Info   : WiFi : Event AP Stopped
18762 : Info   : WiFi : Event AP Started
18763 : Info   : WIFI : AP Mode ssid will be iot-1 with address 192.168.4.1
18764 : Info   : WiFi : WifiDisconnect()
18868 : Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
18869 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
18970 : Info   : WIFI : Set WiFi to OFF
19087 : Info   : INIT : Free RAM:257880
19190 : Info   : INFO : Plugins: 47 [Normal] (ESP32 SDK 4.4.4.20230208)
19293 : Info   : WIFI : Set WiFi to STA
19304 : Info   : WiFi : Event STA Started
19405 : Info   : WiFi : Start network scan all channels
26421 : Info   : WiFi : Scan finished, found: 11
26525 : Info   : WIFI : Set WiFi to AP+STA
26528 : Info   : WiFi : Event AP Started
27089 : Info   : WiFi : Event AP Stopped
27090 : Info   : WiFi : Event AP Started
27091 : Info   : WIFI : AP Mode ssid will be iot-1 with address 192.168.4.1
27096 : Info   : WiFi : Start network scan all channels
34138 : Info   : WiFi : Scan finished, found: 8
34141 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
34147 : Info   : Webserver: start
34211 : Info   : SYS  : 0,34.20,0
`

Console log (`ESP_Easy_mega_20230505_normal_ESP32_4M316k, (GitHub Actions) mega-20230505 (ESP32 SDK 4.4.4.20230310)`) : 

`INIT : Booting version: ESP_Easy_mega_20230505_normal_ESP32_4M316k, (GitHub Actions) mega-20230505 (ESP32 SDK 4.4.4.20230310)
135 : Info   : INIT : Free RAM:281160
136 : Info   : INIT : Soft Reboot #2 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 1704474970 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
137 : Info   : FS   : Mounting...
162 : Info   : FS   : Mount successful, used 141815 bytes of 290156

Build changed!
184 : Info   : CRC  : SecuritySettings CRC   ...OK 
200 : Info   : INIT : SPI not enabled
202 : Info   : Set Network mode: WiFi
305 : Info   : WIFI : Set WiFi to STA
411 : Info   : WiFi : Event STA Started
512 : Info   : WiFi : Start network scan channel 1
2630 : Info   : WiFi : Scan finished, found: 10
2633 : Info   : Setup: Scan all channels
2650 : Info   : INIT : Free RAM:233208
2771 : Info   : INFO : Plugins: 47 [Normal] (ESP32 SDK 4.4.4.20230310)
2777 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
2782 : Info   : Webserver: start
2846 : Info   : SYS  : 0,2.84,0
`

TD-er, 2024-01-05:
Added an issue for it: https://github.com/espressif/arduino-esp32/issues/9069

TD-er, 2024-01-06:
OK, I have a fix ready.
Either this fix will be implemented in Arduino code, or Jason2866 will implement it in a new platform_package for me.

Problem is that we're using a slightly newer version of the ESP-IDF code compared to what the ESP32/Arduino folks are using.
So it is not yet their problem, but it will soon be.
This makes it a bit unclear whether they will already fix it or that we must patch it ourselves until it is fixed in that repo.

TD-er, 2024-01-06:
Can you do a check based on the build here: 
https://td-er.nl/ESPEasy/latest/
Or the GH Actions builds: https://github.com/letscontrolit/ESPEasy/actions/runs/7433945595
It should have been fixed on the ESP32 "LittleFS" builds.

mehdi-song, 2024-01-07:
Hi
I can't use ESP32 "LittleFS" build , can you build ESP_Easy_mega_20240106_normal_ESP32_4M316k.bin ?

TD-er, 2024-01-07:
> Hi I can't use ESP32 "LittleFS" build , can you build ESP_Easy_mega_20240106_normal_ESP32_4M316k.bin ?

That's present in the build:
[image]

mehdi-song, 2024-01-07:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4
�U00.172 : (288964) Info   : 

INIT : Booting version: ESP_Easy_mega_20240106_normal_ESP32_4M316k, (GitHub Actions) HEAD_2e1d436 (ESP32 SDK 4.4.6)
00.174 : (288916) Info   : INIT : Free RAM:288916
00.175 : (288788) Info   : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reset by PRO CPU
00.176 : (288676) Info   : FS   : Mounting...
00.202 : (284288) Info   : FS   : Mount successful, used 141815 bytes of 290156
00.219 : (284072) Info   : CRC  : Settings CRC           ...OK
00.244 : (283912) Info   : ESPEasy console using ESPEasySerial
00.252 : (283796) Info   : CRC  : SecuritySettings CRC   ...OK 
00.271 : (283748) Info   : INIT : SPI not enabled
00.272 : (283696) Info   : Set Network mode: WiFi
00.375 : (283592) Info   : WIFI : Set WiFi to STA
00.566 : (244252) Info   : WiFi : Start network scan all channels
02.287 : (243892) Info   : WiFi : Scan finished, found: 3
02.289 : (243892) Info   : WiFi : Start network scan all channels
04.010 : (243508) Info   : WiFi : Scan finished, found: 4
04.012 : (243620) Info   : Setup: Scan all channels
04.036 : (243556) Info   : ESPEasy console using ESPEasySerial
04.038 : (243508) Info   : INIT : Free RAM:243508
04.063 : (243444) Info   : ESPEasy console using ESPEasySerial
04.065 : (243332) Info   : INFO : Plugins: 48 [Normal] (ESP32 SDK 4.4.6)
04.074 : (243252) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
04.080 : (238788) Info   : Webserver: start
05.428 : (238444) Info   : WD   : Uptime 0  ConnectFailures 0 FreeMem 238588 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
35.425 : (238300) Info   : WD   : Uptime 1  ConnectFailures 0 FreeMem 238444 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED

TD-er, 2024-01-07:
Not sure what you are trying to tell here?
Do you have WiFi credentials configured?

Also the bug for not running DHCP-service on an access point has only been fixed on the LittleFS builds (thus ESP-IDF 5.1 based SDK builds) 
That SDK platform code doesn't have SPIFFS support, so I cannot build a SPIFFS build based on that ESP-IDF 5.1 SDK.

So please tell me what you did expect to be shown, what is configured and what it is you try to show with these logs?

mehdi-song, 2024-01-08:
Thanks for your response.
ESPEasy has AP Mode and When it can't find the wifi SSID then it starts the AP Mode so that it can be easily to set wifi settings.
Now when I change the SSID in `Config->Wifi Settings->SSID` and reboot it , AP Mode not start and only STA Mode started and scanned Wifi SSIDs and it will try to connect to new SSID name.
In old versions such as `ESP_Easy_mega_20230304_normal_ESP32_4M316k Mar 4 2023` AP Mode works well but after that version AP Mode not works and only STA activated .

TD-er, 2024-01-08:
OK, I think I now understand and I'm sorry but then the linked pull request I made does not fix your issue.

commit 77576896 (2024-01-18): [ESP-IDF5.1] Fix starting DHCP on ESP AP during setup

Fixes: #4931
closed 2024-01-18 by TD-er