letscontrolit/ESPEasy #4933: Can't build custom LittleFS image for ESP32-C3

chemmex, 2024-01-09:
env:**custom_ESP32c3_4M316k_CDC** and **normal_ESP32c3_4M316k_LittleFS_CDC** build without issues. However, if I add -DPLUGIN_BUILD_CUSTOM to the build flags, it fails with folliwing:   

```Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\libca6\Adafruit_NeoPixel\esp8266.c.o
In file included from lib/GPIO_DirectAccess/GPIO_Direct_Access.cpp:1:
lib/GPIO_DirectAccess/GPIO_Direct_Access.h: In function 'uint32_t directRead(uint32_t)':
lib/GPIO_DirectAccess/GPIO_Direct_Access.h:150:13: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  150 |     return (GPIO.in.val >> pin) & 0x1;
      |             ^~~~
      |             PI
lib/GPIO_DirectAccess/GPIO_Direct_Access.h: In function 'void directWriteLow(uint32_t)':
lib/GPIO_DirectAccess/GPIO_Direct_Access.h:170:5: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  170 |     GPIO.out_w1tc.val = ((uint32_t)1 << pin);
      |     ^~~~
      |     PI
lib/GPIO_DirectAccess/GPIO_Direct_Access.h: In function 'void directWriteHigh(uint32_t)':
lib/GPIO_DirectAccess/GPIO_Direct_Access.h:189:5: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  189 |     GPIO.out_w1ts.val = ((uint32_t)1 << pin);
      |     ^~~~
      |     PI
lib/GPIO_DirectAccess/GPIO_Direct_Access.h: In function 'void directModeInput(uint32_t)':
lib/GPIO_DirectAccess/GPIO_Direct_Access.h:208:5: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  208 |     GPIO.enable_w1tc.val = ((uint32_t)1 << (pin));
      |     ^~~~
      |     PI
lib/GPIO_DirectAccess/GPIO_Direct_Access.h: In function 'void directModeOutput(uint32_t)':
lib/GPIO_DirectAccess/GPIO_Direct_Access.h:241:5: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  241 |     GPIO.enable_w1ts.val = ((uint32_t)1 << (pin));
      |     ^~~~
      |     PI
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\Esp32_i2s.c.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\HsbColor.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\HslColor.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\HtmlColor.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\HtmlColorNameStrings.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\HtmlColorNames.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\HtmlColorShortNames.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\NeoEsp32RmtMethod.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\NeoEsp32RmtMethod_idf5.cpp.o
Compiling .pio\build\custom_ESP32c3_4M316k_LittleFS\lib044\NeoPixelBus\internal\NeoEsp8266DmaMethod.cpp.o
*** [.pio\build\custom_ESP32c3_4M316k_LittleFS\lib43b\GPIO_DirectAccess\GPIO_Direct_Access.cpp.o] Error 1
===================================================== [FAILED] Took 67.61 seconds =====================================================

Environment                     Status    Duration
------------------------------  --------  ------------
custom_ESP32c3_4M316k_LittleFS  FAILED    00:01:07.608
================================================ 1 failed, 0 succeeded in 00:01:07.608 ================================================ ```

Custom.h is a copy of Custom-sample.h, no plugins added

tonhuisman, 2024-01-09:
It looks like you added that env locally, did you extend it from the correct base configuration?

TD-er, 2024-01-09:
> Custom.h is a copy of Custom-sample.h, no plugins added

I will give it a try.
Can you also post your PlatformIO env if you added one yourself as Ton suggested?

chemmex, 2024-01-09:
Yes, 
```
[env:custom_ESP32c3_4M316k_LittleFS]
extends                   = esp32c3_common_LittleFS
board                     = esp32c3cdc
build_flags               = -DPLUGIN_BUILD_CUSTOM
lib_ignore                = ${esp32c3_common_LittleFS.lib_ignore}
                            ${no_ir.lib_ignore}
```

chemmex, 2024-01-09:
Just managed to build it by adding `${esp32c3_common_LittleFS.build_flags} `

Sorry for disturbance, closing the issue :)

closed 2024-01-09 by chemmex