letscontrolit/ESPEasy #4934: [BUG]WT32-ETH01 ethernet only on cold boot

wutz-ESP, 2024-01-11:
WT32-ETH01 ethernet only on cold boot - reboot on tools Reboot connects to wifi

**Used platform (please complete the following information):**
 - ESP type: WT32-ETH01
 - Build version: ESP_Easy_mega_20231225_normal_ESP32_4M316k_ETH Dec 25 2023

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M
 - Brand/Model: WT32-ETH01
 - Browser + OS Windows 11/Chrome

***  Soft Reboot  ***
`

System Info | Copy info to clipboard (⋄)
-- | --
Unit Number: | 123
Local Time: | 2024-01-11 09:43:45
Time Source: | NTP
Time Wander: | 0.0 [ppm]
Uptime: | 00h03m
Load: | 65.75% (LC=2225)
CPU Eco Mode: | false
Boot: | Soft Reboot (1)
Reset Reason: | CPU0: Software reset CPU CPU1: Software reset CPU
Last Action before Reboot: | Intended Reboot: CommandReboot
SW WD count: | 0
Memory
Heap Size: | 310128
Heap Min Free: | 175488
Free RAM: | 194848 (175488 - sendContentBlocking)
Heap Max Free Block: | 110580
Free Stack: | 7440 (2168 - parseTemplate3)
PSRAM Size: | 0
Network
Network Type: | WiFi
IP Config: | DHCP
IP / Subnet: | 192.168.10.137 / 255.255.255.0
Gateway: | 192.168.10.1
Client IP: | 192.168.10.110
DNS: | 192.168.10.1 / 0.0.0.0
Allowed IP Range: | 192.168.10.1 - 192.168.10.255
Connected: | 44.609
Number Reconnects: | 8
WiFi ?
WiFi Connection: | 802.11n (HT20) (RSSI -48 dBm)
SSID: | FritzBox-WZ (xx:xx:xx:xx:xx:xx)
WiFi TSF time: | 17:12:40.798396
Channel: | 6
Encryption Type: | WPA2/PSK
Last Disconnect Reason: | (1) Unspecified
Configured SSID1: |  
Configured SSID2: | FritzBox-WZ
STA MAC: | 0C:DC:7E:5D:8E:68
AP MAC: | 0C:DC:7E:5D:8E:69
WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | false
Connection Failure Threshold: | 0
Max WiFi TX Power: | 17.50
Current WiFi TX Power: | 1.00
WiFi Sensitivity Margin: | 3
Send With Max TX Power: | false
Extra WiFi scan loops: | 0
Use Last Connected AP from RTC: | false
Extra Wait WiFi Connect: | false
Hidden SSID Slow Connect: | false
Include Hidden SSID: | false
Enable SDK WiFi Auto Reconnect: | false
Firmware
Build:⋄ | 20231225 - Mega32
System Libraries:⋄ | ESP32 SDK 4.4.6
Git Build:⋄ | mega-20231225
Plugin Count:⋄ | 48 [Normal]
Build Origin: | GitHub Actions
Build Time:⋄ | Dec 25 2023 22:00:43
Binary Filename:⋄ | ESP_Easy_mega_20231225_normal_ESP32_4M316k_ETH
Build Platform:⋄ | Linux-6.2.0-1018-azure-x86_64-with-glibc2.35
Git HEAD:⋄ | HEAD_e67583a
System Status
Syslog Log Level: | None
Serial Log Level: | None
Web Log Level: | None
Enable Serial Port Console: | false
Console Serial Port: | -
I2C Bus State: | OK
I2C bus cleared count: | 0
Network Services
Network Connected: | ✔
NTP Initialized: | ✔
ESP Board
ESP Chip ID: | 6131304 (0x5D8E68)
ESP Chip Frequency: | 240 MHz
ESP Crystal Frequency: | 40 MHz
ESP APB Frequency: | 80 MHz
ESP Chip Model: | ESP32-D0WDQ5
ESP Chip Features: | Wi-Fi bgn / BLE
ESP Chip Revision: | 1.0
ESP Chip Cores: | 2
ESP Board Name: | Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS
Storage
Flash Chip ID: | Vendor: 0x5E Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash Chip Speed: | 80 MHz
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 1367 kB (1856 kB free)
Max. OTA Sketch Size: | 1856 kB (1900544 bytes)
SPIFFS Size: | 283 kB (124 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 35

***  cold boot  ***
`

System Info | Copy info to clipboard (⋄)
-- | --
Unit Number: | 123
Local Time: | 2024-01-11 09:57:20
Time Source: | NTP
Time Wander: | 0.0 [ppm]
Uptime: | 00h00m
Load: | 99.58% (LC=12)
CPU Eco Mode: | false
Boot: | Cold Boot (0)
Reset Reason: | CPU0: Vbat power on reset CPU1: for APP CPU, reset by PRO CPU
Last Action before Reboot: | Background Task
SW WD count: | 0
Memory
Heap Size: | 312368
Heap Min Free: | 220176
Free RAM: | 226488 (220284 - sendContentBlocking)
Heap Max Free Block: | 110580
Free Stack: | 7440 (2168 - parseTemplate3)
PSRAM Size: | 0
Network
Network Type: | Ethernet
IP Config: | DHCP
IP / Subnet: | 192.168.10.96 / 255.255.255.0
Gateway: | 192.168.10.1
Client IP: | 192.168.10.110
DNS: | 192.168.10.1 / 0.0.0.0
Allowed IP Range: | 192.168.10.1 - 192.168.10.255
Connected: | 12.198
Number Reconnects: | 0
WiFi ?
WiFi Connection: | -
SSID: | -
Channel: | -
Encryption Type: | -
STA MAC: | 0C:DC:7E:5D:8E:68
AP MAC: | 00:00:00:00:00:00
Ethernet
Eth State: | Link Up
Eth Speed: | 100Mbps
Eth Mode: | Full Duplex
Eth MAC: | 0C:DC:7E:5D:8E:6B
WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | false
Connection Failure Threshold: | 0
Max WiFi TX Power: | 17.50
Current WiFi TX Power: | 0.00
WiFi Sensitivity Margin: | 3
Send With Max TX Power: | false
Extra WiFi scan loops: | 0
Use Last Connected AP from RTC: | false
Extra Wait WiFi Connect: | false
Hidden SSID Slow Connect: | false
Include Hidden SSID: | false
Enable SDK WiFi Auto Reconnect: | false
Firmware
Build:⋄ | 20231225 - Mega32
System Libraries:⋄ | ESP32 SDK 4.4.6
Git Build:⋄ | mega-20231225
Plugin Count:⋄ | 48 [Normal]
Build Origin: | GitHub Actions
Build Time:⋄ | Dec 25 2023 22:00:43
Binary Filename:⋄ | ESP_Easy_mega_20231225_normal_ESP32_4M316k_ETH
Build Platform:⋄ | Linux-6.2.0-1018-azure-x86_64-with-glibc2.35
Git HEAD:⋄ | HEAD_e67583a
System Status
Syslog Log Level: | None
Serial Log Level: | None
Web Log Level: | Info
Enable Serial Port Console: | false
Console Serial Port: | -
I2C Bus State: | OK
I2C bus cleared count: | 0
Network Services
Network Connected: | ✔
NTP Initialized: | ✔
ESP Board
ESP Chip ID: | 6131304 (0x5D8E68)
ESP Chip Frequency: | 240 MHz
ESP Crystal Frequency: | 40 MHz
ESP APB Frequency: | 80 MHz
ESP Chip Model: | ESP32-D0WDQ5
ESP Chip Features: | Wi-Fi bgn / BLE
ESP Chip Revision: | 1.0
ESP Chip Cores: | 2
ESP Board Name: | Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS
Storage
Flash Chip ID: | Vendor: 0x5E Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash Chip Speed: | 80 MHz
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 1367 kB (1856 kB free)
Max. OTA Sketch Size: | 1856 kB (1900544 bytes)
SPIFFS Size: | 283 kB (124 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 35

TD-er, 2024-01-11:
Can you also include the used settings (screenshot is good enough) of the Hardware tab where it shows the Ethernet config?

wutz-ESP, 2024-01-11:
[image]

[image]

[image]

[image]

[image]

TD-er, 2024-01-11:
I will start digging through my parts to see if I have a rev 1.2 here.

Can you try to set GPIO-16 to pull-up on the same Hardware tab where other Ethernet settings are stored?

I think I already have a clue to why this may fail, as GPIO-16 is being used for PSRAM when it is available.
But your board obviously doesn't have PSRAM, so it may cause some high frequency noise on the power signal of the LAN chip at boot.

I think I will disable PSRAM support for ESP32 builds with Ethernet (or maybe add a separate build) as this is really unfortunate they used GPIO-16 for the power pin.

wutz-ESP, 2024-01-11:
doesn'1 make any difference

TD-er, 2024-01-11:
I just made a [build which should have PSRAM disabled](https://www.dropbox.com/scl/fi/hcyq0o04s6szmhsiqk6do/BUG-WT32-ETH01-ethernet-only-on-cold-boot-4934-ESP_Easy_mega_20240111_normal_ESP32_4M316k_ETH.factory.zip?rlkey=ofycav18dnfwkdf4pwpxveqyt&dl=0)
Can you try it while I keep on searching where I keep those WT32-ETH01 boards....

As it is completely untested, please try on boards which are accessible to flash again via serial when things go wrong.

wutz-ESP, 2024-01-11:
thank you so much for your time - first automated reboot  after flash was ok but manually reboot does not get an ip-adress from AVM Fritz!Box, if i disconnect power about 10 seconds is everything ok.
After reboot this unit does not get NTP and is not reachable over ip but gets new parameters from another unit

[image]

[image]

[image]

TD-er, 2024-01-11:
So at least it does reboot fine now?
"only" an issue with ESPEasy's internal state of whether it is connected or not.
Meaning I really should test using a real device.

Maybe you can already backup your settings file and check all checkboxes on the "Factory Reset" page and save those settings.
This will at least make it easier later to also test a build based on IDF5.1.
But since that uses LittleFS, you need to do a factory reset (ESPEasy will do this when you flash a LittleFS build)
By checking all those checkboxes on the Factory Reset page, ESPEasy will store your current network/WiFi settings in NVS and use this when performing a factory reset, making it a lot easier to migrate.

I will make a build for you later this evening based on IDF5.1 to see if this fixes some stuff.

wutz-ESP, 2024-01-11:
Next few days im busy. On monday i will give a try on litte network with another AVM Frit!Box and only this testing unit, maybe there is some problem with dhcp on the avm router.

thank you so much

TD-er, 2024-01-11:
As far as I know, the only issues with AVM hardware is when using WiFi.
Not with LAN.

And I also have lots of AVM stuff here, so I can test (as soon as I can find those boards, I recently saw them...)

wutz-ESP, 2024-01-15:
Build: [ESP_Easy_mega_20231225_normal_ESP32_4M316k_LittleFS_ETH Dec 25 2023](https://github.com/letscontrolit/ESPEasy/releases)
seems working perfect to me

Is it possible flashing LittleFS  Firmware in Tools Update Firmware or only with USB-TTL-Adapter and NodeMC flasher?

tonhuisman, 2024-01-15:
If you want to keep your current configuration, when the device uses SPIFFS (non-LittleFS binaries) then you'll have to backup your config using the Tools/Save button, then use OTA update (Update Firmware) to install the LittleFS build, then restore the config.dat file using the Tools/Load button (rename the .dat file you downloaded before). That should restore (most) of your settings.

wutz-ESP, 2024-01-15:
ESP_Easy_mega_20240111_normal_ESP32_4M316k_ETH.factory.bin
works  often 

> 9+▒scV▒▒+D)$c▒i▒▒▒K▒▒▒▒▒V▒ oscillator MDC Pin: 23 MIO Pin: 18 Power Pin: 16
00.427 : (281956) Info   : ETH power ON
00.828 : (281876) Info   : ethResetGPIOpins()
02.376 : (253688) Info   : ETH event: Started
02.378 : (253800) Info   : ETH event: Connected
02.383 : (253776) Info   : ETH event: Got IP
02.424 : (249404) Info   : Webserver: start
02.436 : (249356) Info   : EVENT: System#Boot
02.439 : (249116) Info   : processEthernetConnected()
02.441 : (248852) Info   : ETH MAC: 0C:DC:7E:5D:8E:6B DHCP IP: 192.168.10.162 (E                                                  SPeasy-T2) GW: 192.168.10.1 SN: 255.255.255.0 DNS: 192.168.10.1/0.0.0.0 FULL_DUP                                                  LEX 100Mbps   duration: 4 ms
02.443 : (248652) Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy                                                   internal wifi status: DISCONNECTED
02.481 : (246368) Info   : NTP  : NTP replied: delay 31 msec Accuracy increased                                                   by 15 msec
02.484 : (248440) Info   : Time set to 2085978497.016
02.485 : (247768) Info   : Current Time Zone:  DST time start: 2036-03-30 02:00:                                                  00 offset: 120 min STD time start: 2036-10-26 03:00:00 offset: 60 min
02.487 : (247664) Info   : Local time: 2036-02-07 07:28:17
02.489 : (245520) Info   : UDP : Start listening on port 8266
02.490 : (245472) Info   : firstLoopConnectionsEstablished
02.497 : (245408) Info   : Internal command: All checked OK
02.501 : (245828) Info   : EVENT: Ethernet#LinkUp
02.506 : (245852) Info   : EVENT: Ethernet#Connected
02.510 : (245364) Info   : EVENT: Time#Initialized
03.217 : (245828) Info   : EVENT: Clock#Time=Thu,7:28
03.772 : (245640) Info   : WD   : Uptime 0  ConnectFailures 0 FreeMem 245784 Eth                                                  SpeedState Link Up Full Duplex 100Mbps ETH status: Conn. IP Init
19.728 : (242720) Info   : static_file: /esp.css to: 192.168.10.110
21.199 : (241944) Info   :      : Rebooting...
ets Jun  8 2016 00:22:57

> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12732
ho 0 tail 12 room 4
load:0x40080400,len:2908
entry 0x400805c4
▒U00.174 : (287976) Info   :

> INIT : Booting version: ESP_Easy_mega_20240111_normal_ESP32_4M316k_ETH, (Self built) bugfix/ESP32_factoryDefaults_a95ad3b (ESP32 SDK 4.4.6)
00.176 : (287832) Info   : INIT : Free RAM:287832
00.178 : (287640) Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 2085978515 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
00.179 : (287576) Info   : FS   : Mounting...
00.203 : (283188) Info   : FS   : Mount successful, used 141815 bytes of 290156
00.222 : (282444) Info   : CRC  : Settings CRC           ...OK
00.248 : (282812) Info   : ESPEasy console using ESPEasySerial
00.256 : (282696) Info   : CRC  : SecuritySettings CRC   ...OK
00.289 : (282648) Info   : INIT : SPI not enabled
00.315 : (282580) Info   : ESPEasy console using ESPEasySerial
00.316 : (282532) Info   : INIT : Free RAM:282532
00.341 : (282468) Info   : ESPEasy console using ESPEasySerial
00.343 : (282356) Info   : INFO : Plugins: 48 [Normal] (ESP32 SDK 4.4.6)
00.344 : (282408) Info   : EVENT: System#Wake
00.427 : (282016) Info   : EVENT: System#BootMode=1,1,0,0,1,0
00.429 : (281932) Info   : ETH PHY Type: LAN8710/LAN8720 PHY Addr: 1 Eth Clock mode: External crystal oscillator MDC Pin: 23 MIO Pin: 18 Power Pin: 16
00.430 : (281932) Info   : ETH power ON
00.831 : (281852) Info   : ethResetGPIOpins()
02.379 : (253632) Info   : ETH event: Started
02.381 : (253744) Info   : ETH event: Connected
02.386 : (253728) Info   : ETH event: Got IP
02.427 : (249332) Info   : Webserver: start
02.439 : (249284) Info   : EVENT: System#Boot
02.442 : (249036) Info   : processEthernetConnected()
02.444 : (248772) Info   : ETH MAC: 0C:DC:7E:5D:8E:6B DHCP IP: 192.168.10.162 (ESPeasy-T2) GW: 192.168.10.1 SN: 255.255.255.0 DNS: 192.168.10.1/0.0.0.0 FULL_DUPLEX 100Mbps   duration: 4 ms
02.446 : (248572) Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
02.720 : (246296) Info   : NTP  : NTP replied: delay 31 msec Accuracy increased by 15 msec
02.723 : (247832) Info   : Time set to 2085978497.016
02.724 : (247688) Info   : Current Time Zone:  DST time start: 2036-03-30 02:00:00 offset: 120 min STD time start: 2036-10-26 03:00:00 offset: 60 min
[... 101 lines cut ...]
02.439 : (249316) Info   : EVENT: System#Boot
02.442 : (249076) Info   : processEthernetConnected()
02.444 : (248996) Info   : EVENT: Ethernet#LinkUp
03.775 : (248900) Info   : WD   : Uptime 0  ConnectFailures 0 FreeMem 249020 EthSpeedState Link Up Full Duplex 100Mbps ETH status: Conn.
04.373 : (248812) Error  : ETH event: Disconnected
04.375 : (248876) Info   : EVENT: Ethernet#Disconnected
05.831 : (248876) Info   : Set Network mode: WiFi
05.934 : (248828) Info   : WIFI : Set WiFi to STA
06.113 : (221868) Error  : WIFI : No valid wifi settings
06.115 : (221804) Info   : WiFi : Start network scan all channels
07.636 : (221452) Info   : WiFi : Scan finished, found: 3
07.638 : (221428) Info   : WiFi : WiFiConnected(), start AP
[... cut ...]
08.281 : (219048) Info   : WIFI : Set WiFi to AP+STA
08.385 : (217220) Info   : WiFi : Start network scan all channels
11.540 : (216868) Info   : WiFi : Scan finished, found: 3
11.543 : (216908) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
11.555 : (216948) Info   : EVENT: WiFi#Disconnected
11.571 : (216988) Info   : EVENT: WiFi#APmodeEnabled
33.789 : (216892) Info   : WD   : Uptime 1  ConnectFailures 0 FreeMem 217036 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED

wutz-ESP, 2024-01-17:
now started again - wondering about wrong date / time   2036-02-07 ? ? ?

> <html>
> <body>

System Info |  
-- | --
Unit Number: | 123
Git Build: | bugfix/ESP32_factoryDefaults_a95ad3b
Local Time: | 2036-02-07 07:55:54
Time Source: | NTP
Uptime: | 00h28m
Load: | 66.37% (LC=5386)
Free RAM: | 224736 (207424 - sendWebPage)
Free Stack: | 7456 (2160 - parseTemplate3)
Network Type: | Ethernet
Eth Speed State: | Link Up Full Duplex 100Mbps

</body>
</html>

TD-er, 2024-01-17:
that's really odd , unless you have setup a specific NTP server?

wutz-ESP, 2024-01-18:
Unit is not visible on other units with older firmware release (ESP_Easy_mega_20231013_normal_ESP32_4M316k_ETH Oct 13 2023), but on this unit you can see units having older firmware releases.

Reverted al my units back to ESP_Easy_mega_20231130_normal_ESP32_4M316k_ETH Nov 30 2023

TD-er, 2024-01-18:
Did you have the UDP port set on the Tools->Advanced page?
In the "X-mas" build, there was an issue with creating proper factory default settings.
In this actions build the factory default settings bug should be fixed: https://github.com/letscontrolit/ESPEasy/actions/runs/7568834175

wutz-ESP, 2024-01-22:
Yes, after upgrade checked the port an tested 8265 an then back to 8266 on this unit.
Wondered because this unit can see older ones, but older releases did not see the new ones.

TD-er, 2024-01-22:
That might be a bug in the p2p code which is already fixed in the latest code in the 'mega' branch.

wutz-ESP, 2024-01-29:
Build: [ESP_Easy_mega_20240118_normal_ESP32_4M316k_ETH Jan 18 2024](https://github.com/letscontrolit/ESPEasy/releases)

p2p works and ntp works -  Works in my condition as expected   ;-)

System Info |  
-- | --
Unit Number: | 123
Git Build: | mega_e015d1b
Local Time: | 2024-01-29 12:09:56
Time Source: | NTP
Uptime: | 00h07m
Load: | 66.79% (LC=3773)
Free RAM: | 219820 (208556 - sendWebPage)
Free Stack: | 7456 (2156 - parseTemplate3)
Network Type: | Ethernet
Eth Speed State: | Link Up Full Duplex 100Mbps
Eth IP Address: | 192.168.10.96

TD-er, 2024-01-29:
Can you maybe also check this build as it is soon to be merged:
https://github.com/letscontrolit/ESPEasy/actions/runs/7682112311

But please only do this on units which are already on your desk so you can recover easily when it failed for whatever reason.

N.B. please use a Custom LittleFS ETH build, which may also wipe your file system if you were using SPIFFS.

wutz-ESP, 2024-01-29:
[ESP_Easy_mega_20240128_normal_ESP32_4M316k_ETH Jan 28 2024](https://github.com/letscontrolit/ESPEasy/releases)
works - reboot ok - cold boot ok - p2p ok - ntp ok

System Info |  
-- | --
Unit Number: | 123
Git Build: | HEAD_b56555a
Local Time: | 2024-01-29 12:30:20
Time Source: | NTP
Uptime: | 00h01m
Load: | 62.35% (LC=4743)
Free RAM: | 219664 (213096 - sendWebPage)
Free Stack: | 7456 (2156 - parseTemplate3)
Network Type: | Ethernet
Eth Speed State: | Link Up Full Duplex 100Mbps
Eth IP Address: | 192.168.10.96

[ESP_Easy_mega_20240128_normal_ESP32_4M316k_LittleFS_ETH Jan 28 2024](https://github.com/letscontrolit/ESPEasy/releases)
works - reboot ok - cold boot ok - p2p ok - ntp ok
p2p seems to be a little slower but Load is about 5% lower

System Info |  
-- | --
Unit Number: | 123
Git Build: | HEAD_b56555a
Local Time: | 2024-01-29 13:13:23
Time Source: | NTP
Uptime: | 00h02m
Load: | 57.58% (LC=11981)
Free RAM: | 222804 (202344 - sendWebPage)
Free Stack: | 7360 (2172 - parseTemplate3)
Network Type: | Ethernet
Eth Speed State: | Link Up Full Duplex 100Mbps
Eth IP Address: | 192.168.10.96

[image]

[image]

TD-er, 2024-01-29:
And did you notice anything "new" when looking at the root page (or sysinfo page)? ;)
And when looking through the Ethernet options ;)

TD-er, 2024-01-29:
Ah you used the "normal" build, then you won't see the new stuff on the root page or sysinfo page....

wutz-ESP, 2024-01-29:
[ESP_Easy_mega_20240128_collection_A_ESP32_4M316k_ETH Jan 28 2024](https://github.com/letscontrolit/ESPEasy/releases)

i can see  mDNS but at my home network it is not reachable 

http://espeasy-t2.local/      this  will work for me   http://espeasy-t2

[image]

TD-er, 2024-01-29:
That was not what I meant...

I meant the IPv6 I added.
Right now I only added it to "Custom" builds so we can just start testing with IPv6 as it is still very preliminary.

I will have a look at the mDNS why it is not working.
N.B. it can also be a policy in your network or on your PC to ignore mDNS.

No need to enable mDNS as it is a bit of an odd protocol and could be used to do malicious stuff, so if it is disabled, just leave it as is.