letscontrolit/ESPEasy #4935: [BUG] Enable SDK WiFi Auto Reconnect does not correctly show in Info page (it's always at false state)

iz8mbw, 2024-01-12:
I have the setting "Enable SDK WiFi Auto Reconnect:" enabled in the Tools -> Advanced page:
[image]

When I go on the page Tools -> Info the "Enable SDK WiFi Auto Reconnect:" is always at false (also if it's enabled):
[image]

TD-er, 2024-01-12:
Which build?

TD-er, 2024-01-12:
Never mind, found the bug, will fix it

iz8mbw, 2024-01-12:
All build :-)

TD-er, 2024-01-12:
> All build :-)

The question was more like build date :)
But I have already found it and will be merged soon.

iz8mbw, 2024-01-12:
On a very very FRESH build but the bug is "old" since "Enable SDK WiFi Auto Reconnect" was introduced.
Thanks.

iz8mbw, 2024-01-12:
Just built a Custom build using [this change](https://github.com/TD-er/ESPEasy/commit/6f6ec09809eea626e2e251c4be49b25ec1f3ecee) and:
[image]

commit 6f6ec098 (2024-01-18): Fix "Enable SDK WiFi Auto Reconnect" state on sysinfo page (#4935)

Fixes: #4935
closed 2024-01-18 by TD-er