letscontrolit/ESPEasy #4939: P073_7DGT; 7-segment; 6-digit TM1637 commands are no longer executed after a certain time (cash overflow?)

Sminter4, 2024-01-15:
**Project:**
I built a dashboard that shows the time, day and outside temperature alternately every 3 seconds.
(And other stats via a Neopixel, but that's irrelevant for this)

[image]

**Bug:**
Unfortunately, the commands are only initially implemented correctly. After a certain time (approx. 1 hour) the old text is no longer deleted, but is only written over it (spaces contain numbers from the old text). At some point the text stops (manual commands are no longer executed) and I can only see in the log page that there should actually be a different text there. It feels like a memory overflow.

**Code:**
```
on system#boot do
 timerSet,1,5
endon

On Rules#Timer=1 do //7Seg-Wechsel
 if [var#40]=0
  7dtext,%syshour%.%sysmin_0%
 endif
 if [var#40]=1
  7dtext,%sysday%. %sysmonth_s%
 endif
 if [var#40]=2
  7dtext,[var#41]^C
 endif
 if [var#40]>1
  let,40,0
 else
  let,40,[var#40]+1
 endif
 timerSet,1,3
endon
```

**Log:**
```
3232240: EVENT: Clock#Time=Mon,17:45
3234621: EVENT: Rules#Timer=1,1
3234648: ACT : 7dtext,0.2^C
3234662: ACT : let,40,0
3234667: ACT : timerSet,1,3
3237668: EVENT: Rules#Timer=1,1
3237681: ACT : 7dtext,17.45
3237710: ACT : let,40,0+1
3237714: ACT : timerSet,1,3
3240715: EVENT: Rules#Timer=1,1
3240739: ACT : 7dtext,15. Jan
3240760: ACT : let,40,1+1
3240763: ACT : timerSet,1,3
3243764: EVENT: Rules#Timer=1,1
3243792: ACT : 7dtext,0.2^C
...
```
**Facts:**
Build: ESP_Easy_mega_20231225_neopixel_ESP8266_4M1M Dec 25 2023
HW: ESP 8266, Wemos D1 mini

TD-er, 2024-01-15:
First you can also use the looptimer, which will be running at a more constant interval as it is rescheduled based on the previous scheduled time.

And now a bug in your rules...
```
 if [var#40]>1
  let,40,0
 else
  let,40,[var#40]+1
 endif
```

Not sure how this variable could be anything other than `0` or `1`.

Sminter4, 2024-01-15:
I don't see any added value from this. But I tested some suggestions.

As a rule, I don't see any errors. I have 3 states (0,1,2). When variable 40 reaches 2, it starts again at 0. Or am I wrong?
It doesn't say `[var#40]>=1`.

New code:
```
on system#boot do
 loopTimerSet,1,3
endon

On Rules#Timer=1 do //7Seg-Wechsel
 if [var#40]=0
  7dtext,%syshour%.%sysmin_0%
 endif
 if [var#40]=1
  7dtext,%sysday%. %sysmonth_s%
 endif
 if [var#40]=2
  7dtext,[var#41]^C
 endif
 if [var#40]=2
  let,40,0
 else
  let,40,[var#40]+1
 endif
endon
```
Now I've installed the new code and rebooted. The code has now been running for 20 minutes and the display also freezes.
It's probably less to do with the rules or timers. Maybe I could produce the same error with manual command entry after 50 prompts.
```
  7dtext,%sysday%. %sysmonth_s%
  7dtext,[var#41]^C
```

tonhuisman, 2024-01-15:
Can we rule out ESP temperature and power supply stability?

Sminter4, 2024-01-15:
I have not connected a temperature sensor and constant voltage monitoring.

However, a 1000µF capacitor is connected. I can also say that all other functions work without any problems:
5 x switches and a HW-MS03 via PCF8574
1x buzzer via GPIO
1 x NeoPixel (BusFX) It shows different states from the house.

Only the display freezes. As an amateur, I would exclude the temperature and voltage here.

tonhuisman, 2024-01-15:
I was suggesting that the _TM1637 display unit_ might get overheated, maybe because it is in a small enclosure or confined space?

Sminter4, 2024-01-16:
Astonishing. I reduced the brightness to half and the display has been running stable for 12 hours. I would not have thought that. Thanks for the tip!

You obviously already know these symptoms. From where?

tonhuisman, 2024-01-16:
> You obviously already know these symptoms. From where?

Well, the hint you gave was that the rest of the devices was running fine, so the most likely 'candidate/culprit' is the display module itself, and as it didn't manifest immediately, it would probably be heat related 🥵 Reducing the heat source then should improve things 👍

TD-er, 2024-01-16:
Also you're not the first one to have "strange" issues caused by underpowered power supply or heat.
And "underpowered" can also be caused by thin wires or bad connections which cause a voltage drop when dealing with a current surge.