letscontrolit/ESPEasy #4943: [BUG] - Git Build: pygit2_not_installed

iz8mbw, 2024-01-18:
Just built a Custom build of ESP_Easy_mega_20240118_max_ESP32_16M8M_LittleFS.bin and this is what I see in Home Page:

[image]

not have the same issue for ESP_Easy_mega_20240118_custom_ESP32_4M316k_LittleFS.bin
[image]

iz8mbw, 2024-01-18:
Done but same issue.
During building:
[image]

iz8mbw, 2024-01-18:
[image]

TD-er, 2024-01-18:
The Python env in power shell or Windows cmd is not the same as the env used by PlatformIO in VS code.
How do you start the build of ESPEasy?
Do you have Python2 and Python3 installed on your system?
Maybe you should run `pip3` instead of `pip` as it may be set as an alias on your system to the Python 2.x version of pip?

iz8mbw, 2024-01-20:
Python version is 3.12.
Also if I execute
```
pip3 install -r .\requirements.txt
```

it answer me all packages are already installed.

```
PS C:\ESPEasy-mega> pip3 install -r .\requirements.txt
Requirement already satisfied: platformio>=6.1.9 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from -r .\requirements.txt (line 1)) (6.1.13)
Requirement already satisfied: pygit2>=1.10.1 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from -r .\requirements.txt 
(line 2)) (1.13.3)
Requirement already satisfied: bottle==0.12.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r 
.\requirements.txt (line 1)) (0.12.25)
Requirement already satisfied: click<9,>=8.0.4 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (8.1.7)
Requirement already satisfied: colorama in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (0.4.6)
Requirement already satisfied: marshmallow==3.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (3.20.2)
Requirement already satisfied: pyelftools==0.30 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (0.30)
Requirement already satisfied: pyserial==3.5.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (3.5)
Requirement already satisfied: requests==2.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (2.31.0)
Requirement already satisfied: semantic-version==2.10.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (2.10.0)
Requirement already satisfied: tabulate==0.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (0.9.0)
Requirement already satisfied: ajsonrpc==1.2.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (1.2.0)
Requirement already satisfied: starlette<0.36,>=0.19 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (0.35.1)
Requirement already satisfied: uvicorn<0.26,>=0.16 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (0.25.0)
Requirement already satisfied: wsproto==1.* in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from platformio>=6.1.9->-r .\requirements.txt (line 1)) (1.2.0)
Requirement already satisfied: packaging>=17.0 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from marshmallow==3.*->platformio>=6.1.9->-r .\requirements.txt (line 1)) (23.2)
Requirement already satisfied: charset-normalizer<4,>=2 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from requests==2.*->platformio>=6.1.9->-r .\requirements.txt (line 1)) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from requests==2.*->platformio>=6.1.9->-r .\requirements.txt (line 1)) (3.6)
Requirement already satisfied: urllib3<3,>=1.21.1 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from requests==2.*->platformio>=6.1.9->-r .\requirements.txt (line 1)) (2.1.0)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from requests==2.*->platformio>=6.1.9->-r .\requirements.txt (line 1)) (2023.11.17)
Requirement already satisfied: h11<1,>=0.9.0 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from wsproto==1.*->platformio>=6.1.9->-r .\requirements.txt (line 1)) (0.14.0)
Requirement already satisfied: cffi>=1.16.0 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from pygit2>=1.10.1->-r .\requirements.txt (line 2)) (1.16.0)
Requirement already satisfied: setuptools in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from pygit2>=1.10.1->-r .\requirements.txt (line 2)) (69.0.3)
Requirement already satisfied: pycparser in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from cffi>=1.16.0->pygit2>=1.10.1->-r .\requirements.txt (line 2)) (2.21)
Requirement already satisfied: anyio<5,>=3.4.0 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from starlette<0.36,>=0.19->platformio>=6.1.9->-r .\requirements.txt (line 1)) (4.2.0)
Requirement already satisfied: sniffio>=1.1 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from anyio<5,>=3.4.0->starlette<0.36,>=0.19->platformio>=6.1.9->-r .\requirements.txt (line 1)) (1.3.0)
PS C:\ESPEasy-mega> 
```

Also executing "pip3 install pygit2" I have this answer:
```
PS C:\ESPEasy-mega> pip3 install pygit2
Requirement already satisfied: pygit2 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (1.13.3)
Requirement already satisfied: cffi>=1.16.0 in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from pygit2) (1.16.0)
Requirement already satisfied: setuptools in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from pygit2) (69.0.3)
Requirement already satisfied: pycparser in c:\users\fabio\appdata\local\programs\python\python312\lib\site-packages (from cffi>=1.16.0->pygit2) (2.21)
```

TD-er, 2024-01-20:
Hmm as far as I can remember, we had specifically set the Python version to a lower version due to issues in the build proces.
But I don't remember what those issues were.
@tonhuisman Do you remember where and why we set it to a lower Python version?

tonhuisman, 2024-01-20:
We had to use Python 3.11 as pygit2 1.12.x did not support Python 3.12, but the latest pygit2 1.13.3 _should_ support Python 3.12 (according to the description in their RTD), so we can remove that restriction again, I think.
Locally I'm still on Python 3.10, and pygit2 works as expected, but I'll upgrade to 3.12 to see if that's an issue.

tonhuisman, 2024-01-20:
Just updated my system-Python to 3.12.1.
After a clean build my just-updated C3 unit reports this:
`INIT : Booting version: ESP_Easy_mega_20240120_normal_ESP32c3_4M316k_CDC, (Self built) mega_e015d1b (ESP32 SDK 4.4.6)`
So, `mega_e015d1b` is the current branch, and it also works just fine for any development branch.

NB: The Python version in my virtual env. seems to be 3.11.4, and updating that is a bit more tedious than updating the system Python version 🤔 
NB2: Even made a PlatformIO build with a Custom config, outside of VSCode, but it's still working as intended.

iz8mbw, 2024-01-22:
so guys, what I can/should do to don't have more the message "pygit2_not_installed"?
More, @tonhuisman it looks like your build you referred "ESP_Easy_mega_20240120_normal_ESP32c3_4M316k_CDC, (Self built) mega_e015d1b (ESP32 SDK 4.4.6)" is not using LittleFS (newest IDF 5.1.2).

tonhuisman, 2024-01-22:
Correct, but this issue is not related to IDF 5, but to the `pygit2` Python component not being installed correctly or not working as expected.

tonhuisman, 2024-03-03:
> so guys, what I can/should do to don't have more the message "pygit2_not_installed"?

@iz8mbw has this issue been resolved already?

If not, did you run it as instructed by TD-er, from a _PlatformIO: New Terminal_ window ? (either using the toolbar button or the menu entry in the PlatformIO Quick Access menu-area in VSCode)

Jason2866, 2024-03-04:
Usually Platformio installs it's own python pyenv. Running Python from CLI does not start the Platformio Python. So if issues arrives first easy step. Deleting hidden folder `.platformio` and Platformio will install everything from scratch (Python too).

iz8mbw, 2024-03-09:
still have this issue, also after removing ".platformio" directory.
I confirm Platformio installed everything from scratch (Python too), but when going to build I still have:

[image]

TD-er, 2024-03-09:
I can try to add a command to install required packages in the same call where I show exactly this text you show in your screenshot.
What Python version do you use?

TD-er, 2024-03-09:
Can you try this:

Create a file: `ESPEasy/tools/pio/install-requirements.py`
```python
Import("env")

import subprocess
import sys

subprocess.check_call([sys.executable, "-m", "pip", "install", "-r", "requirements.txt"])
```

Then in `platformio.ini` add it to the `extra_scripts_default` like this:
```
[extra_scripts_default]
extra_scripts             = pre:tools/pio/install-requirements.py
                            pre:tools/pio/set-ci-defines.py
                            pre:tools/pio/generate-compiletime-defines.py
                            tools/pio/copy_files.py
```

Jason2866, 2024-03-09:
@TD-er Like you suggested in your last post is the way to go. Like you do currently it is not safe that the python modules are installed in the Platformio Python env.

TD-er, 2024-03-09:
Ah cool didn't see the PIO docs, but used StackOverflow.
I will adapt it to `env.Execute`

TD-er, 2024-03-09:
So the `install-requirements.py` will be something like this:
```python
Import("env")

try:
    from pygit2 import Repository
except ImportError:
    env.Execute("$PYTHONEXE -m pip install -r requirements.txt")
```

Jason2866, 2024-03-09:
Mhh, you don't know which module is missing. You check maybe the wrong one, and the missing one gets not installed.

TD-er, 2024-03-09:
Yep, this is currently the one that may not be present.
And I don't like to duplicate the requirements.txt file here as it will eventually get out of sync.

I will have to think about it a bit more.

iz8mbw, 2024-03-09:
so what to try?
this https://github.com/letscontrolit/ESPEasy/issues/4943#issuecomment-1986831198 or this https://github.com/letscontrolit/ESPEasy/issues/4943#issuecomment-1986812986 ?

TD-er, 2024-03-09:
@iz8mbw 
My latest suggestion.
The suggested snippet from @Jason2866 still needs some tweaks for ESPEasy.

Jason2866, 2024-03-09:
You can "feed" `deps` from `requirements.txt`

TD-er, 2024-03-09:
> You can "feed" `deps` from `requirements.txt`

Yep, was already thinking about this.

iz8mbw, 2024-03-09:
@TD-er  ok so this https://github.com/letscontrolit/ESPEasy/issues/4943#issuecomment-1986829855 for "install-requirements.py" and this (last part) https://github.com/letscontrolit/ESPEasy/issues/4943#issuecomment-1986812986 for "platformio.ini"?
Right?

TD-er, 2024-03-09:
That's correct

iz8mbw, 2024-03-09:
ok, trying.

iz8mbw, 2024-03-09:
Now: "No_.git_dir":

[image]

iz8mbw, 2024-03-09:
but VSC is still building

TD-er, 2024-03-09:
Are you building from a git folder?
Thus is it created via `git ch3ckout` or `git clone` ?

iz8mbw, 2024-03-09:
Bulding by click on "Build":

[image]

TD-er, 2024-03-09:
That I understand, but it doesn't tell me if your source files are in a git folder or did you just download the ZIP file and extracted it?

A Git folder does have a (hidden) `.git` folder in them.

iz8mbw, 2024-03-09:
If for "download the ZIP file and extracted it" you mean ESPEasy the answer is yes, I donwlaoed it (ESPEasy-mega.zip) via GitHub

TD-er, 2024-03-09:
Then it isn't a Git folder so pygit2 cannot get some idea of any Git tags or Git branches from it.

iz8mbw, 2024-03-09:
Yes, not git folder insede "ESPEasy-mega.zip" downloaded file

iz8mbw, 2024-03-09:
Therefore?

TD-er, 2024-03-09:
> Therefore?

Not sure what you mean or like to happen here...
I can't extract information about a branch if the information simply isn't there.

It is just included in the sysinfo as extra information tag in case of support questions as I often also make test builds for people so it is useful for me to see from their posted sysinfo what source/branch may have been used.
For the official builds it isn't showing the branch name, but the latest tag so you can see something like the official git tag there.

iz8mbw, 2024-03-09:
I'm not an expert but I didn't understand how to fix this (original) issue when I build by myself:
[image]

TD-er, 2024-03-09:
The original issue was because the Python lib `pygit2` was not installed.

Just for completeness, please have a look at the code:
```python
def get_git_description():
    try:
        from pygit2 import Repository
        try:
            repo = Repository('.')
            return "{0}_{1}".format(repo.head.name
                                    .replace("refs/heads/","",1)
                                    .replace("refs/tags/","",1), repo.revparse_single('HEAD').short_id)
        except:
            return 'No_.git_dir'
    except ImportError:
        return 'pygit2_not_installed'
```

So if we can't import that library, you get the error `pygit2_not_installed`
Then we move on to the next part...
Try to get some information from the git folder.
Well you apparently don't have a `.git` folder as you didn't fetch the sources via Git, but downloaded the sources as a ZIP.

This doesn't make much of a difference if you don't plan on submitting any pull requests for code changes but merely want to build ESPEasy.
However if you plan on contributing to the project, you should use Git to maintain a record of any changes on either side (either in the original repository or on your forked repository).
So this is just about showing some information on what version of the source was used to build it.
But this is only possible if you fetched it using Git.

PR #5002 (2024-03-09): [PlatformIO] Fix installing pygit2
Fixes: #4943
commit 92383a82 (2024-03-09): [PlatformIO] Fix installing pygit2

Fixes: #4943
closed 2024-03-09 by TD-er