letscontrolit/ESPEasy #4955: [BUG] Degree symbol not shown in ESPEasy webinterface

bartgrefte, 2024-01-30:
**Describe the bug**
I have a SSD1306 display hooked up to a NodeMCU showing temps from two DS18B20 sensors. The degree-symbol, entered as & deg ; (with spaces to prevent parsing here), is shown on the display but not on the ESPEasy webinterface.

On the display I get 17.38&deg;C, on the webinterface 17.38C, since I didn't check the display right away I thought both had an issue with the degree symbol, turned out is was only the webinterface.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to config page for attached SSD1306 display
2. Enter for example [Temperatuur#Aanvoer]&deg;C on one of the line-fields (if sensor-value path applies)
3. Click submit
4. Go to ESP_IP-address/devices and see missing degree symbol next to the display-task in Values-column.

**Expected behavior**
That the degree-symbol is displayed on the ESPEasy webinterface as well.

**Screenshots**
[image]

**Used platform (please complete the following information):**
 - ESP type: ESP8266
 - Build version: mega-20231225
 - Build set: normal

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M1M
 - Brand/Model: NodeMCU
 - Browser + OS: Firefox, Windows 10 LTSB, on Chrome there's a space where the degree-symbol should be.

tonhuisman, 2024-01-30:
Hm, not sure if this can (or will) be fixed.
The content shown in the web UI, is the predefined content, or any inserted via commands, _after_ any variable replacements. Instead of using the degree symbol, you could have also used the `{D}` variable, that's replaced by the matching degree symbol for the display. For compatibility reasons, also some Unicode characters, like the degree you used, are translated to fit the display.

The character set used is limited to single byte characters (ASCII), in contrary to the Unicode character set used by browsers on most current OS-es like Windows, Mac OS, Linux, iOS, iPadOS and Android, and also adjusted to show correctly on the display (some special characters deviate from the most commonly used ASCII characters), so that implies it _may_ not show correct in the web UI.

And you've also seen that different browsers handle these characters differently, despite the fact they all claim to follow the HTML standards...

bartgrefte, 2024-01-31:
"you could have also used the {D} variable"
After initially having copy-pasted the degree symbol from somewhere, which wasn't shown in the webinterface (didn't check the display), I tried  {D} as well, same problem in the webinterface. Wasn't until after trying & deg ; I checked the display as well, because I figured that if the webinterface didn't show it, the display didn't either....

"And you've also seen that different browsers handle these characters differently, despite the fact they all claim to follow the HTML standards..."
I always thought it was just IE handling things differently than the rest, seeing a difference between Chrome and FF is a new one for me.