letscontrolit/ESPEasy #4956: [BUG] Switch state triggers on reboot or power on

MartinAngelov, 2024-01-31:
**Describe the bug**
I have NodeMCU ideaspark version. I have attached relay module (LOW trigger) and I control it via Switch input using Rules. So far all looks good, but when the device is rebooted form the web or cold boot it will cause Sensor#State event to be triggered

I also did test using Monitor GPIO,5 and a rule for its sate - all is stable and as expected, no unwanted triggers.

I have tested with and without boot state checked and with different pins for input and output. 

Not sure if it is a bug or expected behavior but for my case this makes no sense when wanting to use external physical switch.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure device Switch input:
-    GPIO 5
-    Name: Sensor
-    Internal PullUp: checked 
-    Inverse Logic: checked 
-    Switch type: Normal switch 
-    Send boot: tested with both checked and unchecked 
-    Value Name: State

3. Rule1:
```
   On Sensor#State=1 Do
       Pulse,4,0,300
   Endon
```
5. Reboot or plugin power - in few seconds once decice is booted rule will be triggered and relay will pulse as per Rule1

**Expected behavior**
I do expect when GPIO5 is shorted to trigger Rule1 and once reboot or power on Rule1 not to be executed because of Sensor#State

**Screenshots**
[image]
[image]
[image]

**Used platform (please complete the following information):**
 - ESP type: ESP8266
 - Build version: 20231225 - Mega
 - Build set normal

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M1M
 - Brand/Model: NodeMCU
 - Browser + OS: Edge Chromium, Windows 11

TD-er, 2024-01-31:
You could set the GPIO boot state on the hardware tab.
Since this is set before any rules are started, it should not trigger any event.

MartinAngelov, 2024-01-31:
I have edited my first post looks like the problem is not the Switch input but the GPIO behavior on boot.

MartinAngelov, 2024-02-01:
Further testing with Monitor, GPIO and disabled switch.

When GPIO-12 is configured to Internal PullUp on triggers the pulsation on boot, when it is set to Default the behavior is as expected - no action on boot and works as per rule logic. Also I don't see the gpio event in the log on reboot.

TD-er, 2024-02-01:
OK, so we have to look into what the switch task does when it is started.
You could also try with giving commands to disable/enable the switch task (this way nothing gets saved, thus much easier to test)

```
taskdisable,1
taskenable,1
```

Assuming the switch task is task nr 1.
Do you see an extra event when enabling the task?
Does it make a difference when you booted with the task disabled and then calling `taskenable` to enable the task?

MartinAngelov, 2024-02-01:
@TD-er , the results with details below.
When enabled with command:
```
212981: HTTP: taskenable,1
213108: EVENT: TaskInit#Sensor=1,1
213324: EVENT: Sensor#State=1
213329: ACT : Pulse 4,0,300
213633: GPIO : port 4. Pulse set for 300 ms
213665: WD : Uptime 4 ConnectFailures 0 FreeMem 21600 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
```

On reboot task enabled:

```
> 608845: WD : Uptime 10 ConnectFailures 0 FreeMem 20624 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
> 838: Set Network mode: WiFi
> 944: WIFI : Set WiFi to STA
> 1050: ESPEasy console using ESPEasySerial
> 1051: INIT : Free RAM:27016
> 1059: ESPEasy console using ESPEasySerial
> 1060: INFO : Plugins: 47 [Normal][No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
> 1062: EVENT: System#Wake
> 1151: WIFI : Connecting **redacted** (0dBm)- attempt #0
> 2163: Webserver: start
> 2176: EVENT: System#Boot
> 2201: EVENT: WiFi#APmodeDisabled
> 2277: EVENT: TaskInit#Sensor=1,1
> 2355: EVENT: Sensor#State=1
> 2359: ACT : Pulse 4,0,300
> 2666: GPIO : port 4. Pulse set for 300 ms
> 2881: WIFI : DHCP IP:  **redacted**  duration: 51 ms
> 2912: NTP : NTP replied: delay 10 mSec Accuracy increased by 636 msec
> 2914: Time set to 1706822183.637
> 2916: Current Time Zone: STD time start: 2024-10-27 03:00:00 offset: 0 min
> 2920: Local time: 2024-02-01 21:16:23
> 2925: UDP : Start listening on port 8266
> 2926: firstLoopConnectionsEstablished
> 2930: EVENT: WiFi#ChangedAccesspoint
> 2935: Webserver: stop
> 2938: Webserver: start
```

When disabled:

Config details:
```
 Hardware page:
 output pins: GPIO4, GPIO5 Internal pull-up (only 4 is used in rules)
 Input (switch) GPIO12 : Default
```
Switch configuration:
```
 GPIO12
 Normal switch
 Pull-up: ❌
 Reveres: input ✅
 Send boot state: ❌
```
Rule1:
```
On Sensor#State=1 Do
  Pulse 4,0,300
Endon
```

tonhuisman, 2024-02-02:
@MartinAngelov 
Trying to wrap my head around this, but so far I've been unable to replay your situation.
There seems to be some confusion (at least to me), as you initially write about a task named `Sensor`, but the rule you show is named `Card`, and they also have different GPIO pins assigned (5, 12 ?), when looking at to your later messages.

Please provide a clear picture of what GPIOs are used, and what's connected to them (switch? relay? )

MartinAngelov, 2024-02-03:
@tonhuisman, ahh my bad. I had 2 switches respectively Card and Sensor, and 2 relays on gpio 5 and gpio 4, now I do have only switch named Sensor on gpio12. Both had the same behaviour and that is why narrowed the topic to only 1 input. 

[image]
[image]

tonhuisman, 2024-02-04:
Is I2C enabled in your setup? As by default for ESP8266 that is configured for GPIO-4 and 5 ([also see docs](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)), which may cause the extra pulse(s) on these GPIO pins, but doesn't explain why this would happen on GPIO-12.

MartinAngelov, 2024-02-04:
I2C is disabled, I tested different pins, same thing is also happening with another device esp-wroom-32. Every time TaskInit is enabled it kicks the rule because of event switch state. If i remove the rule or disable the switch it won't send signal.

MartinAngelov, 2024-02-04:
@TD-er on **esp32 wroom device** when I send command **taskenable,1** I do see double event for switch state=1, this is because of send boot state with unchecked "send boot state" enabling the task sends only single event. I don't see same events on reboot in the log on **nodemcu based on esp8266**
[image]
 But on boot there is always a single click 
With boot state unchecked:
[image]

With boot state checked:
[image]