letscontrolit/ESPEasy #4957: ESP32 looses MQTT Broker connection

TommoT1, 2024-02-02:
Hi there!
I have two different ESP32 (Sonoff Dual R3),
espeasy release mega-20231225,
wich loose after some time the MQTT connection to the Broker. Sometimes it Takes some hours, Sometimes two dass.
Those units are in different locations and use different brokers but show the same logentry:

MQTT : Broker C005 connection failed (389/0)

In those two networks where the Sonoff Dual reside are more nodes, based on ESP8266 which show no problems staying conneced to MQTT.

Any ideas?

TD-er, 2024-02-02:
Can you see the disconnect reason in the logs of the MQTT broker?

Is the WiFi connection of this ESP unit stable?
Can you try to force to use 802.11g instead of 802.11n? (Force B/G on the tools->Advanced page)
Do you have ECO mode enabled? If so, try disabling it.
Which build are you using, one with "LittleFS" in the name or not? (if you change to a LittleFS build, please be aware the file system will be erased, so backup the settings and rules first)

TommoT1, 2024-02-02:
Must searchbfor the logs....

WiFi ist stable, -35 dBm,
Eco Mode disabled
Build ist without LittleFS, ESP_Easy_mega_20231225_normal_ESP32_4M316k

I set it now to forcw 892.11g....

TD-er, 2024-02-02:
If it is still a problem, can you pick a build from [this GH Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/7755079304)?

It is using the latest ESP-IDF commits and latest commits from the ESP32/Arduino repo.

tonhuisman, 2024-02-02:
..._if_ you pick one of the `LittleFS` builds... after making backups of your configuration (Tools/Save) and rules.

TommoT1, 2024-02-02:
I just ota flashed the normal ESP32 one from the Action build and will report...
Thanks!

TD-er, 2024-02-02:
Which "normal" one?
The "LittleFS" builds are based on latest ESP-IDF code, the others aren't.

But as mentioned before, switching to a LittleFS build will clear the file system and thus requires you to backup your settings and rules before updating.

TommoT1, 2024-02-02:
Ah ok got it....
No ,I took the non LittleFS build, than things have to wait... One node ist about 600km away, the other 200....
Thanks for your help

TD-er, 2024-02-02:
Hmm do these nodes also have to communicate with a MQTT broker which isn't in the same local network?

If so, can you please take a look at the timeout?
The default of 100 msec is perfect for local LAN, but it is probably too short for long distance (or even any online) host.

TommoT1, 2024-02-02:
Yes, thanks for the hint but the timeout is already at 1000ms...

TD-er, 2024-02-02:
You can turn the controller off/on.
```
controllerdisable,1
controllerenable,1
```
Assuming controller 1 is your MQTT controller.

Do you access your MQTT broker via IP or DNS hostname?

ESP32 does have a strange bug where it may not receive a DNS server via DHCP on each renew.
This will then clear the DNS server entry.
On each host lookup I check if this is cleared and if so, I restore the cached IP for DNS server.
However it might be possible the lookup somehow already failed and this gets cached in the ESP IDF code.
This would then explain why it keeps failing.

I will have a look at this to see if I can detect this failing.

If you restart WiFi, any caches will be flushed, so that would explain why a reconnect can succeed after you restart WiFi.

TommoT1, 2024-02-02:
Thanks for explanation, 
so if I get you right, then for now it might be better to stay with wifidisconnect instead of disabling and enabling the MQTT controller because all caches are flushed on wifireconnect.
Have a nice weekend!

TommoT1, 2024-02-02:
One more question:
Is it ok to just do a wifidisconnect and wait for the reconnect or is it saver to start a timer and do a explicit wifireconnect?

TD-er, 2024-02-02:
You can also set a "Connection Failure Threshold:" 
This is the number of failed reconnects after which ESPEasy will reboot.

However if the reconnect to MQTT broker fails quickly you may hit such a threshold quite fast and this may make it harder for you to reconnect to it to change some settings.

I think setting a timer on a MQTT#disconnect and clearing this on a MQTT#connected event is probably the most elegant solution.

TommoT1, 2024-02-02:
OK, I'll do it that way 👍

TD-er, 2024-02-02:
> The event ist called _**MQTT#Disconnected**_ right?

Yep
See: https://espeasy.readthedocs.io/en/latest/Reference/Events.html#internal-core-events

TD-er, 2024-03-30:
Not sure if this is "the same" issue as it might also be related to something else, but can you perhaps test the latest test build which is available via the [webflasher](https://td-er.nl/ESPEasy/latest/)

TommoT1, 2024-03-31:
Good morning!
I'll do so and report back

TD-er, 2024-03-31:
I would like to make a new build today, so unless you do find extremely tricky issues, I would like to continue with making the build.
When do you think you can have some update?

TommoT1, 2024-03-31:
Since about 4,5h no issues so far!

TommoT1, 2024-03-31:
So it seems solved! What about closing the issue?

TD-er, 2024-03-31:
Good idea :)

closed 2024-03-31 by TD-er