letscontrolit/ESPEasy #4967: [FR] No function of notification

comtechnik, 2024-02-07:
I have the latest version of ESPEasy. I use the “normal” version once and the “display” version once. In the “Display” version the notification works normally, but unfortunately this does not work in the “normal” version! But use the same SMTP settings for both. Is this possibly a bug?
Please help! Thanks!

TD-er, 2024-02-07:
Which exact build file did you use? (filename, see sysinfo tab)

comtechnik, 2024-02-07:
ESP_Easy_mega_20231225_normal_ESP8266_4M1M

comtechnik, 2024-02-07:
I don't understand your answer? Does that mean that I should create my own build? I don't understand why, because the "normal" version includes notification, but it doesn't work.

tonhuisman, 2024-02-07:
Can you see any message(s) in the log, related to notifications? Possibly error messages or connection failures?

TD-er, 2024-02-07:
This isn't an answer, but it is a label.
I did read your reply, but set it to unread for myself as I need to look into this later this evening.

With labels I can find things again later and also easily search through existing topics with the same label.
So for now it is: Have to dig into this a bit deeper.

comtechnik, 2024-02-07:
Sorry, thank you very much!

comtechnik, 2024-02-07:
This is what is written in the LOG when you press send:

16279149: Timeout while reading input data!
16280149: Timeout while reading input data!
16281149: Timeout while reading input data!
16282149: Timeout while reading input data!

TD-er, 2024-02-07:
Do you use DHCP or static assigned IP?
Can you check the DNS server and gateway IP addresses as shown on the sysinfo page?
Which SMTP port do you use?

comtechnik, 2024-02-07:
I use a static IP
the DNS and gateway are correct

comtechnik, 2024-02-07:
[image]
[image]

TD-er, 2024-02-07:
Can you try on the same node (with this settings) to use the display build (or switch to the normal build on the working node) ?
Just to make sure it isn't any setting or maybe even a conflict on your network (e.g. IP is used by some other device too, or some firewall block is present in the router for this IP)

The code for both builds should be the same. If there would be some difference, then it should be something like "Notification" is included or not included in a build.

comtechnik, 2024-02-07:
Don't quite understand what they mean. They are two separate devices, each has its own static IP. Not in the dhcp area. Both have the same notification settings, the same SMTP server, the same user name and the same password.

TD-er, 2024-02-07:
What you're seeing is this:
- 2 devices with same settings (apart from static IP) 
- Only difference on devices is build ("normal" vs "display" build
- device with "normal" build can't send, device with "display" build can

Since this should not happen, there are 3 options:
- There is some bug in ESPEasy (not impossible, but harder to check quickly)
- Settings perhaps not as "the same" as you would think
- Some external factor in your network causing this behavior.

The last 2 options can be excluded by simply flashing the "display" build to the node running the "normal" build.

If this then "solves" the issue of not being able to send, then it is clearly a bug in ESPEasy.
If the node still isn't able to send, then it must be either some setting which is different, or there is some external factor.

External factors can be:
- There is another device with that same IP present on your network
- Router/firewall may be set to block device on that IP to connect to some online service/host
- WiFi connection bad causing timeouts or slow reply from other hosts

comtechnik, 2024-02-07:
OK, first of all, thank you. I will take another Wemos d1 mini and flash it once with the “normal” version and then with the “display” version. I'll give it a completely new static IP address and try the whole thing.
I don't want to re-flash my original "normal" one because otherwise all my controllers and devices would be gone.

tonhuisman, 2024-02-07:
> I don't want to re-flash my original "normal" one because otherwise all my controllers and devices would be gone.

No, that's not true, you can update the Build installed, without losing your configuration, as long as the flash configuration (`4M1M`) is the same.
You can do an Over The Air (OTA) update via the Tools / Update Firmware button, browse to the 'other' .bin file, and upload that.

TD-er, 2024-02-07:
And if you're worried about loosing the configuration, you can download the config.dat and security.dat files from the node.

comtechnik, 2024-02-07:
I just flashed the "display" firmware via OTA. It's the same problem as with the "normal" firmware. With the same settings. So there must be a conflict somewhere in the network that the Fritzbox has another device with this IP in there somewhere. despite static ip

TD-er, 2024-02-07:
You can power off the unit with issues, to see if you can still ping it.
What is a bit strange is that you can apparently still access the web interface of the ESPEasy node, so I'm really curious to what other device might have the same IP.
Could it be another ESPEasy device?

comtechnik, 2024-02-08:
I'm always looking for a second device with the same IP address. Unfortunately in vain. There is only this one device with the IP in the network. I'm at a loss as to why this isn't working?

comtechnik, 2024-02-08:
No SMTP block, MQTT messages every 180 seconds, SMTP settings are correct otherwise it wouldn't work on the other device either.
I give up!
the same message always comes up.

211117: Timeout while reading input data!
212117: Timeout while reading input data!
213117: Timeout while reading input data!
214117: Timeout while reading input data!
215253: WD : Uptime 4 ConnectFailures 0 FreeMem 18104 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init

comtechnik, 2024-02-08:
If I press send on the other device with the "display" firmware, the message is there within 2 seconds.

TD-er, 2024-02-08:
Finally had some time to look into this...

That timeout error is being generated in the function `safeReadStringUntil`
In the email code it is trying to read until a newline.
Is it possible the mail body doesn't have a newline?

It is a bit hard to see from the source alone what might be wrong here.

Can you send me a zip file containing the notification.dat of both nodes via email? (maybe also config.dat)
Or you can perhaps try to diff both (binary) files to see where there is a difference.

comtechnik, 2024-02-08:
I don't change the email text. on both devices.? Which email address can I send the zip file to?

TD-er, 2024-02-08:
Another idea....
Since the error is reported in waiting for a response of the mail server, could it be that the mail server still keeps the network connection open for some odd reason?
So you could maybe also disconnect the working one and try again with the not working one...

email: gijs <dot> noorlander <at> gmail <dot> com

comtechnik, 2024-02-08:
I sent you the config.bat and notifications.bat from both devices by email. Unfortunately I can't compare them. Is there software that can be used to open and view the files.

comtechnik, 2024-02-08:
I saw that I had a blank line in the message in the "normal" device, I took it out and now everything works. Thank you very much for your great help! you are the best. Thanks alot! I opened the two notificatons.dat in notepad++ and saw that there were 2 lines and in the display notifications.dat there was only one line. perfect!

TD-er, 2024-02-08:
Still ESPEasy shouldn't handle it like this, so I will take another look at it.

I used `hexdump -C <filename>` under Linux to create the hexdump files and then just `diff` to see the differences

You can also hexdump with Notepad++, then save those to a separate file and then use any diff tool (also one in Notepad++ if I'm not mistaken) to compare 2 files

comtechnik, 2024-02-08:
thank you for everything!
They were a great help to me

closed 2024-02-08 by TD-er

TD-er, 2024-02-08:
Ah found the bug.....
```c++
    aMesg.replace(F("\r"), F("<br/>")); // re-write line breaks for Content-type: text/html
```
This should check for the char `'\r'` and not the string `"\r"`

Will fix it for the next build.

comtechnik, 2024-02-09:
So it's a little problem with ESPEasy. If ESPEasy should still process the blank line. That would be great if it was fixed in the next build, THANK YOU.
I read that you are the owner and main developer of ESPEasy. I think ESPEasy is very good! TOP!
Thank you

commit f56453f2 (2024-02-08): [Email] Fix sending multiple lines in body (#4967)

Fixes: #4967